MichaIng/DietPi #2885: Grocy ERP Beyond Fridge

fanul, 2019-06-03:
### Creating a software request

#### Formal software information
- Software title | Grocy
- Short description | grocy is a web-based self-hosted groceries & household management solution for your home.
- Official URL (if available) | https://grocy.info/
- GitHub URL (if available) | https://github.com/grocy/grocy
- (Official) install documentation | https://github.com/grocy/grocy

#### Are there similar/alternative software titles available with DietPi-Software?
none that i know of

#### What makes your requested software better than the above solutions, if available?
its fully opensources and its needed for home enthusiast

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?
According to the official install instructions there are a lot of manual steps to do: Setting up a webserver-database-PHP stack and configure them accordingly. DietPi could automate those steps, which saves end users a lot of time and allows installation as well to less experienced Linux users.

#### Can you provide the installation steps that you would suggest DietPi-Software to do?
1. Install webserver stack
2. Install Redis and the related PHP module, which is recommended
3. Adjust/Add the following PHP settings: enable sqllite (PDO extension)
4. copy Github material
5. done

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.
no

MichaIng, 2019-06-03:
@fanul 
Many thanks for your request. We'll have a look when we can.

LexiconCode, 2020-03-24:
Some up-to-date install instructions that are pretty simple
https://github.com/grocy/grocy/issues/201#issuecomment-487074462

MichaIng, 2021-09-24:
The FeatHub link to vote for it: https://feathub.com/MichaIng/DietPi/+90