MichaIng/DietPi #2886: Make backups with locations beyond /mnt

diveyez, 2019-06-03:
## Non standard /mnt/ for backups are needed

I encountered an issue where a previous system labeled all the large media devices and now, I just cannot backup due to this. Is there a work around or do I need to make this feature request and work one myself?

MichaIng, 2019-06-03:
@diveyez 
Many thanks for your request.

You mean e.g. `/media`?
It is strange since on DietPi-PREP the `fstab` should have been reset, so all non-root/non-boot drives should get new mount points inside `/mnt`.

Could you please give step-by-step instructions how you ran into this issue, starting with the base image you used?

diveyez, 2019-06-03:
Hmm, thats odd.

diveyez, 2019-06-03:
openmediavault forces mounts at /srv/dev-disk-by-label-<name>/
that is your culprit bro!

MichaIng, 2019-06-03:
@diveyez 
That is the reason why we dropped OMV support a while ago. It simply is too intrusive, requires specific uncommon setups and besides covers much of what DietPi aims in it's own way, so you have many features doubled, conflicting each other. So either go with OMV OR with DietPi is the right way. OMV even states itself to be an own OS 😉.

But generally there is no reason to not allow `dietpi-backup` to target locations outside of /mnt/, as long as it is a dedicated drive, which can be checked.

Until then there are many alternatives to `dietpi-backup` that you are able to use, most likely even OMV has it's own backup system?

diveyez, 2019-06-04:
I modified it. ;) Been using it since the beta.
If you can get dietpi to do what it does, let me know!

diveyez, 2019-06-04:
> 
> 
> @diveyez
> That is the reason why we dropped OMV support a while ago. It simply is too intrusive, requires specific uncommon setups and besides covers much of what DietPi aims in it's own way, so you have many features doubled, conflicting each other. So either go with OMV OR with DietPi is the right way. OMV even states itself to be an own OS 😉.
> 
> But generally there is no reason to not allow `dietpi-backup` to target locations outside of /mnt/. It just needs to be added to the "exclude" list then.
> 
> Until then there are many alternatives to `dietpi-backup` that you are able to use, most likely even OMV has it's own backup system?

I just did a 6 hours sweep.
It appears that OMV mounted those drives 6 times each, confusing both me, and deb.
They really do not know how to sanitize their work.
I am clean, with no conflicting mount points and symbolic links. No need for prep and now, yes, I can make the mount point for the backup.

Recommend me a panel solution for SMB/etc for my lady? I need to replace OMV because she dont  know how to use anything linux lol

diveyez, 2019-06-12:
No recommends?
hmmm, interesting.

Rebuild of that system starts tomorrow! Deadlined!

MichaIng, 2019-06-12:
@diveyez 
How exactly should your setup look like? If you need local network file share, I generally NFS which is usually faster and more lightweight then SMB/CIFS/Samba. Windows also supports NFS for a while, however check this first if required: Control panel > Programs & Features > Enable/Disable Windows features > NFS services (or similar).

If you mount your network share on DietPi to /mnt/something, you can also use it as dietpi-backup target, working around the still missing feature.

MichaIng, 2021-12-04:
Implemented with: #5016

closed 2021-12-11 by MichaIng