MichaIng/DietPi #2888: Installing Motion Cam via dietpi-software manager fails - "you have held broken packages"

infinitejones, 2019-06-04:
#### Details:
- Date           | Wed  5 Jun 09:20:43 AEST 2019
- Bug report     | N/A
- DietPi version | v6.24.1 (MichaIng/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi Zero W (armv6l) (index=1)
- Kernel version | #1219 Tue May 14 21:16:38 BST 2019
- Distro         | stretch (index=4)
- Command        | G_AGI v4l-utils python python-dev curl libssl-dev libcurl4-openssl-dev libjpeg-dev zlib1g-dev motion
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:
1. Select Motion Cam for install via dietpi-software
2. ...

#### Expected behaviour:
Motion should install correctly

#### Actual behaviour:
Raspbian reports apt error

#### Additional logs:
```
Log file contents:
E: Unable to correct problems, you have held broken packages.
```
Further output:
```
root@weatherPi:~# apt install -f
Reading package lists... Done
Building dependency tree
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
root@weatherPi:~# apt-get install motion
Reading package lists... Done
Building dependency tree
Reading state information... Done
motion is already the newest version (4.0-1).
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
root@weatherPi:~# apt-get -y install v4l-utils python python-dev curl libssl-dev libcurl4-openssl-dev libjpeg-dev zlib1g-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
python is already the newest version (2.7.13-2).
python-dev is already the newest version (2.7.13-2).
curl is already the newest version (7.64.0-3).
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libcurl4-openssl-dev : Depends: libcurl3 (= 7.52.1-5+deb9u9) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```

infinitejones, 2019-06-04:
Further to this:

```
root@weatherPi:~# apt install libcurl3
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages will be REMOVED:
  curl libcurl4 php7.3-curl
The following NEW packages will be installed:
  libcurl3
0 upgraded, 1 newly installed, 3 to remove and 1 not upgraded.
Need to get 262 kB of archives.
After this operation, 523 kB disk space will be freed.
Do you want to continue? [Y/n]
Get:1 http://raspbian.melbourneitmirror.net/raspbian stretch/main armhf libcurl3 armhf 7.52.1-5+deb9u9 [262 kB]
Fetched 262 kB in 1s (217 kB/s)
(Reading database ... 30781 files and directories currently installed.)
Removing curl (7.64.0-3) ...
Removing php7.3-curl (7.3.4-2) ...
Removing libcurl4:armhf (7.64.0-3) ...
Selecting previously unselected package libcurl3:armhf.
(Reading database ... 30757 files and directories currently installed.)
Preparing to unpack .../libcurl3_7.52.1-5+deb9u9_armhf.deb ...
Unpacking libcurl3:armhf (7.52.1-5+deb9u9) ...
Setting up libcurl3:armhf (7.52.1-5+deb9u9) ...
Processing triggers for libc-bin (2.28-10+rpi1) ...
Processing triggers for php7.3-fpm (7.3.4-2) ...
root@weatherPi:~# apt-get -y install v4l-utils python python-dev curl libssl-dev libcurl4-openssl-dev libjpeg-dev zlib1g-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
python is already the newest version (2.7.13-2).
python-dev is already the newest version (2.7.13-2).
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 curl : Depends: libcurl4 (= 7.64.0-3) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
root@weatherPi:~# apt install libcurl4
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages will be REMOVED:
  libcurl3
The following NEW packages will be installed:
  libcurl4
0 upgraded, 1 newly installed, 1 to remove and 1 not upgraded.
Need to get 0 B/293 kB of archives.
After this operation, 46.1 kB of additional disk space will be used.
Do you want to continue? [Y/n]
(Reading database ... 30766 files and directories currently installed.)
Removing libcurl3:armhf (7.52.1-5+deb9u9) ...
Selecting previously unselected package libcurl4:armhf.
(Reading database ... 30757 files and directories currently installed.)
Preparing to unpack .../libcurl4_7.64.0-3_armhf.deb ...
Unpacking libcurl4:armhf (7.64.0-3) ...
Setting up libcurl4:armhf (7.64.0-3) ...
Processing triggers for libc-bin (2.28-10+rpi1) ...
root@weatherPi:~# apt-get -y install v4l-utils python python-dev curl libssl-dev libcurl4-openssl-dev libjpeg-dev zlib1g-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
python is already the newest version (2.7.13-2).
python-dev is already the newest version (2.7.13-2).
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libcurl4-openssl-dev : Depends: libcurl3 (= 7.52.1-5+deb9u9) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```

MichaIng, 2019-06-05:
@infinitejones 
Many thanks for your report.

This is related to the PHP7.3 install from Buster repo. `libcurl4-openssl-dev` needs to be installed from Buster as well. Please try:
```
sed -i 's/curl /curl libcurl4-openssl-dev /' /etc/apt/preferences.d/dietpi-php
apt inststall libcurl4-openssl-dev
```

closed 2019-07-03 by MichaIng