MichaIng/DietPi #2890: Image | New Raspberry Pi images

MichaIng, 2019-06-05:
Hey guys,
please help testing the new images for Raspberry Pi: https://dietpi.com/downloads/testing/
- If you like to stay on bleeding edge or help preparing DietPi to be fully compaitble with Raspbian Buster, check out the Buster image. Note that not yet all software titles have been tested throughoutly on Raspbian Buster, but we will help resolving issues quickly.

PeterLacknase, 2019-06-07:
Just worx. ;-)

`Bug report sent, reference code: ebd04696-bd2b-4963-8147-38ff2ebfe982`

Would You recommend a fresh installation any time or would a repo-upgrade suffice?

MichaIng, 2019-06-07:
@PeterLacknase 
Many thanks for testing. You tested the Stretch or Buster image, or both?

> Would You recommend a fresh installation any time or would a repo-upgrade suffice?

You mean from Stretch to Buster?
- This depends a bid on how much software/packages you have installed. You can simply try a dist-upgrade after doing a full SDcard backup/clone at best. At least according to my experience, Stretch => Buster worked much better than Jessie => Stretch or even Wheezy => Jessie that time.
- But generally, if you did not do too much custom system changes, I recommend a fresh install, since there will always be dirty leftovers, possibly not fully compatile settings files and stuff after a distro upgrade, which sometimes do not show up directly but at a later time. If you want to go that way, especially remember to create database .sql dumps of MySQL/MariaDB and in case other persistent databases, which is safer to restore than migrating the raw database files.
- Note that Raspbian Buster is still the testing branch. Debian Buster is close to release but Raspbian usually needs some months more to fully test/fix the customisations.

PeterLacknase, 2019-06-07:
Buster.

MichaIng, 2019-06-08:
Okay I moved both new RPi images in place now: https://dietpi.com/downloads/images/

closed 2019-06-08 by MichaIng