MichaIng/DietPi #2891: Docker Buster cant install on buster

Qarasique, 2019-06-05:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=24
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version | buster/sid
- Kernel version | 4.19.42-v7+ #1219 SMP Tue May 14 21:20:58 BST 2019 armv7l GNU/Linux
- SBC device | RPi 3 Model B+ (armv7l)
- Power supply used | 5V2A
- SDcard used | 32gb Sony

#### Additional Information (if applicable)
- Software title | Docker
- Was the software title installed freshly or updated/migrated - Freshly
- Can this issue be replicated on a fresh installation of DietPi - Yes

#### Steps to reproduce

1. ... Install docker on fresh machine through Dietpi-launcher

#### Expected behaviour

- Docker just works

#### Actual behaviour
- docker mirror show 404 error 
Err:4 https://download.docker.com/linux/raspbian buster Release
  404  Not Found [IP: 13.33.76.253 443]

#### Extra details
- Docker partly install but 'docker' command dont work, so it cant install without files from his repo

MichaIng, 2019-06-05:
@Qarasique 
Many thanks for your report. Will run some test on Debian Buster to see if it's an RPi-only issue or even kernel-related.

MichaIng, 2019-06-15:
Verified it's a bug with Buster of the official https://get.docker.com/ installer script. (or lets say the script simply does not support Buster yet)

```
		debian|raspbian)
			dist_version="$(sed 's/\/.*//' /etc/debian_version | sed 's/\..*//')"
			case "$dist_version" in
				9)
					dist_version="stretch"
				;;
				8)
					dist_version="jessie"
				;;
			esac
		;;
```
What needs to be added:
```
				10)
					dist_version="buster"
				;;
```

You can easily fix it:
```
sed -i 's/10/buster/' /etc/apt/sources.list.d/docker.list
G_AGUP
G_AGI docker-ce docker-ce-cli
```

MichaIng, 2019-07-01:
PR has been merged, Debian Buster will work now and I just added a workaround for Raspbian Buster, as there is no branch yet for it in the Docker repo: https://github.com/MichaIng/DietPi/commit/093ca14d749c9dcf8f6639a869bbf630a72e17cd

closed 2019-07-03 by MichaIng