MichaIng/DietPi #2892: Latest updates on https://dietpi.com ?

Sopor, 2019-06-05:
I was downloading DietPi_RPi-ARMv6-Stretch.7z but that seems to be an old version (6.21.1). Shouldn't those image files be the latest versions? It is annoying to start a new installation with a large update :(

MichaIng, 2019-06-06:
@Sopor 
It is expected that the images provided on your website are not always on the latest version of DietPi. Firstrun update anyway brings it on the current version. The update should go smoothly and quick since it is done before any software titles are installed, so the amount of incremental update steps is quite small. However if you experience any issues, let us know.

For RPi in particular I have two new images in the pipeline, which only requires 1-2 successful boot tests: https://github.com/MichaIng/DietPi/issues/2890

If you try out the Stretch image with success (expected, we only want to be failsafe here by testing one more than just a single RPi), I will move it in place to be the default download.

MichaIng, 2019-06-27:
New images are live for a while and again a new RPi4 capable (Buster-only) image is in pipeline. I mark this as closed. Feel free to reopen if required.

closed 2019-06-27 by MichaIng