MichaIng/DietPi #2893: Nextcloud internal server error via https, http up and running

benjaminpreiss, 2019-06-06:
### Creating a bug report/issue

#### Required Information
- DietPi version:

G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=24
G_DIETPI_VERSION_RC=1
G_GITBRANCH=master
G_GITOWNER=MichaIng

- Distro version: 9.9
- Kernel version:

Linux DietPi 4.9.162+ #1 SMP PREEMPT Sat Mar 23 15:52:32 CET 2019 aarch64 GNU/Linux

- SBC device:

Odroid N2 (aarch64)

- Power supply used: 12V, 2A
- SDcard used

#### Additional Information (if applicable)
- Software title: Nextcloud
- Was the software title installed freshly or updated/migrated?

Freshly installed

- Can this issue be replicated on a fresh installation of DietPi?

Yes.

#### Steps to reproduce

1. Issue can be replicated by following [this guide](https://www.digitalocean.com/community/tutorials/how-to-create-a-self-signed-ssl-certificate-for-apache-in-ubuntu-16-04) but changing document root in config files to `/var/www/nextcloud` instead of `/var/www/html`.
2. I dont have a firewall installed though, so skip that part.

#### Expected behaviour

- When accessing the cloud via https it should load (maybe with an error complaining about self-signed certificate)

#### Actual behaviour

- I get an internal server error when accessing the cloud via https. http access is working though.

#### Extra details

- I did configure apache to forward any http request to https. http access is only working when i disable this setting of course.

benjaminpreiss, 2019-06-06:
Okey, so i solved the problem...

Following the guide was completely correct except that I had to change my document root to `/var/www` instead of `/var/www/nextcloud`.

now http and https access is working correctly.

closed 2019-06-06 by benjaminpreiss

MichaIng, 2019-06-06:
@benjaminpreiss 
Jep the webservers document root on DietPi is /var/www. `dietpi-letsencrypt` btw does all the configuration automatically and you don't have the warnings in browser due to self-signed certificates.

And jep to support multiple websites on DietPi, webserver depending websites are always reachable via their sub directory then, e.g. my.domain.org/nextcloud. You can change that by changing the document root or setup some rewrite rule, but other `dietpi-software` installs might then not work OOTB or in cases it might be reverted, just keep that in mind.

benjaminpreiss, 2019-06-06:
@MichaIng Hmm, I thought I had selected dietpi-letsencrypt for installation...

I think I will do a new install again in a week or so to test it again.

MichaIng, 2019-06-06:
@benjaminpreiss 
If you did so, run `dietpi-letsencrypt` command to get and apply the certificate. The guide you linked is about manually applying a self-signed certificate in which case you don't require `dietpi-letsencrypt`.