MichaIng/DietPi #2894: Trying to toggle state to Enable WiFi Hotspot on Pi

murraykessell, 2019-06-07:
#### Details:
- Date           | Fri  7 Jun 13:55:16 AWST 2019
- Bug report     | N/A
- DietPi version | v6.24.1 (MichaIng/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B+ (armv7l) (index=3)
- Kernel version | #1219 SMP Tue May 14 21:20:58 BST 2019
- Distro         | stretch (index=4)
- Command        | systemctl start hostapd
- Exit code      | 1
- Software title | DietPi-Config

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
Job for hostapd.service failed because the control process exited with error code.
See "systemctl status hostapd.service" and "journalctl -xe" for details.
```

MichaIng, 2019-06-07:
@murraykessell 
Many thanks for your report.

Please try to upgrade to the Raspbian repo version: `G_AGI --reinstall hostapd`

murraykessell, 2019-06-08:
Thanks that's fixed it

closed 2019-06-08 by murraykessell

MichaIng, 2019-06-08:
@murraykessell 
Thanks for feedback. Will implement it as default into DietPi-Software.