MichaIng/DietPi #2895: installation: 7sa command not found

rozza-m, 2019-06-07:
#### Details:
- Date           | Fri  7 Jun 12:14:04 BST 2019
- Bug report     | N/A
- DietPi version | v6.21.1 (Fourdee/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B (armv7l) (index=3)
- Kernel version | #1159 SMP Sun Nov 4 17:50:20 GMT 2018
- Distro         | stretch (index=4)
- Command        | 7za x -y 7.7z -offmpeg_rpi
- Exit code      | 127
- Software title | DietPi-Software

#### Steps to reproduce:
Use dietpi-software to install Home-Assistant

#### Expected behaviour:
Home-assistant is installed.

#### Actual behaviour:
Installatino crashes out with message that "7za x -y 7.7z -offmpeg_rpi" command isn't found.

#### Extra details:
Dietpi 
told me to post this here :)
#### Additional logs:
```
Log file contents:
/DietPi/dietpi/func/dietpi-globals: line 1373: 7za: command not found
```

MichaIng, 2019-06-07:
@rozza-m 
Many thanks for your report. You flashed a fresh image, right?
- Please update it to the latest version first: `dietpi-update`, or
- You can also flash the image I have currently in testing pipeline, which does not have this issue: https://github.com/MichaIng/DietPi/issues/2890
- Or last but not least, it is also possible to simply install the required package: `G_AGI p7zip-full`

We dropped the need for this package with v6.22. I did a slight mistake to create the image already without it but still with stable code that required it :wink:. However resolves with any of the above steps.

rozza-m, 2019-06-08:
Thanks for your reply. It's not a brand new install and I hadn't done dietpi-update before installing. Of course, not having a brand new install is still a valid use case.
It was fixed by installing p7zip-full. If you're happy that the underlying issue is fixed for new installs please feel free to close (or I can). Thanks!

MichaIng, 2019-06-08:
@rozza-m 
Usually firstrun setup should do an update automatically as long as you do not cancel or have no valid network connection at that time. However, new image will be placed to official download dir/link, so jep, issue will be closed.

closed 2019-06-08 by MichaIng

rozza-m, 2019-06-09:
Thank-you for your attention and fast replies.