MichaIng/DietPi #2896: move dietpi from 4 to 8gb sd card

Sopor, 2019-06-07:
If i install DietPi on a 4GB SD card and i later want to create a new image of the 4GB card and write it to a 8GB SD card with Rufus or balenaEtcher. When i install DietPi it seems to do this automatically, but how do i expand it to use the full 8GB card?

PeterLacknase, 2019-06-07:
‘sudo dietpi-drive_manager‘

MichaIng, 2019-06-07:
@Sopor 
With `dietpi-drive_manager` one cannot expand the root file system directly, but it allows to enable the boot service that does this on next reboot: `dietpi-drive_manager` > `/dev/mmcblk0p2` > `Resize`

Or you enable the service manually, at best already on the running 4G SDcard system before cloning/copying it to the 8G SDcard: `systemctl enable dietpi-fs_partition_resize`

Sopor, 2019-06-07:
Thanks, now i know how to do it!

closed 2019-06-07 by Sopor