MichaIng/DietPi #2897: Installation of Home Assistant (Software) stuck

Gill-Bates, 2019-06-07:
`WARNING: seems you still have not added 'pyenv' to the load path.

# Load pyenv automatically by adding
# the following to ~/.bashrc:

export PATH="/home/homeassistant/.pyenv/bin:$PATH"
eval "$(pyenv init -)"
eval "$(pyenv virtualenv-init -)"
Downloading Python-3.6.3.tar.xz...
-> https://www.python.org/ftp/python/3.6.3/Python-3.6.3.tar.xz
Installing Python-3.6.3...

`

MichaIng, 2019-06-07:
@Gill-Bates 
It is expected that Home Assistant takes a very long time to install due to the pyenv and Python compilation. The 100% CPU load just verifies this.

Not sure about the warning, will have a look if this might imply an issue, but don't think so since it worked in the path and we didn't change something.

We are thinking about skipping the pyenv and start Home Assistant with the system-wide Python: https://github.com/MichaIng/DietPi/issues/2374
But this requires intense testing/verification since the reason for pyenv is that Python + modules and their versions meet exactly the needs to Home Assistant. And it a slight security enhancement as well.

So try to be partitioned for now. Which RPi model is it?

Gill-Bates, 2019-06-07:
I am using Versin 3. A Popup Message that the installation can take up to 1h before the installation starts would be helpful.

After installation the service is not bindend to Port 8123. went anything wrong? I get a timeout after trying to get access

MichaIng, 2019-06-24:
@Gill-Bates 
> Jun 08 00:54:32 dietpi homeassistant-start.sh[1060]: 2019-06-08 00:54:32 INFO
(MainThread) [homeassistant.setup] Setting up tts

I know from recent Mycroft AI setup that TTS compilation indeed can take a long while. I am just trying on VM now to check which steps take how long here.

MichaIng, 2019-10-15:
Added hint to online docs: https://dietpi.com/phpbb/viewtopic.php?p=10416#p10416
Reviewing install process and adding hint to dietpi-software with v6.27.

marioarranzr, 2019-11-02:
For me, it took nearly 20min with just the message `Installing Python-3.6.3.. `. So be patient, it finally makes it. 
In the meantime, you can open a new ssh connection and set the missing path `export PATH="/home/homeassistant/.pyenv/bin:$PATH"`

MichaIng, 2019-11-06:
PR up to inform about long install process and update Python to latest version: https://github.com/MichaIng/DietPi/pull/3220

closed 2020-01-01 by MichaIng

akiltopu, 2020-11-12:
I just installed home assistant with diet-pi latest version to my Pi3+ but i could not get the server running on port 8123
I just did a clean install but it gives me some errors. What can i do ? 

root@DietPi:~# service home-assistant status
● home-assistant.service - Home Assistant (DietPi)
   Loaded: loaded (/etc/systemd/system/home-assistant.service; disabled; vendor preset: enabled)
   Active: active (running) since Thu 2020-11-12 12:58:44 +03; 33min ago
 Main PID: 594 (hass)
    Tasks: 9 (limit: 2063)
   CGroup: /system.slice/home-assistant.service
           └─594 /home/homeassistant/.pyenv/versions/3.8.6/bin/python3.8 /home/homeassistant/.pyenv/versions/3.8.6/bin/hass
 -c /mnt/dietpi_userdata/homeassistant

Nov 12 13:00:11 DietPi homeassistant[594]: 2020-11-12 13:00:11 ERROR (MainThread) [homeassistant.setup] Unable to set up de
pendencies of map. Setup failed for dependencies: frontend
Nov 12 13:00:11 DietPi homeassistant[594]: 2020-11-12 13:00:11 ERROR (MainThread) [homeassistant.setup] Setup failed for ma
p: Could not set up all dependencies.
Nov 12 13:00:11 DietPi homeassistant[594]: 2020-11-12 13:00:11 ERROR (MainThread) [homeassistant.setup] Unable to set up de
pendencies of logbook. Setup failed for dependencies: frontend, http, recorder
Nov 12 13:00:11 DietPi homeassistant[594]: 2020-11-12 13:00:11 ERROR (MainThread) [homeassistant.setup] Setup failed for lo
gbook: Could not set up all dependencies.
Nov 12 13:00:11 DietPi homeassistant[594]: 2020-11-12 13:00:11 ERROR (MainThread) [homeassistant.setup] Unable to set up de
pendencies of tts. Setup failed for dependencies: http
Nov 12 13:00:11 DietPi homeassistant[594]: 2020-11-12 13:00:11 ERROR (MainThread) [homeassistant.setup] Setup failed for tt
s: Could not set up all dependencies.
Nov 12 13:00:11 DietPi homeassistant[594]: 2020-11-12 13:00:11 ERROR (MainThread) [homeassistant.setup] Unable to set up de
pendencies of media_source. Setup failed for dependencies: http
Nov 12 13:00:11 DietPi homeassistant[594]: 2020-11-12 13:00:11 ERROR (MainThread) [homeassistant.setup] Setup failed for me
dia_source: Could not set up all dependencies.
Nov 12 13:00:15 DietPi homeassistant[594]: 2020-11-12 13:00:15 ERROR (MainThread) [homeassistant.setup] Setup failed for up
dater: Unable to import component: No module named 'distro'
Nov 12 13:00:27 DietPi homeassistant[594]: 2020-11-12 13:00:27 ERROR (MainThread) [homeassistant.setup] Setup failed for ss
dp: Unable to import component: No module named 'defusedxml'

Thanks

Joulinar, 2020-11-12:
did you followed the installation steps as describe on our online docs? https://dietpi.com/docs/software/home_automation/#home-assistant-open-source-home-automation-platform-running-on-python-3

HA would need to be restarted 2 times to get installation done

akiltopu, 2020-11-12:
> did you followed the installation steps as describe on our online docs? https://dietpi.com/docs/software/home_automation/#home-assistant-open-source-home-automation-platform-running-on-python-3
> 
> HA would need to be restarted 2 times to get installation done

Thanks i just reboot the system again and do the instructions on the link. 

log changes to below, but i could not login to web console of home asistant :/

root@DietPi:~# service home-assistant status
● home-assistant.service - Home Assistant (DietPi)
   Loaded: loaded (/etc/systemd/system/home-assistant.service; disabled; vendor preset: enabled)
   Active: active (running) since Thu 2020-11-12 14:27:38 +03; 1min 48s ago
 Main PID: 605 (hass)
    Tasks: 11 (limit: 2063)
   CGroup: /system.slice/home-assistant.service
           ├─605 /home/homeassistant/.pyenv/versions/3.8.6/bin/python3.8 /home/homeassistant/.pyenv/versions/3.8.6/bin/hass -c /mnt/dietpi_userdata/homea
ssistant
           └─786 /home/homeassistant/.pyenv/versions/3.8.6/bin/python3.8 -m pip install --quiet pillow==7.2.0 --upgrade --constraint /home/homeassistant/
.pyenv/versions/3.8.6/lib/python3.8/site-packages/homeassistant/package_constraints.txt --user --prefix=

Nov 12 14:27:38 DietPi systemd[1]: Started Home Assistant (DietPi).
Nov 12 14:27:45 DietPi homeassistant[605]: 2020-11-12 14:27:45 WARNING (Recorder) [homeassistant.components.recorder] Ended unfinished session (id=1 from
 2020-11-12 11:20:09.428311)

akiltopu, 2020-11-12:
> did you followed the installation steps as describe on our online docs? https://dietpi.com/docs/software/home_automation/#home-assistant-open-source-home-automation-platform-running-on-python-3
> 
> HA would need to be restarted 2 times to get installation done

thanks it is now working 👍

Joulinar, 2020-11-12:
yes, it's taking quite some time. You could have watched `htop` to see how installation is going. 😃