MichaIng/DietPi #2898: update failed to v6.24.1 php 7.3

megusta-01, 2019-06-07:
#### Details:                                                                                                                                                               
- Date           | Fri  7 Jun 22:26:52 CEST 2019                                                                                                                            
- Bug report     | N/A                                                                                                                                                      
- DietPi version | v6.22.3 (Fourdee/master)                                                                                                                                 
- Img creator    | DietPi Core Team                                                                                                                                         
- Pre-image      | Raspbian Lite                                                                                                                                            
- SBC device     | RPi Zero W (armv6l) (index=1)                                                                                                                            
- Kernel version | #1200 Tue Feb 12 20:11:02 GMT 2019                                                                                                                       
- Distro         | stretch (index=4)                                                                                                                                        
- Command        | G_AGI php7.3-fpm php7.3-apcu php7.3-curl php7.3-gd php7.3-mbstring php7.3-xml php7.3-zip php7.3-sqlite3                                                  
- Exit code      | 100                                                                                                                                                      
- Software title | DietPi-Software                                                                                                                                          

#### Steps to reproduce:                                                                                                                                                    
i maked dietpi-update                                                                                                                                                                   

#### Expected behaviour:                                                                                                                                                    
an update                                                                                                                                                                    

#### Actual behaviour:                                                                                                                                                      
wifi is broken                                                                                                                                                                   

#### Extra details:                                                                                                                                                         
first time it works, until a maked a restore and installed unbound 1.6                                                                                                                                                                

#### Additional logs:                                                                                                                                                       
```                                                                                                                                                                         
Log file contents:                                                                                                                                                          
Setting up php7.3-fpm (7.3.4-2) ...                                                                                                                                         
Job for php7.3-fpm.service failed because a fatal signal was delivered to the control process.                                                                              
See "systemctl status php7.3-fpm.service" and "journalctl -xe" for details.                                                                                                 
invoke-rc.d: initscript php7.3-fpm, action "start" failed.

MichaIng, 2019-06-07:
@megusta-01 
Many thanks for your report.

Initial failure is php7.3-fpm. Please paste:
```
journalctl -u php7.3-fpm
cat /var/log/php7.3-fpm.log
```

> wifi is broken

Hmm `libssl1.1` is upgraded and [unbound](https://packages.debian.org/stretch/unbound) at least depends on it. Not sure if it stops/stuck because of this which then breaks DNS resolving. Although `resolvconf` should be still available 🤔. What you mean by WiFi is broken, is the connection/interface completely down or does only connections to domains fail (due to DNS resolving fails)? `ping 8.8.8.8` does this still succeed?

Perhaps unbound needs to be installed from the Buster repo as well, but lets first check why PHP-FPM is failing.

megusta-01, 2019-06-10:
if i make ifup wlan0, i get a message:
wpa_supplicant daemon failed to start / exited return code 1 / failed to bring up wlan 0

journal php says:
main process exited, code=killed, status 4/ILL
failed to start php 7.3 fast cgi
unit entered failed state
failed with result "signal"

i use raspi headless and log to ram, so without wifi log file is lost :(
i try again, maybe i get something

MichaIng, 2019-06-10:
@megusta-01 
Why you check PHP log when WiFi fails?

Please paste: `journalctl -t wpa_supplicant`

MichaIng, 2019-06-10:
@megusta-01 
Okay lets check WiFi first, please paste: `journalctl -t wpa_supplicant`

megusta-01, 2019-06-10:
so, i maked a restore again, uninstall unbound, update can't install php7.3
i hope it's not to much

un 10 16:08:05 DietPi systemd[1]: Starting The PHP 7.3 FastCGI Process Man…r...
Jun 10 16:08:05 DietPi systemd[1]: php7.3-fpm.service: Main process exited,…/ILL
Jun 10 16:08:05 DietPi systemd[1]: Failed to start The PHP 7.3 FastCGI Proc…ger.
Jun 10 16:08:05 DietPi systemd[1]: php7.3-fpm.service: Unit entered failed …ate.
Jun 10 16:08:05 DietPi systemd[1]: php7.3-fpm.service: Failed with result '…al'.
Hint: Some lines were ellipsized, use -l to show in full.
dpkg: error processing package php7.3-fpm (--configure):
 subprocess installed post-installation script returned error exit status 1
Setting up php-apcu (5.1.17+4.0.11-1) ...
Obsolete conffile /etc/php/7.0/mods-available/apcu.ini has been modified by you.
Saving as /etc/php/7.0/mods-available/apcu.ini.dpkg-bak ...
Setting up libcurl4:armhf (7.64.0-3) ...
Setting up php7.3-curl (7.3.4-2) ...

Creating config file /etc/php/7.3/mods-available/curl.ini with new version
Setting up curl (7.64.0-3) ...
Setting up php-curl (2:7.3+69) ...
Processing triggers for libc-bin (2.28-10+rpi1) ...
Processing triggers for systemd (232-25+deb9u11) ...
Errors were encountered while processing:
 php7.3-fpm
E: Sub-process /usr/bin/dpkg returned an error code (1)
[FAILED] DietPi-Software | G_AGI php7.3-fpm php7.3-apcu php7.3-curl php7.3-gd php7.3-mbstring php7.3-xml php7.3-zip php7.3-sqlite3

--------------------------------------------------------------------                                                                                                        
- DietPi has encounted an error, and, is unable to continue        -                                                                                                        
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -                                                                                                       
- Copy and paste the BLUE lines below, into the ticket             -                                                                                                        
--------------------------------------------------------------------                                                                                                        

#### Details:                                                                                                                                                               
- Date           | Mon 10 Jun 16:09:51 CEST 2019                                                                                                                            
- Bug report     | N/A                                                                                                                                                      
- DietPi version | v6.22.3 (Fourdee/master)                                                                                                                                 
- Img creator    | DietPi Core Team                                                                                                                                         
- Pre-image      | Raspbian Lite                                                                                                                                            
- SBC device     | RPi Zero W (armv6l) (index=1)                                                                                                                            
- Kernel version | #1200 Tue Feb 12 20:11:02 GMT 2019                                                                                                                       
- Distro         | stretch (index=4)                                                                                                                                        
- Command        | G_AGI php7.3-fpm php7.3-apcu php7.3-curl php7.3-gd php7.3-mbstring php7.3-xml php7.3-zip php7.3-sqlite3                                                  
- Exit code      | 100                                                                                                                                                      
- Software title | DietPi-Software                                                                                                                                          

#### Steps to reproduce:                                                                                                                                                    
                                                                                                                                 
1. ...                                                                                                                                                                      
2. ...                                                                                                                                                                      

#### Expected behaviour:                                                                                                                                                    
                                                                                                                                          
[... cut ...]
Jun 10 15:50:07 DietPi wpa_supplicant[3104]: wlan0: WPA: Group rekeying completed with 44:4e:6d:3c:9d:cc [GTK=CCMP]
Jun 10 16:00:07 DietPi wpa_supplicant[3104]: wlan0: WPA: Group rekeying completed with 44:4e:6d:3c:9d:cc [GTK=CCMP]
Jun 10 16:10:07 DietPi wpa_supplicant[3104]: wlan0: WPA: Group rekeying completed with 44:4e:6d:3c:9d:cc [GTK=CCMP]
root@DietPi:~# ping google.de
PING google.de (172.217.16.67) 56(84) bytes of data.
64 bytes from ham11s01-in-f3.1e100.net (172.217.16.67): icmp_seq=1 ttl=56 time=8.59 ms

megusta-01, 2019-06-10:
root@DietPi:~# journalctl -u php7.3-fpm

-- Logs begin at Sat 2019-06-08 16:26:55 CEST, end at Mon 2019-06-10 16:20:07 CEST. --
Jun 10 16:08:05 DietPi systemd[1]: Starting The PHP 7.3 FastCGI Process Manager...
Jun 10 16:08:05 DietPi systemd[1]: php7.3-fpm.service: Main process exited, code=killed, status=4/ILL
Jun 10 16:08:05 DietPi systemd[1]: Failed to start The PHP 7.3 FastCGI Process Manager.
Jun 10 16:08:05 DietPi systemd[1]: php7.3-fpm.service: Unit entered failed state.
Jun 10 16:08:05 DietPi systemd[1]: php7.3-fpm.service: Failed with result 'signal'.
Jun 10 16:10:52 DietPi systemd[1]: Starting The PHP 7.3 FastCGI Process Manager...
Jun 10 16:10:52 DietPi systemd[1]: php7.3-fpm.service: Main process exited, code=killed, status=4/ILL
Jun 10 16:10:52 DietPi systemd[1]: Failed to start The PHP 7.3 FastCGI Process Manager.
Jun 10 16:10:52 DietPi systemd[1]: php7.3-fpm.service: Unit entered failed state.
Jun 10 16:10:52 DietPi systemd[1]: php7.3-fpm.service: Failed with result 'signal'.
Jun 10 16:11:12 DietPi systemd[1]: Starting The PHP 7.3 FastCGI Process Manager...
Jun 10 16:11:12 DietPi systemd[1]: php7.3-fpm.service: Main process exited, code=killed, status=4/ILL
Jun 10 16:11:12 DietPi systemd[1]: Failed to start The PHP 7.3 FastCGI Process Manager.
Jun 10 16:11:12 DietPi systemd[1]: php7.3-fpm.service: Unit entered failed state.
Jun 10 16:11:12 DietPi systemd[1]: php7.3-fpm.service: Failed with result 'signal'.

cat /var/log/php7.3-fpm.log
error: no file at the direction

megusta-01, 2019-06-10:
one thing i wounder, i've php 7.0 and 7.3 installed parallel

[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : webmin
[  OK  ] DietPi-Services | stop : home-assistant
[  OK  ] DietPi-Services | stop : mosquitto
[  OK  ] DietPi-Services | stop : aria2
[  OK  ] DietPi-Services | stop : logitechmediaserver
[  OK  ] DietPi-Services | stop : squeezelite
[  OK  ] DietPi-Services | stop : shairport-sync
[  OK  ] DietPi-Services | stop : lighttpd
[  OK  ] DietPi-Services | stop : php7.3-fpm
[  OK  ] DietPi-Services | stop : php7.0-fpm
[  OK  ] DietPi-Services | stop : smbd
[  OK  ] DietPi-Services | stop : nmbd
[  OK  ] DietPi-Services | stop : avahi-daemon

MichaIng, 2019-06-10:
@megusta-01 
Try: `G_AGP php7.0-* && G_AGA`
Then: `G_AGF && dpkg-configure php7.3-fpm`

megusta-01, 2019-06-10:
> Ah you have phpMyAdmin installed, right?

i'm not sure, what is the package name, phpmyadmin? if i list packages there is no phpmyadmin

>  Try: G_AGP php7.0-* && G_AGA
Then: G_AGF && dpkg-configure php7.3-fpm

both fails, if i run restart php:

php.service fails because a fatal signal was deliverd to the control process. 

journal -xe:
wlan0 is not configured / code 1
ifup: failed to bring up wlan0

MichaIng, 2019-06-10:
@megusta-01 
> if i list packages there is no phpmyadmin

Okay, since there is some package that depends on `php-fpm` `php-gd` `php-...` meta packages. However should not hurt.

> php.service fails because a fatal signal

You mean `php7.3-fpm`, right? There is no `php.service`.
And no log file was created at `/var/log/php7.3-fpm.log`?

> wlan0 is not configured / code 1

So there is clearly something deeper broken. Have you tried `G_AGI haveged`?
Also please paste: `dmesg | tail`

megusta-01, 2019-06-10:
i've package 
php7.3-fpm/testing,testing,now 2:7.3+69 all
installed, is this right so or i've installed wrong package? (testing rep)

MichaIng, 2019-06-10:
@megusta-01 
This is correct.

megusta-01, 2019-06-10:
> G_AGI haveged
fails also, code 100 / no internet connection

> dmesg | tail

brcmfmac: power managemet disabled

MichaIng, 2019-06-10:
@megusta-01 
> fails also, code 100 / no internet connection

Ah of course, me <<< dump 😉.

Hum very strange, so you need to first get the WiFi working again. Perhaps some reboot helps or redo the config via `dietpi-config` > `Network Options: Adapters`. And you said you removed `unbound`, right? As it seems to conflict with resolvconf or something.

megusta-01, 2019-06-10:
> Network Options: Adapters

i tried different options, nothing works

>  removed unbound?

yes, now I maked restore again, removed unbound, webmin and pihole, i could update to 6.24 but dhcpcd.service doesn't work anymore. This problem i had also before. Linux makes me crazy, arggh

root@DietPi:~# systemctl status dhcpcd.service
● dhcpcd.service - dhcpcd on all interfaces
   Loaded: loaded (/lib/systemd/system/dhcpcd.service; enabled; vendor preset: enabled)
  Drop-In: /etc/systemd/system/dhcpcd.service.d
           └─wait.conf
   Active: failed (Result: exit-code) since Mon 2019-06-10 21:12:04 CEST; 8min ago

Jun 10 21:12:03 DietPi systemd[1]: Starting dhcpcd on all interfaces...
Jun 10 21:12:03 DietPi dhcpcd[206]: Not running dhcpcd because /etc/network/interfaces
Jun 10 21:12:03 DietPi dhcpcd[206]: defines some interfaces that will use a
Jun 10 21:12:03 DietPi dhcpcd[206]: DHCP client or static address
Jun 10 21:12:04 DietPi systemd[1]: dhcpcd.service: Control process exited, code=exited status=6
Jun 10 21:12:04 DietPi systemd[1]: Failed to start dhcpcd on all interfaces.
Jun 10 21:12:04 DietPi systemd[1]: dhcpcd.service: Unit entered failed state.
Jun 10 21:12:04 DietPi systemd[1]: dhcpcd.service: Failed with result 'exit-code'.

MichaIng, 2019-06-11:
@megusta-01 
dhcpcd is not intended to run and not the DHCP client we use (ics-dhcp-client/dhclient). I know Pi-hole installs it for some not too clear reason (to me) to then enable a static IP.

Simple remove it: `G_AGP dhcpcd5`

megusta-01, 2019-06-11:
thx for quick support and information, it seems an unbound problem

closed 2019-06-11 by megusta-01