MichaIng/DietPi #2899: If i change hostname it will disconnect me

Sopor, 2019-06-07:
If i change the name `DietPi` it will lost connection to the RPi. Is this normal? I manage it over SSH so after i have change the name i get disconnected and can't login again. The RPi is connected to a public IP address with a ddns and even if i try to connect with the IP address it won't work. It is not even reply to a ping :(

I wondering if it has something to do with `Search domains`?

Sopor, 2019-06-07:
If i install it behind a router it will show `localdomain` in the `Search domains` but where will it get this information from?

Sopor, 2019-06-07:
I think i found it `/run/systemd/resolve/resolv.conf`

MichaIng, 2019-06-07:
@Sopor 
`/etc/resolv.conf` contains the DNS resolving info btw, but with `resolvconf` package it is a symlink to `/etc/resolvconf/run/resolv.conf`

I am not sure what/how webmin does things, but usually changing the hostname should not affect connectivity. But in case behind a router it depends on a router how it handles it. Possibly some routers do not only bind assigned local IPs to the MAC address but also to the hostname, and, disconnect devices or reassign a new IP if the hostname changes, not sure. But this is not related to DietPi.

That you loose connectivity when being connected to www directly (no NAT), then I am even more confused. The local hostname btw. in this case should usually match the FQDN/domain where the device is reachable, but it should still not affect connectivity, especially not via bare IP.

Did you try to change the hostname via `dietpi-config` instead? I am just doing a webmin test install to see how/what it does.

Sopor, 2019-06-07:
After i did a reboot i can now login again. This is the one running in my LAN. The other one i need to wait for assistance because it is not located where i am. It is located severals kilometres from me. So it is a bit frustrating when i can't connect again :(

MichaIng, 2019-06-07:
Confirmed, changing the hostname through webmin breaks connection completely. Actually it is only changing two files, no idea what webmin does there... It seems to completely bring down network interfaces, why ever... Even pinging the static local IP fails.

Sopor, 2019-06-08:
I will ask Jamie and see what he is saying about it, but now it is time for the bed. Thanks for your help.
I don't know if sourceforge is used anymore? I will use https://github.com/webmin/webmin/issues

MichaIng, 2019-06-08:
@Sopor 
> but now it is time for the bed

Dito 😴.

> I don't know if sourceforge is used anymore? I will use https://github.com/webmin/webmin/issues

Ah perfect, jep much easier then to crosslink issues, just found the sourceforge site starting from http://webmin.com/.

MichaIng, 2019-06-24:
Ref: https://github.com/webmin/webmin/issues/1067

MichaIng, 2019-06-24:
@Sopor 
As stated in your webmin issue, the reason is indeed related to the way we add interfaces via `allow-hotplug`. `ifdown -a && ifup -a` or `systemctl restart networking` deconfigures `hotplug` devices but only reconfigures `auto` devices.

So as a workaround, when using Webmin, you should do `sed -i 's/allow-hotplug/auto/g' /etc/network/interfaces` to switch all interfaces to auto, which as well allows Webmin to detect them correctly as active.