MichaIng/DietPi #2900: Two IP addresses?

Sopor, 2019-06-08:
I have assigned 192.168.0.10 to my RPi but still it will show something else in the `interfaces`

```
# Local
auto lo
iface lo inet loopback

# Ethernet
allow-hotplug eth0
iface eth0 inet dhcp
	address 192.168.0.100
	netmask 255.255.255.0
	gateway 192.168.0.1
	dns-nameservers 192.168.0.1
#dns-nameservers 8.8.8.8 8.8.4.4
```

As you can see here in Webmin it will show 192.168.0.10

[image]

And 192.168.0.100

[image]

Why i'm asking is that i have problem to connect to servers i'm running on the RPi. What i know, the iptables is not installed in DietPi, correct? Do you have any other firewall running in DietPi?

MichaIng, 2019-06-09:
@Sopor 
Many thanks for your report.

Yeah it looks like Webmin is incompatible with the way DietPi creates `/etc/network/interfaces` and the other way round. What I already found in your other opened issue:
- `allow-hotplug eth0` is not detected as active interface by Webmin, even that/if it is active. Webmin expects and uses `auto eth0`, which has the issue that it will fail on boot when the cable is not connected or WiFi not available on boot. `allow-hotplug eth0` is actually the quite common default for this reason.
  - I guess this is also the source of your doubled IP issue: Since the `allow-hotplug eth0` part for Webmin is not "active", I guess it creates a new interface somwhere which is then detected as status "Up/active" while the other one shown as "Down/not active".
- Webmin re-arranges (one tab forward) the interfaces address/netmask/gateway, which breaks `dietpi-config` detection, AFAIK. The latter would be not hard to solve.

Please paste: `cat /etc/network/interfaces{,.d/*}`

Sopor, 2019-06-09:
```
$ cat /etc/network/interfaces{,.d/*}
#/etc/network/interfaces
#Please use DietPi-Config to modify network settings.

# Local
auto lo
iface lo inet loopback

# Ethernet
allow-hotplug eth0
iface eth0 inet dhcp
        address 192.168.0.100
        netmask 255.255.255.0
        gateway 192.168.0.1
        dns-nameservers 192.168.0.1
#dns-nameservers 8.8.8.8 8.8.4.4

# Wifi
#allow-hotplug wlan0
iface wlan0 inet dhcp
address 192.168.0.100
netmask 255.255.255.0
gateway 192.168.0.1
wireless-power off
wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
#dns-nameservers 8.8.8.8 8.8.4.4
cat: '/etc/network/interfaces.d/*': No such file or directory
```

MichaIng, 2019-06-09:
@Sopor 
Okay so the "active" interface is not added to the file at all. My next guess is now, when I see the two menus:
- "Edit Active Interface" means the changes are only applied to the current interface session, so applied most likely via `ip a` command and are not boot persistent.
- "Edit Bootup Interface" is the menu to apply changes boot persistent, thus changes here are stored to `/etc/network/interfaces`.

You can compare:
`ip a` # Should show `192.168.0.10` as currently active IP for the eth0 interface.
But after reboot, 192.168.0.100 will be applied again.

Sopor, 2019-06-09:
I do use DHCP but i have it locked at 192.168.0.10 in pfsense, so don't misunderstand me. I have not set a static IP.

MichaIng, 2019-06-09:
@Sopor 
Ah yes sorry of course. Then
```
        address 192.168.0.100
        netmask 255.255.255.0
        gateway 192.168.0.1
        dns-nameservers 192.168.0.1
```
has no influence anyway. Strange that Webmin obviously still show this info even that it is a DHCP interface. I bet if you edit the address field to `address 192.168.0.10`, the "Edit Bootup Interface" menu will reflect his.

diveyez, 2019-06-12:
Duplicate #2899

Both issues related to each other.

MichaIng, 2019-06-12:
@diveyez 
Yeah, however two different issues. I leave them open until reported to Webmin.

Sopor, 2019-06-12:
@MichaIng There is even more incompatible between Webmin and DietPi. You can check the issue https://github.com/authentic-theme/authentic-theme/issues/1339
It seems that the fake Cron job you have created make the edit_cron.cgi into a loop that will create very high CPU usage.

MichaIng, 2019-06-12:
@Sopor 
Thanks for the hint, see my reply: https://github.com/authentic-theme/authentic-theme/issues/1339#issuecomment-501459105

MichaIng, 2019-06-24:
I opened an issue on Webmin GitHub page to implement correct `allow-hotplug` detection and optionally allowing to explicitly select hotplug as third interface state. For external (e.g. USB) adapters this is recommended, only for onboard adapters with static cable connection `auto` should be used: https://github.com/webmin/webmin/issues/1074