MichaIng/DietPi #2905: External Seagate USB 3.0 drive not spinning down

tomillr, 2019-06-10:
I've mounted an external 5tb USB 3.0 drive made by Seagate and set the idle spindown timer to 5 minutes in die DietPi Drive Manager. Unfortunately, the drive just keeps on spinning, even though I'm running a minimal install of DietPi with no user activity to the drive.

Manually testing hdparm just gives me an error message. Apparently, there is an issue with the kernel version and how it handles "ATA command pass through". As a workaround you can blacklist the device by running this call:

```
echo "options usb-storage quirks=0bc2:ab30:u" >> /etc/modprobe.d/usb-storage-quirks.conf
update-initramfs -u
reboot
```

Afterwards I can at least have the drive spin down manually by using `hdparm -y /dev/sda`.

Unfortunately, the automatic spindown still doesn't work. Once I access the drive and it spins up, the drive just keeps on running.

Any idea what I should do here?

[https://dietpi.com/phpbb/viewtopic.php?f=11&t=6031](https://dietpi.com/phpbb/viewtopic.php?f=11&t=6031)

PeterLacknase, 2019-06-11:
Please post the output of 

`sudo hdparm -I /dev/sdX`

What is the exact type/ model of the case / enclosing?

MichaIng, 2019-06-11:
@tomillr
Many thanks for your report. 

Besides drive model, which SBC or PC are you using?

tomillr, 2019-06-11:
I'm using a Apollo Lake-based "AP34" mini pc made by Beelink. Please see the requested info below.

```
root@DietPi:~# uname -a
Linux DietPi 4.19.0-5-amd64 #1 SMP Debian 4.19.37-3 (2019-05-15) x86_64 GNU/Linux
```

```
root@DietPi:~# sudo hdparm -I /dev/sda

/dev/sda:

ATA device, with non-removable media
	Model Number:       ST5000LM000-2AN170                      
	Serial Number:      WCJ0H32F
	Firmware Revision:  0001    
	Transport:          Serial, ATA8-AST, SATA 1.0a, SATA II Extensions, SATA Rev 2.5, SATA Rev 2.6, SATA Rev 3.0
Standards:
	Used: unknown (minor revision code 0x006d) 
	Supported: 10 9 8 7 6 5 
	Likely used: 10
Configuration:
	Logical		max	current
	cylinders	16383	16383
	heads		16	16
	sectors/track	63	63
	--
	CHS current addressable sectors:    16514064
	LBA    user addressable sectors:   268435455
	LBA48  user addressable sectors:  9767541168
	Logical  Sector size:                   512 bytes
	Physical Sector size:                  4096 bytes
	Logical Sector-0 offset:                  0 bytes
	device size with M = 1024*1024:     4769307 MBytes
	device size with M = 1000*1000:     5000981 MBytes (5000 GB)
	cache/buffer size  = unknown
	Form Factor: 2.5 inch
	Nominal Media Rotation Rate: 5526
Capabilities:
	LBA, IORDY(can be disabled)
	Queue depth: 32
	Standby timer values: spec'd by Standard, no device specific minimum
	R/W multiple sector transfer: Max = 16	Current = ?
	Advanced power management level: 127
	Recommended acoustic management value: 208, current value: 208
	DMA: mdma0 mdma1 *mdma2 udma0 udma1 udma2 udma3 udma4 udma5 udma6 
	     Cycle time: min=120ns recommended=120ns
	PIO: pio0 pio1 pio2 pio3 pio4 
	     Cycle time: no flow control=120ns  IORDY flow control=120ns
Commands/features:
	Enabled	Supported:
	   *	SMART feature set
	    	Security Mode feature set
	   *	Power Management feature set
	   *	Write cache
	   *	Look-ahead
	   *	Host Protected Area feature set
	   *	WRITE_BUFFER command
	   *	READ_BUFFER command
	   *	DOWNLOAD_MICROCODE
	   *	Advanced Power Management feature set
	    	Power-Up In Standby feature set
	   *	SET_FEATURES required to spinup after power up
	    	SET_MAX security extension
	   *	48-bit Address feature set
	   *	Device Configuration Overlay feature set
	   *	Mandatory FLUSH_CACHE
	   *	FLUSH_CACHE_EXT
	   *	SMART error logging
	   *	SMART self-test
	   *	General Purpose Logging feature set
	   *	WRITE_{DMA|MULTIPLE}_FUA_EXT
	   *	64-bit World wide name
	   *	WRITE_UNCORRECTABLE_EXT command
	   *	{READ,WRITE}_DMA_EXT_GPL commands
	   *	Segmented DOWNLOAD_MICROCODE
	    	unknown 119[6]
	   *	Gen1 signaling speed (1.5Gb/s)
	   *	Gen2 signaling speed (3.0Gb/s)
	   *	Gen3 signaling speed (6.0Gb/s)
	   *	Native Command Queueing (NCQ)
	   *	Phy event counters
	   *	READ_LOG_DMA_EXT equivalent to READ_LOG_EXT
	    	DMA Setup Auto-Activate optimization
	    	Device-initiated interface power management
	    	Software settings preservation
	    	unknown 78[7]
	   *	SMART Command Transport (SCT) feature set
	   *	SCT Write Same (AC2)
	   *	SCT Data Tables (AC5)
	    	unknown 206[7]
	    	unknown 206[12] (vendor specific)
	    	unknown 206[13] (vendor specific)
Security: 
	Master password revision code = 65534
		supported
	not	enabled
	not	locked
	not	frozen
	not	expired: security count
		supported: enhanced erase
	818min for SECURITY ERASE UNIT. 818min for ENHANCED SECURITY ERASE UNIT.
Logical Unit WWN Device Identifier: 5000c500ac249461
	NAA		: 5
	IEEE OUI	: 000c50
	Unique ID	: 0ac249461
Checksum: correct
```

MichaIng, 2019-06-11:
@tomillr 
APM is supported and since after you workaround `hdparm -y /dev/sda` works, generally the automated spin down should work as well 🤔.

Can you please paste:
```
cat /etc/hdparm.conf
```

And compare power states once when drive is active, once after it has been idle for definitely the defined spin-down time and again once after `hdparm -y /dev/sda`. Perhaps on after spin-down time it changes the power state but does not stop spinning completely (different power state).
```
hdparm -C /dev/sda
```

Another thing to try (afterwards) is to edit `/etc/hdparm.conf` => `apm = 1`, which is the most aggressive power saving mode. I am not 100% sure (and this varies as well) about the influence, but generally values 1-127 allow spin-down while 128-255 do not: https://manpages.debian.org/stretch/hdparm/hdparm.8.en.html

tomillr, 2019-06-11:
See below for the hdparm.conf used:

```
root@DietPi:~# hdparm -C /dev/sda

/dev/sda:
 drive state is:  standby
root@DietPi:~# cat /etc/hdparm.conf
/dev/mmcblk1
{
   apm = 127
   spindown_time = 241
}
/dev/sda
{
   apm = 127
   spindown_time = 241
}
```
How about using a tool like TLP to manage spin down and also power down used USB ports?

MichaIng, 2019-06-11:
@tomillr 
> drive state is:  standby

Was this after `hdparm -y /dev/sda` or was the drive still spinning?

> How about using a tool like TLP or powertop to manage spin down and also power down used USB ports?

You can try it, however if it's about drive power management I don't believe that other tools can do what hdparm can't (using apm events).

> Is there a way to manually trigger an HDD to spin up so that I can test the automatic spin down function?

`lsblk` should spin up all drives, at least in my case this worked. Usually I would say, simple write to the disk, but `async` caches small writes 😉.

tomillr, 2019-06-11:
Hm, noticed another strange behavior:

I was able to manually spin up the drive my going into the dietpi-drive_manager. I then went and edited the spin down timer to 300 seconds and exited out of the drive manager.

Calling `hdparm -C /dev/sda` immediately afterwards brings up:

```
root@DietPi:~# hdparm -C /dev/sda

/dev/sda:
 drive state is:  unknown
```

I only get a proper reading back after using `root@DietPi:~# sudo hdparm -I /dev/sda` again.

```
root@DietPi:~# sudo hdparm -I /dev/sda

/dev/sda:

ATA device, with non-removable media
	Model Number:       ST5000LM000-2AN170                      
	Serial Number:      WCJ0H32F
	Firmware Revision:  0001    
	Transport:          Serial, ATA8-AST, SATA 1.0a, SATA II Extensions, SATA Rev 2.5, SATA Rev 2.6, SATA Rev 3.0
Standards:
	Used: unknown (minor revision code 0x006d) 
	Supported: 10 9 8 7 6 5 
	Likely used: 10
Configuration:
	Logical		max	current
	cylinders	16383	16383
	heads		16	16
	sectors/track	63	63
	--
	CHS current addressable sectors:    16514064
	LBA    user addressable sectors:   268435455
	LBA48  user addressable sectors:  9767541168
	Logical  Sector size:                   512 bytes
	Physical Sector size:                  4096 bytes
	Logical Sector-0 offset:                  0 bytes
	device size with M = 1024*1024:     4769307 MBytes
	device size with M = 1000*1000:     5000981 MBytes (5000 GB)
	cache/buffer size  = unknown
	Form Factor: 2.5 inch
	Nominal Media Rotation Rate: 5526
Capabilities:
	LBA, IORDY(can be disabled)
	Queue depth: 32
	Standby timer values: spec'd by Standard, no device specific minimum
	R/W multiple sector transfer: Max = 16	Current = ?
	Advanced power management level: 127
	Recommended acoustic management value: 208, current value: 208
	DMA: mdma0 mdma1 *mdma2 udma0 udma1 udma2 udma3 udma4 udma5 udma6 
	     Cycle time: min=120ns recommended=120ns
	PIO: pio0 pio1 pio2 pio3 pio4 
	     Cycle time: no flow control=120ns  IORDY flow control=120ns
Commands/features:
	Enabled	Supported:
	   *	SMART feature set
	    	Security Mode feature set
	   *	Power Management feature set
	   *	Write cache
	   *	Look-ahead
	   *	Host Protected Area feature set
	   *	WRITE_BUFFER command
	   *	READ_BUFFER command
	   *	DOWNLOAD_MICROCODE
	   *	Advanced Power Management feature set
	    	Power-Up In Standby feature set
	   *	SET_FEATURES required to spinup after power up
	    	SET_MAX security extension
	   *	48-bit Address feature set
	   *	Device Configuration Overlay feature set
	   *	Mandatory FLUSH_CACHE
	   *	FLUSH_CACHE_EXT
	   *	SMART error logging
	   *	SMART self-test
	   *	General Purpose Logging feature set
	   *	WRITE_{DMA|MULTIPLE}_FUA_EXT
	   *	64-bit World wide name
	   *	WRITE_UNCORRECTABLE_EXT command
	   *	{READ,WRITE}_DMA_EXT_GPL commands
	   *	Segmented DOWNLOAD_MICROCODE
	    	unknown 119[6]
	   *	Gen1 signaling speed (1.5Gb/s)
	   *	Gen2 signaling speed (3.0Gb/s)
	   *	Gen3 signaling speed (6.0Gb/s)
	   *	Native Command Queueing (NCQ)
	   *	Phy event counters
	   *	READ_LOG_DMA_EXT equivalent to READ_LOG_EXT
	    	DMA Setup Auto-Activate optimization
	    	Device-initiated interface power management
	    	Software settings preservation
	    	unknown 78[7]
	   *	SMART Command Transport (SCT) feature set
	   *	SCT Write Same (AC2)
	   *	SCT Data Tables (AC5)
	    	unknown 206[7]
	    	unknown 206[12] (vendor specific)
	    	unknown 206[13] (vendor specific)
Security: 
	Master password revision code = 65534
		supported
	not	enabled
	not	locked
	not	frozen
	not	expired: security count
		supported: enhanced erase
	818min for SECURITY ERASE UNIT. 818min for ENHANCED SECURITY ERASE UNIT.
Logical Unit WWN Device Identifier: 5000c500ac249461
	NAA		: 5
	IEEE OUI	: 000c50
	Unique ID	: 0ac249461
Checksum: correct
root@DietPi:~# hdparm -C /dev/sda

/dev/sda:
 drive state is:  active/idle
root@DietPi:~# 

```

After that the drive automatically spins down after the set value of 300 seconds.

```
root@DietPi:~# hdparm -C /dev/sda

/dev/sda:
 drive state is:  standby
```

Any idea why that happens only after requesting the drive information?

MichaIng, 2019-06-11:
@tomillr 
Hmm, what `dietpi-drive_manager` does when choosing the the spin-down time is:
```
hdparm -B 127 /dev/sda # Set apm to 127
hdparm -S 60 /dev/sda # Set spin-down time to 60*5=300s
```
https://manpages.debian.org/stretch/hdparm/hdparm.8.en.html#OPTIONS

Now I see that `-S` not only changes the spin-down time but also puts the drive into "idle (low-power) mode". Not sure if this second call is required. Setting the value in `/etc/hdparm.conf` might already work for the current session, not sure.
Also not sure if there is a difference between "standby" and "idle" mode, although does not make much sense if there was one.
But you can play with these two commands to check which one leads to "unknown" `hdparm -C /dev/sda` output.
If indeed the timeout set by `-S` has an effect, but in your case only after `-I` was called once, I can add this to drive manager code to be called silently as well. At least does not hurt.

diveyez, 2019-06-12:
This is a common problem with Seagate drives. Not related to dietpi usage.  I have one that does the same thing, unless it is ran with the bundled software. Which happens to be windows and mac software only.

PeterLacknase, 2019-06-12:
In some cases disabling SMART scrubbing helps:

`smartctl -c /dev/sd?`

If it shows „Auto Offline Data Collection“ as enabled, spin down is not possible while running SMART tests. To disable those tests (permanently):

`smartctl --offlineauto=off /dev/sd?`

MichaIng, 2019-06-12:
@diveyez 
Thanks for your hint. Yeah Seagate drives seem to have their own internal power saving mechanism, e.g. I found this in hdparm man page:
```
-Z
    Disable the automatic power-saving function of certain Seagate drives (ST3xxx models?), to prevent them from idling/spinning-down at inconvenient times.
```
- But in @tomillr case even without `-Z` the defined idle time finally works. So not sure if this applies in that case.

@PeterLacknase 
Good hint as well, however at least by default we do not install smartutils on DietPi, so this daemon is not active and cannot interfere.

So @tomillr if hdparm -I restores a working drive state in you case and allows spin-down in the defined time, I will add this call to drive manager after chaning the value.

MichaIng, 2019-06-14:
Ref: https://dietpi.com/phpbb/viewtopic.php?f=11&t=6031

MichaIng, 2019-06-24:
Forgot about `echo "options usb-storage quirks=0bc2:ab30:u" >> /etc/modprobe.d/usb-storage-quirks.conf`.
Understood now that it disables UAS. E.g. this is known to be now (with their last large kernel upgrade to 4.19) available on RPi but not yet fully implemented.

@tomillr 
Which device/SBC do you actually use and with which kernel version?
```
echo $G_HW_MODEL_DESCRIPTION
uname -a
```

tomillr, 2019-06-24:
> I'm using a Apollo Lake-based "AP34" mini pc made by Beelink. Please see the requested info below.
> 
> ```
> root@DietPi:~# uname -a
> Linux DietPi 4.19.0-5-amd64 #1 SMP Debian 4.19.37-3 (2019-05-15) x86_64 GNU/Linux
> ```

MichaIng, 2019-06-24:
@tomillr 
Okay this is drive specific then indeed. I am thinking if we should leave it as is, or detect this drive and add the kernel quirk automatically. But this starts go getting complicated then.

I am not too experienced with this, does `lsusb` show `0bc2:ab30` and does this match that drive, regardless on which USB port it is attached?

MichaIng, 2019-06-25:
Okay, device identifier is the same on all systems+ports, all x86_64 Debian versions use the usb-storage as kernel module and since the newest v4.19 requires the fix, I will add this now by default to all x86_64 systems on DietPi-PREP + DietPi-Update to v6.25.

closed 2019-07-03 by MichaIng