MichaIng/DietPi #2906: Feature

diveyez, 2019-06-12:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- ...

#### Describe the solution you'd like:

- ...

#### Describe alternatives you've considered:

- ...

#### Additional context

- ...

#### Vote for this feature on FeatHub: https://feathub.com/MichaIng/DietPi/

MichaIng, 2019-06-12:
@diveyez 
Many thanks for your request.

There is [autofs](https://manpages.debian.org/stretch/autofs/autofs.5.en.html) which automatically mounts attached drives. But it does not create the fstab entries, AFAIK.

The issue is that either we need to recognize kernel events (which is not easily possible with shell scripts, AFAIK) or we need to daemon that constantly checks for attached drives, which would lead to a tee high constant CPU usage. So it is 3rd party tools like the above that we need to go with. But perhaps we can implement them into the drive manager, e.g. enable/disable automount and such.

diveyez, 2019-06-15:
``
blkid | grep /dev/sd | sed -e 's/:.* UUID/;UUID/' -e 's/ TYPE.*//' | while read re; do sed -i "s;$re;" /etc/fstab; done
``

diveyez, 2019-06-15:
Updated to explain +1

MichaIng, 2019-06-15:
Actually `dietpi-drive_manager` already does what we want. But we cannot run an additional process in the background that is constantly scanning for new drives, that would be a complete overhead and overkill.

Actually udev rules should do it:
```
echo 'ACTION=="add", KERNEL=="sd[a-z]", RUN+="/DietPi/dietpi/dietpi-drive_manager 3"' > /etc/udev/rules.d/99-dietpi-auto_mount.rules
udevadm control --reload-rules
```

But I need to add the argument `3` that only does the drive scan + fstab creation loop and then exits.

diveyez, 2019-06-15:
Can you @diveyez notify when this is pushed to stable/master? I really need the lady to be able to do this as, she cannot run a CLI for *NIX or *Win lol

MichaIng, 2019-06-15:
@diveyez 
Jep will do.

Did a start: https://github.com/MichaIng/DietPi/commit/73a555639cff1ff418cbdc942b60a20ed9dff512
However currently unmounted drives are added in commented way to fstab. We need to adjust the input argument so that unmounted drives are added un-commented and mounted automatically.

This would actually render `autofs`/`automount` obsolete on DietPi 😃. I hope the udev method works reliable.

diveyez, 2019-06-15:
The commenting is for a reason ;)
its 

> **"beta"**

 in formation lol

hifitime, 2020-01-13:
echo 'ACTION=="add", KERNEL=="sd[a-z]", RUN+="/DietPi/dietpi/dietpi-drive_manager 3"' > /etc/udev/rules.d/99-dietpi-auto_mount.rules
udevadm control --reload-rules

That's what I really need, but it doesn't work

MichaIng, 2020-01-13:
@hifitime 
This feature (argument 3) has not yet been added to drive manager :wink:. It cannot work yet.

street-grease-coder, 2021-01-16:
bump. especially reconnecting usb drives that have been (e.g. accidentally) plugged out or not plugged in at boot are very annoying to setup for noobs (i.e. udev rules)

MichaIng, 2021-01-16:
`dietpi-drive_manager 3` now works to re-scan drives and rewrite fstab. What I didn't think about is that it does not mount the drive by default. The following could work:
```
echo 'SUBSYSTEM=="block", KERNEL="sd[a-z][1-9]", ACTION=="add", RUN+="mount /dev/%k || { mkdir -p /mnt/%k && mount /dev/%k /mnt/%k && /boot/dietpi/dietpi-drive_manager 3 && mount -o remount /mnt/%k; }"' > /etc/udev/rules.d/99-dietpi-auto_mount.rules
udevadm control --reload-rules
```
I adjusted the rule a bit so that it matches only partitions, no raw drives and only block devices, just to be sure. It mounts the drive, if it is present already in fstab (mount command without mount point succeeds), else it creates a new mount point based on the device name (I couldn't find a way to get UUID from within udev rules easily), mounts the drive with default options to that mount point and then runs `dietpi-drive_manager` to have the fstab entry created with mount options based on drive details, then remounts the drive to apply mount options 😅. Requires some testing, especially how it behaves at boot where we don't want that overhead.

`dietpi-drive_manager` creates a commented fstab entry for non-mounted drives. So another possibility would be to run `/boot/dietpi/dietpi-drive_manager 3` first, then uncomment commented fstab entries (a `sed` call) and then mount the drive.

street-grease-coder, 2021-01-16:
> `dietpi-drive_manager 3` now works to re-scan drives and rewrite fstab. What I didn't think about is that it does not mount the drive by default. The following could work:
>
> ```
> echo [...]

wow, thanks a lot. Somehow this doesn't work for me (ntfs drive...). I don't get any errors, it just doesn't mount it. U/Mounting it manually works

MichaIng, 2021-01-17:
Thanks for sharing. Yes a real script is probably better since I am not even sure it that shell syntax I added to the udev rule is even correctly interpreted (curly brackets etc). Wrapping it into a systemd unit has the benefit that the output is not lost but can be reviewed via `journalctl -u usb-mount`. udev otherwise does not log anywhere, AFAIK. But the same could be archived from within the script itselt.

What I wonder is why everyone uses `SUBSYSTEM=="usb"` while in my case checking all USB-attached drives clearly state `SUBSYSTEM=block`: `udevadm info /dev/sda | grep SUBSYSTEM`

But yeah, all a matter of testing, at least we have collected some ideas.

street-grease-coder, 2021-01-17:
@diveyez @hifitime in case time allows for it