MichaIng/DietPi #2911: xserver - all desktops fail

diveyez, 2019-06-14:
5v wall plug, Rpi3b+
Fresh install, buster 
https://imgur.com/gallery/EYFpc8p

diveyez, 2019-06-14:
Solution was to do ‘’sudo apt-get install task-xfce-desktop’’

MichaIng, 2019-06-14:
Not the first time a bug like this occurred in LightDM from Debian repo: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=749243

Most bad about this is that according to FHS all dirs+files inside /var should be able to be recreated by programs (besides their own parent dir, e.g. /var/lib/lightdm of course).

However easy to fix:
```
mkdir -p /var/lib/lightdm/data
chown lightdm:lightdm /var/lib/lightdm/data
```
____________
`task-xfce-desktop` installs the whole default Debian desktop with all the overhead that we do not want to ship by default with DietPi desktop selections. However glad it worked in your case.

MichaIng, 2019-06-14:
@diveyez 
Hmm unable to replicate the issue on Debian Buster. The directory was auto-created in my case when LightDM is called the first time.

You
1. Installed XFCE via `dietpi-software`
2. Enabled LightDM auto login via `dietpi-autostart`
3. `reboot`
4. Error you posted

Right?

diveyez, 2019-06-14:
Bo,I do not use autologins for security reasons, it happens with LightDM greeter

diveyez, 2019-06-14:
Your solution still did not work, LightDM greeter shows 100% cpu usage and frozen to a black screen.

diveyez, 2019-06-14:
It appears a package was moved that the desktop needs which means you will need to include it, or switch apt line to the default  install line.

diveyez, 2019-06-14:
@MichaIng during my rework of the problem I noticed a problem, ‘command reboot not found’ when in a root shell,
Is it possible that root/sudo is not registering?

diveyez, 2019-06-14:
@MichaIng any graphical software for the buster image causes me to hang at a black screen on boot only escape switching tty and manually installing software by hand without using the dietpi installer menu.

MichaIng, 2019-06-15:
@diveyez 
Can you list which other software you tried and how you installed it manually afterwards? Will compare with our install steps. Indeed most likely some X.org package is missing then for Raspbian Buster.

I think I have some Raspbian Buster SDcard to run some tests as well.

diveyez, 2019-06-15:
Just xfce4 and lxde were the softwares trying to install, I also did the headless chromium kiosk and that also failed.

MichaIng, 2019-06-24:
@diveyez 
And how did you install XFCE or LXDE by hand to make it run?

MichaIng, 2020-01-01:
I mark this as closed. Did many tests or desktops/xserver/graphical software on RPi meanwhile, hence there is not a general issue (anymore). However feel free to reopen if required.

closed 2020-01-01 by MichaIng

diveyez, 2020-01-06:
Hey there! Sorry I have been really busy.

I need to speak to you in private.

Something has come up regarding needing a baby image for an extended development cycle.

MichaIng, 2020-01-08:
@diveyez 
In case email me, the mail address given in my profile should be public.