MichaIng/DietPi #2913: Improvement & refactoring :  Move LED settings and CPU governor adjustment into dedicated systemd units.

FredericGuilbault, 2019-06-15:
From https://github.com/MichaIng/DietPi/pull/2833#issuecomment-494604437

> Largest impact is of course the two child scripts (CPU governor + LED settings) that are executed in subshell and load dietpi-globals in case. Actually it should be at least possible to skip led settings, an additional setting to revert to system defaults, which removes the settings file and then scripts the execution. Actually we could move this as well as the CPU governor adjustment into an own systemd unit each. It does not really matter when they run, just after DietPi-RAMdisk. So they can be Type=simple to not delay any other boot scripts. This also makes it again simpler to revert to system defaults => disable the related systemd unit. Once applied settings (settings file) can be preserved then while toggling the service.

FredericGuilbault, 2019-06-15:
After this it will be possible to remove `DietPi-PreBoot.service` and it's script `/DietPi/dietpi/preboot`

MichaIng, 2019-06-15:
@FredericGuilbault 
Jep thanks, good idea. I am not yet 100% sure about it, have 1-2 other things that might fit into the preboot script. But at least CPU and LED script should only be called, if settings have been actually chosen and it should be possible to revert settings to system defaults.