MichaIng/DietPi #2916: Intel Quick Sync not working on x64 build

tomillr, 2019-06-16:
The Plex Media Server allows for hardware transcoding by using the "Quick Sync"-technologie build into modern Intel chips.

Unfortunately, I've yet to get this working when using DietPi x64 on my Intel Apollo Lake-based mini PC.

The Plex is capable of using Quick Sync on other Linux systems, as I've successfully tested hardware transcoding in Ubuntu, Debian as well as OpenMediaVault on the very same device.

Could it be that there is some kind of driver missing in DietPi or do I have to manually enable support for Quick Sync somewhere in the settings

MichaIng, 2019-06-16:
@tomillr
Many thanks for your report.

AFAIK FFmpeg transcoding is required for this.

Try: `G_AGI ffmpeg`

Possibly some transcoder link needs to be corrected as well, I will have a look.

tomillr, 2019-06-17:
Hardware transcoding was already enabled in the PMS settings.

Unfortunately, neither adding a Plex user or installing ffmpeg worked. 

Could it be, that getting Intel Quick Sync to work requires some additional GPU driver to be installed on the system?

MichaIng, 2019-06-17:
@tomillr 
Possible, although I am wondering if this would be pre-installed on other OS then. Which Intel CPU/GPU generation is it?

Intel GEN8+ drivers: https://packages.debian.org/buster/intel-media-va-driver
- Available on Debian Buster only

Debug tools, which might help and pulls some libs as depedencies as well: https://packages.debian.org/stretch/intel-gpu-tools

tomillr, 2019-06-17:
> @tomillr
> Intel GEN8+ drivers: https://packages.debian.org/buster/intel-media-va-driver
> The nonfree version, not sure if/what the difference is: https://packages.debian.org/buster/intel-media-va-driver-non-free

intel-media-va-driver-non-free at least mentions "Apollo Lake" as a supported system.

Can I download and install the driver right over SSH?

MichaIng, 2019-06-17:
@tomillr 
This is the official Debian repo, so: `G_AGI intel-media-va-driver-non-free`

tomillr, 2019-06-17:
Thanks, I've installed the driver alongside the Intel GPU Tools but Plex still only uses the CPU for transcoding.

MichaIng, 2019-06-17:
@tomillr 
Ah least a hint. Does this file or something similar exist? `ls -Al /dev/dri/`
But honestly I have no idea how to map the iGPU to a certain dev file... Needs investigation...

You have no dedicated GPU installed, right?

tomillr, 2019-06-18:
That's right, the mini pc doesn't have a dedicated GPU but rather uses an integrated Intel GPU. The folks over at Plex also mentioned that "udev" has to be installed. Otherwise the Plex Media Server can't access the hardware. Not sure that's going to help us through.

tomillr, 2019-06-19:
I managed to get hardware transcoding working by using `usermod -a -G render plex` and restarting the system.

MichaIng, 2019-06-20:
@tomillr 
> usermod -a -G render plex

Very interesting, this user does not even exist on my machines. Lol first match on duckduckgo: https://bbs.archlinux.de/viewtopic.php?id=31345
1. What is this?
2. Found that the group was changing the name from "video" to "render". But "video" still exists!
3. "render" is such a systemd story...
4. Looks like it has something to do with systemd.

Yeah okay I just checked my RPi2 (Buster) and indeed: `render:x:114:`
- Also new: `kvm:x:113:`: https://help.ubuntu.com/community/KVM/Installation
  - A group which allows to run KVMs (Kernel-based Virtual Machines).

MichaIng, 2019-06-20:
@tomillr 
> intel-media-va-driver-non-free

Ah: `apt purge intel-media-va-driver-non-free`

MichaIng, 2019-06-24:
@tomillr 
You use a Buster image, right? I got confused above, I recognised that `intel-media-va-driver-non-free` is available on Buster only and then obviously forgot when writing back the install command.

So looks like we should add all AV software users to the `render` group on Buster to solve hardware acceleration on all of them together.

tomillr, 2019-06-24:
> @tomillr
> You use a Buster image, right?
That's right. Buster rc1 amd64.

MichaIng, 2019-06-24:
@tomillr 
Okay I will add media users to this group on Buster then by default.

closed 2019-07-03 by MichaIng