MichaIng/DietPi #2919: Dietpi preboot service as a dependency for succecent service might be depricated.

FredericGuilbault, 2019-06-17:
In the branch dev, Since we move moved most out of DietPi-PreBoot. (only CPU & LED management left). I tried to dissable DietPi-PreBoot service. But I got those errors : 
```

[DEPEND] Dependency failed for DietPi-Boot.
[DEPEND] Dependency failed for DietPi-PostBoot.
[DEPEND] Dependency failed for DietPi-FirstBoot setup.
```

In the current state of PreBoot. is it still a dependency for others services ?

MichaIng, 2019-06-17:
@FredericGuilbault 
Besides the CPU and LED scripts, `dietpi-obtain_hw_model` is also something we need to run at another place first to be able to remove preboot.
Although, after first boot, this is actually required for RPi only where this allows to switch the SDcard between different RPi models. On other machines this is either not possible or the model differences have no effect on DietPi handling.

I am not 100% sure now if it is best to create another new systemd units for this, or to merge at another place. `dietpi-boot` does not work, since `dietpi-firstboot`, which runs first, requires at least `dietpi-obtain_hw_model` to be called.

FredericGuilbault, 2019-06-18:
Oh I didin't know `dietpi-obtain_hw_model` where making work. This justify the dependency.

closed 2019-06-18 by FredericGuilbault