MichaIng/DietPi #2924: Rpi3b onboard Broadcom Wlan0 not available

diveyez, 2019-06-18:
DietPi Stable Stretch
Rpi3b, Clean install. 
wlan0 is not available, installing net-tools reveals no devices in ifconfig, as well as other various tools.

Desktop configuration failed as no adapter is visible.

MichaIng, 2019-06-19:
@diveyez 
Many thanks for your report.

Forget about `net-tools`, it's outdated and obsolete on modern systems.
Please check: `ip r`
- This should list all available adapters, regardless if being configured or not.

> pre-req apt packages were not found, installer failed, had to manually install the packages.

Can you paste the exact messages about this error? Actually all WiFi packages should be installed by default on our images, and if manually removed, of course they should be installed automatically.

> especially since it also failed to allow me to change keyboard language/region.

As of the other issue, it is an external issue on Buster only, which requires fix on Debian side. We will only invest effort for a workaround if Debian does not fix it until official Buster release.

MichaIng, 2019-07-26:
@diveyez 
Is missing onboard WiFi issue still apparent?