MichaIng/DietPi #2925: OpenMediaVault

diveyez, 2019-06-18:
I have gotten openmediavault to play nice with dietpi, however it is buggy. Question is, if I can modify and optimize it, for dietpi, will we get a reinstatement of omv?

MichaIng, 2019-06-19:
@diveyez 
Thanks for your suggestion.

This issue is that both, OMV and DietPi, have similar aims but with different implementations/solutions. It is a bid similar to some issues with Webmin, where e.g. network setup of Webmin web UI is not fully compatible with DietPi, although in that case it is easy to fix. But OMV is much more intrusive, more like an own OS than only a web UI.

We invested much effort, time and nerves to get it working on DietPi but finally gave as the effort (not only an initial implementation but especially constant maintenance due to changes in OMV) was just not worth it.

I suggest, if you could get it working, to contribute fixes/changes that e.g. make OMV more flexible/less intrusive to the underlying OS, directly to OMV. I fork IMO is not a good solution, since it becomes either outdated (compared to the original) or the effort of merging upstream changes is undoable.

We can then have another look.

diveyez, 2019-06-21:
Ok, an "OMV mini"
fork would be sick, and a dish of  OPENSOURCE to them

MichaIng, 2019-06-21:
@diveyez 
It's a bid comparable to the NextCloudPi case: https://github.com/MichaIng/DietPi/issues/1778
- A wonderful peace of software for user who want to quickly have a ready-to-run and well configured Nextcloud-only machine, with nice web interface etc etc.
- But it turns the machine into a single-purpose device, being simply to intrusive for DietPi to implement as software option. Would require huge rework and custom adjustments to have it run nicely besides our other software options. And since we have an own ready-to-run Nextcloud install option...

There are other cases where great FOSS software exists that is totally great on itself but collides with the aims and uses of DietPi. Of course one can always install those manually, it will in most cases run just fine on DietPi, like on any other Debian, but it makes sense that we only implement software titles into DietPi-Software that can run well concurrently besides all (lets say most) other software titles we offer, and as well does not too badly collide with the way `dietpi-config` does things.
- Lol just remember the Webmin issues, where exactly this is the case with their and our network setup: https://github.com/MichaIng/DietPi/issues/2900, https://github.com/MichaIng/DietPi/issues/2899 However we are already working on fixing those cases, which are not too hard, make sense in general and Webmin devs are quite open and fast in reply 😃.

I am also no fans of forks. I mean sometimes the ideas of different developers/contributors simply go too far apart, or bad atmosphere between them starts to drain productivity and motivation, then there is no other chance. But generally I am a friend of working together on the original project instead of dividing forces and making the life for users harder as well, who finally stay in front of a large number of forks to choose from. Think of SickBeard => SickRage => SickChill => SiCKRAGE and not to forget the older fork that we offer: Medusa... what a mess... SickChill/SiCKRAGE was a trademark quarrel in the end...

diveyez, 2019-06-21:
I have dietpi with a desktop running omv and stretch now.
So, its possible, a few things break, but nothing that cannot be worked around.

However, I would rather not, have the OMV EXTREMELY long boot up... That would flood the issues on this github. No thanks.

closed 2019-06-21 by diveyez

MichaIng, 2021-09-22:
Btw, here is a nice current video about how to setup OMV on DietPi: https://www.youtube.com/watch?v=yCVpzetpM4k
German language, but I think even without audio the video is helpful already. Important to note that `dietpi-drive_manager` and the `dietpi-config` network setup shouldn't be used anymore after having this done via OMV interface, so it's an either on in this regards, same with the file servers that one can setup via OMV. But otherwise it seems to work very well.

Joulinar, 2021-09-22:
Nice video, but he recommend not to install anything in addition to OMV. Well, in this case why not using plain OMV img? :wink:

MichaIng, 2021-09-22:
The major reason it that they provide only x86_64 images as far as I can see. It is an installer image which flashes a Debian Buster to the target drive and is definitely heaver compared to DietPi, also after installing the `openmediavault` package + dependencies from their repository (which is basically what the installer does): https://packages.openmediavault.org/public/dists/shaitan/main/binary-amd64/Packages

I just see it pulls in as well cpufrequtils, rsyslog and logrotate: The first is not required on DietPi as DietPi-PreBoot applies the CPU governor already and the other two at least don't make sense in combination with RAM log at least:
```sh
systemctl disable --now cpufrequtils rsyslog logrotate
```
Or uninstalling DietPi-RAMlog, switching to "full logging" mode in DietPi-Software. Probably OMV actively uses the plain text syslog files (instead of reading from journal).

It sets up `systemd-networkd` and `systemd-resolved` as network stack, so once it was used to configure network interfaces, to reduce some obsolete overhead at boot, the ifupdown networking service should be disabled as well as the config file moved or removed:
```sh
systemctl disable --now networking
mv /etc/network/interfaces{,_bak}
```
Probably `ifupdown` can even be removed (`iproute2` should be kept!):
```sh
apt-mark auto iproute2
apt purge --autoremove ifupdown
```

However, it's good to know that it works OOTB even that there is room to tweaks and conflicts with our tools (same as with Webmin e.g.).