MichaIng/DietPi #2927: DietPi update from 6v.22.3 to v6.24.1 failed

Ksylian, 2019-06-19:
#### Details:
- Date           | Wed 19 Jun 11:23:52 BST 2019
- Bug report     | N/A
- DietPi version | v6.22.3 (Fourdee/master)
- Img creator    | n/a
- Pre-image      | n/a
- SBC device     | RPi 3 Model B (armv7l) (index=3)
- Kernel version | #1200 SMP Tue Feb 12 20:27:48 GMT 2019
- Distro         | stretch (index=4)
- Command        | G_AGUP
- Exit code      | 100
- Software title | DietPi-Update

#### Steps to reproduce:

1. Run dietpi-update
2. ...

#### Expected behaviour:

- Dietpi should update to v6.24.1

#### Actual behaviour:

- Update fails

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
Hit:1 http://download.mono-project.com/repo/debian raspbianstretch InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Ign:3 http://archive.raspbian.org/raspbian wheezy InRelease
Ign:4 http://download.opensuse.org/repositories/home:/emby/xUbuntu_14.04  InRelease
Err:5 http://archive.raspbian.org/raspbian wheezy Release
  404  Not Found
Hit:6 http://download.opensuse.org/repositories/home:/emby/xUbuntu_14.04  Release
Hit:7 https://archive.raspberrypi.org/debian stretch InRelease
Reading package lists...
E: The repository 'http://archive.raspbian.org/raspbian wheezy Release' does no longer have a Release file.
```

--------------------------------------------------------------------

[FAILED] DietPi-Update | Unable to continue, DietPi-Update will now terminate.

MichaIng, 2019-06-19:
@Ksylian 
You have multiple Linux distribution repositories and some outdated branches in your APT sources, wherever they came from.
1. Check those files: `/etc/apt/sources.list` + all files in `/etc/apt/sources.list.d/`
2. Remove lines containing: `wheezy` and `opensuse`.

Ksylian, 2019-06-19:
Well spotted, I had one line containing `wheezy ` which I removed and then managed to update the OS. 
Thank you!

MichaIng, 2019-06-20:
@Ksylian 
Perfect, I mark this as closed.

Btw I see the opensuse repo was for Emby. We used this in the past, but switched to more current official deb packages now. But no issue with keeping it, actually even good, in case some other dependencies has been pulled from there, to keep it it available for updates and such.

closed 2019-06-20 by MichaIng