MichaIng/DietPi #2932: unable to resolve host every time i run sudo

Sopor, 2019-06-22:
When i run something as sudo i get this message every time:
```
xxxxx@hostname:~$ sudo cpu
sudo: unable to resolve host hostname
[  OK  ] DietPi-CPU_info | Root access verified.
```
I have replaced the real hostname and user name in case you wondering :)

DietPi v6.24.1
RPi 3 Model B+ (armv7l)

MichaIng, 2019-06-22:
@Sopor 
Please assure that entries in /etc/hostname and /etc/hosts match, and as well not all kind of characters are allowed.

E.g. on my Buster test machine:
```
root@VM-Buster:~# cat /etc/hostname
VM-Buster
root@VM-Buster:~# cat /etc/hosts
127.0.0.1 localhost
192.168.178.30 VM-Buster
```

Sopor, 2019-06-22:
```
xxxxx@hostname:~$ cat /etc/hostname
hostname
xxxxx@hostname:~$ cat /etc/hosts
127.0.0.1       localhost dietpi
```
It seems to be hosts that still having the old name `dietpi`

Sopor, 2019-06-22:
And now it is working:
```
$ sudo cpu
[  OK  ] DietPi-CPU_info | Root access verified.
```

closed 2019-06-22 by Sopor

MichaIng, 2019-06-22:
@Sopor 
Note that on Debian there is the `127.0.1.1` definition by default (can be the local static IP of the machine as well), which should be associated with the hostname.
`127.0.0.1` should only have `localhost`.

I do not 100% understand the practical difference, but what I understood is that e.g. `127.0.0.1 localhost dietpi` makes "dietpi" just an alias for "localhost" instead of being an own hostname entry. So Debian chose to add `127.0.1.1` (which is as well a loopback to the same machine, like 127.0.0.1) to their default /etc/hosts some years ago and expect to add the hostname (that matches /etc/hostname) there, so it is a dedicated hostname entry an not just an alias.

However for this reason to be failsafe, /etc/hosts should look like this:
```
127.0.0.1 localhost
127.0.1.1 <hostname>
```
with <hostname> matching /etc/hostname.

Here the related bug report when this was introduced: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=316099
- Ah looks like `127.0.0.1 localhost <hostname>` makes the command `hostname` print "localhost" instead of expected "<hostname>".