MichaIng/DietPi #2935: Image | Raspberry Pi 4

MichaIng, 2019-06-24:
Device is ordered, initial hardware detection and ID has been added: https://github.com/MichaIng/DietPi/commit/19ab1667e87b760d48414887191ed0b407f13574

MichaIng, 2019-06-24:
Woooooh, totally overseen: Its BUSTER only!!

Just saw it here: https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=243387
And didn't see on the download page: https://www.raspberrypi.org/downloads/raspbian/

Okay due to needs for RPi 4 it seems that Raspbian is faster this time to release Buster than Debian. This changes much, we can hope that some left Buster issues with Raspbian have been solved then and as well the firmware/RPi repo is now fully maintained.

MattL0, 2019-06-24:
see thi on diet pi buster upgrade :

MichaIng, 2019-06-24:
@MattL0 
See in first post the 2nd code box contains all files that are present on clean /boot. These are most probably more then you see in your case since the new files for RPi4 support are already inside. But any files aside from them are most probably junk, so you can move them out of the place, e.g.
`mv /boot/junk /mnt/dietpi_userdata/junk_bak`

MattL0, 2019-06-24:
on a upgrade the boot partition is like 50m , an by looking on your first post i can't remove the large files... so i guess i should reinstall dietpi from a fresh buster image

MichaIng, 2019-06-25:
@MattL0 
> can't i remove one of these files ?

`LICENSE.oracle` can be removed, but yeah indeed non of the large files can be removed, respectively some could, but those are recreated on every firmware update.

So you did already upgrade. Any errors? `df` should also say you if 100% is used or not.

As can be seen above, the new firmware fits closely onto the partition. Neither DietPi, not the firmware files, nor the partition should have any different sizes on all RPis, the "dietpi" dir only varies between some kbits. So it should definitely fit.

> so i guess i should reinstall dietpi from a fresh buster image.

Our RPi images are not yet based on the new Raspbian Lite Buster with new partition size. This was just released the last days. We will ship new images soon, at lastest after the RPi 4 arrived to run some tests.
So reflashing will not solve the close to full /boot partition.

MattL0, 2019-06-25:
ha ok thanks a lot . I'll wait :P 

here is the output of df ( i removed some .dtb files not related to  my pi model... but as you aid I don't know if there will come back ) :

/dev/mmcblk0p1     42131   41064      1067  98% /boot

MichaIng, 2019-06-25:
@MattL0 
Hmm interesting that it is even closer in your case. Could you paste `ls -Alh /boot` to compare?

Which sound card do you use (if any)? There are some where custom dtoverlays are required that are added to `/boot/overlays`.

MattL0, 2019-06-25:
yes 

total 39M
-rwxr-xr-x 1 root root  27K Jun 24 19:39  bcm2710-rpi-3-b-plus.dtb
-rwxr-xr-x 1 root root  25K Jun 24 19:39  bcm2710-rpi-cm3.dtb
-rwxr-xr-x 1 root root  52K Jun 24 19:41  bootcode.bin
-rwxr-xr-x 1 root root  142 Mar  3 15:09  cmdline.txt
-rwxr-xr-x 1 root root 3.1K Jun 24 19:53  config.txt
-rwxr-xr-x 1 root root  19K Jun 24 19:40  COPYING.linux
drwxr-xr-x 4 root root 3.0K Mar 30 03:24  dietpi
-rwxr-xr-x 1 root root  12K Jun 24 19:53  dietpi.txt
-rwxr-xr-x 1 root root 3.0K Jun 24 19:41  fixup4cd.dat
-rwxr-xr-x 1 root root 6.0K Jun 24 19:41  fixup4.dat
-rwxr-xr-x 1 root root 9.0K Jun 24 19:41  fixup4db.dat
-rwxr-xr-x 1 root root 9.0K Jun 24 19:41  fixup4x.dat
-rwxr-xr-x 1 root root 2.6K Jun 24 19:41  fixup_cd.dat
-rwxr-xr-x 1 root root 6.6K Jun 24 19:41  fixup.dat
-rwxr-xr-x 1 root root 9.6K Jun 24 19:41  fixup_db.dat
-rwxr-xr-x 1 root root 9.6K Jun 24 19:41  fixup_x.dat
-rwxr-xr-x 1 root root 5.1M Jun 24 19:40  kernel7.img
-rwxr-xr-x 1 root root 5.4M Jun 24 19:40  kernel7l.img
-rwxr-xr-x 1 root root 4.8M Jun 24 19:40  kernel.img
drwxr-xr-x 2 root root  15K Jun 24 19:40  overlays
-rwxr-xr-x 1 root root 745K Jun 24 19:40  start4cd.elf
-rwxr-xr-x 1 root root 4.5M Jun 24 19:41  start4db.elf
-rwxr-xr-x 1 root root 2.7M Jun 24 19:40  start4.elf
-rwxr-xr-x 1 root root 3.6M Jun 24 19:41  start4x.elf
-rwxr-xr-x 1 root root 670K Jun 24 19:41  start_cd.elf
-rwxr-xr-x 1 root root 4.7M Jun 24 19:41  start_db.elf
-rwxr-xr-x 1 root root 2.8M Jun 24 19:40  start.elf
-rwxr-xr-x 1 root root 3.7M Jun 24 19:41  start_x.elf
drwxr-xr-x 2 root root  512 Mar 20  2018 'System Volume Information'

MattL0, 2019-06-25:
I have no ext sound card, I use theses 3 PI as a squeezelite player + shairport.  Maybe this is the difference

MichaIng, 2019-06-25:
@MattL0 
Okay nope in this case no additional overlays were installed then.

Another one that I see: `rm -R '/boot/System Volume Information'`
Also nearly no size, but since Mac has their own files again, Windows might additionally place thumbs.db and desktop.ini and if you ever somehow plug the SDcard to a camera again some dirs/files could be created, this might some up to hit 100%.

MattL0, 2019-06-25:
I just saw that i got this when i load :  dietpi-software
[image]

MichaIng, 2019-06-25:
@MattL0 
Could you open a new issue about this? We are already going to far off-topic 😉.

Fourdee, 2019-06-25:
@MichaIng 

All that work you did on Buster will pay off here, nice one :+1: 

I'd also recommend we switch to Buster for RPi image by default. As Raspbian official has removed Stretch links: https://www.raspberrypi.org/downloads/raspbian/

When my device arrives (after 5th July) I will chip in with some testing + implementations where needed.

MichaIng, 2019-06-25:
@Fourdee 
> I'd also recommend we switch to Buster for RPi image by default.

Jep, to keep it easy we even have to. Looks like RPi 4 simply works (or is at least officially only supported) on Buster. Otherwise we would need to provide two separate images (RPi4 and non-RPi4) and as of the above issue with close-to-full /boot, extend the boot partition manually. Totally not keen to do so.

There has been just another firmware update just some hours after the first. I think I will wait for another day or two, to assure things are settled, and then redo our RPi image based on the new Raspbian Lite Buster (+ move Stretch image to legacy). So RPi4 support is already there and the issue with small boot partition solved.
Will also create this based on beta code, do allow me test new dietpi-firstboot on a non-VM system.

Fourdee, 2019-06-27:
@MichaIng 

Initial RPi buster image done:
https://dietpi.com/downloads/images/DietPi_RPi-ARMv6-Buster.7z

Quick testing my end with general server use + WiFi is fine. Things that need testing are:
- GPU/desktop related items.

Notes to self:
- Update dietpi.com info for RPi 4

MichaIng, 2019-06-27:
@Fourdee 
Btw Kodi APT package can be added by /etc/apt/sources.list.d/raspi.list, adding the stretch branch besides buster branch. We do this by default when changing APT mirror on RPi btw.

But the question is if this Kodi 18 package still works.

lordkane, 2019-06-27:
what is the login ?
dietpi root not work
root dietpi not work

MichaIng, 2019-06-27:
@lordkane 
Should be user=root, password=dietpi, if Fourdee did not somehow changed this?
Let me do a quick check on the image.

lordkane, 2019-06-27:
user=root, password=dietpi does 100 prozent not work :(

SimplyGardner, 2019-06-27:
Is it just me... but I get this error everything I try and flash the image. Tried re-downloading and starting again. 

[image]

lordkane, 2019-06-27:
> 
> 
> @lordkane
> Should be user=root, password=dietpi, if Fourdee did not somehow changed this?
> Let me do a quick check on the image.

sorry i am so stupid i have a second dietpi image run.... with pihole and connect not to the new pi

MichaIng, 2019-06-27:
@lordkane 
Okay, glad you recognised it 🙂.

@SimplyGerdner
Strange, extracts and mounts well here without issues and seems to have worked for @lordkane as well. You could check the hashes from the hash.txt file that is contained. Perhaps the download corrupted.

SimplyGardner, 2019-06-27:
@MichaIng I have following in the file. Although, I deleted the previous download, re-downloaded again and same thing happens. Strange.

```
FILE:	DietPi_v6.24_RPi-ARMv6-Buster.img
DATE:	Thu 27 Jun 15:10:42 BST 2019
MD5:	a402f96e4f8169e783f4fbdb9d6f43ee
SHA1:	c70b181d41d0b01a7efb33ea7516cec6a4041448
SHA256:	1d1ff3d7e36e9b85f1cb41fb63743d1f8716c20f02a0762a37d2d5e3588f0b49
```

MichaIng, 2019-06-27:
@SimplyGardner 
Okay when redownloaded, then a corrupt download is very unlikely. Did you try to flash with Ether?
If you are on a Windows machine, please try it with [Rufus](https://github.com/pbatard/rufus/releases/download/v3.5/rufus-3.5p.exe.)

SimplyGardner, 2019-06-27:
@MichaIng Etcher, yes. On macOS. Just tried reinstalling Etcher - no luck. Only thing I can think of is I've just upgraded to the beta of macOS (Catalina).

MichaIng, 2019-06-27:
@SimplyGardner 
The OS should not be related, however beta, one can not be 100% sure. If you have another machine to try it on, it is at least work a test.

SimplyGardner, 2019-06-27:
@MichaIng Just tried on another Mac that is not running the beta and seemed to have worked! Thank you.... now on to the setup :)

Know that release is an experimental release... is it easy to upgrade to the official dietpi release once release or will this require another re-flash of the SD card?

MichaIng, 2019-06-27:
@SimplyGardner 
As long as we do no essential changes to the image, and I don't think we need to, you can stay with this image. Once v6.25 is released it will reapply the DietPi update to include all changes done until release, including some changes to Raspberry Pi firmware repo and one additional package (raspi-copies-and-fills) to be installed, that I already applied to dev code.

Just for general interest: Could you paste the output of: `dmesg | grep 'crng init done'`

SimplyGardner, 2019-06-27:
@MichaIng 
Gotcha, so just ``sudo apt-get update && sudo apt-get upgrade`` and that should put me on the latest stable version when released? 

Is it normal the pi4 runs hot? Like 55 degrees when idle (with a heat sick too). Cant remember my 3b+ being that hot.  

Sure, output for ``dmesg | grep 'crng init done'`` is...
```
[   45.751574] random: crng init done
```

MichaIng, 2019-06-27:
@SimplyGardner 
> sudo apt-get update && sudo apt-get upgrade

`sudo dietpi-update`, which includes the APT upgrade as well. You will also get a notification max 1 day after v6.25 has been released.

> Is it normal the pi4 runs hot? Like 55 degrees when idle (with a heat sick too). Cant remember my 3b+ being that hot.

Should not be 🤔. I have no values but read that at least RPi4 takes longer to run that hot that it throttles down clocks automatically compared to RPi3.
Ah 3 dietpi-survey uploads with RPi4: https://dietpi.com/survey/#benchmark
- Indeed average idle temp higher than older RPi models.
- Not sure if it's related to current Raspbian Buster or kernel in combination with RPi4. Note that this image is also still in development on Raspbian side. Some things are still missing and after some more feedback I think some optimisations can be expected.

> [   45.751574] random: crng init done

Ayayay okay nothing changed. Very long...
If you face issues with service restart taking long or boot itself takes long until login is possible: `G_AGI haveged`
If you are interested about this, see the related issue I linked above.

ricardoandren, 2019-06-27:
Strange ram benchmark by the way, avarage (write) on the rpi 3/3+ is 366 mb/s and the pi4 is 343, despite it being lpddr4 ram...

MichaIng, 2019-06-27:
@ricardoandren 
Yeah I was also wondering. Perhaps it is indeed related to `raspi-copies-and-fills` missing on Buster images currently, but can be installed now (it was not compatible with Buster systemd/udev before): `G_AGI raspi-copies-and-fills`

But I have no idea if these functions are used natively when accessing RAM, or just when some binary explicitly implements and calls them.

You might want to run `dietpi-benchmark` once without (current state) and once with the above package installed to compare. Also possibly we need to increase the written file size to increase accuracy. Since services are stopped it is safe to use e.g. 80% of free RAM 🤔.

ricardoandren, 2019-06-27:
@MichaIng Will definitely try tomorrow when I get my pi. As it was just a thing I noticed.

SimplyGardner, 2019-06-27:
@MichaIng When installing something from ``dietpi-software`` (Avahi-Daemon) and then when I try to uninstall, it does not uninstall. Looks like it is stuck and I can not get rid of the software from this screen. 
[image]

MichaIng, 2019-06-27:
@SimplyGardner 
Good find, many thanks! Bug due to some v6.25 change about how we handle whiptail menu return values. Just fixed it with: https://github.com/MichaIng/DietPi/commit/8d382e6b2aee84306902ba6cd1e5f64b50a48996

SimplyGardner, 2019-06-28:
@Michalng
Thanks! So did the software uninstall and is just showing still on the menu section. Or, did the bug prevent it from uninstalling? Would this go in the next update or is it live now?

ricardoandren, 2019-06-28:
Chromium  with widevine: Check.
sudo reboot makes the board shutdown instead on occasion.

MichaIng, 2019-06-28:
@SimplyGardner 
> Or, did the bug prevent it from uninstalling? Would this go in the next update or is it live now?

Jep it prevented it from uninstall. Fix is not yet online, will be released with Beta v6.25.1. Note that this RPi4 image is with `dev` code.

I can't say something with Netatalk, but you can generally install any software on DietPi when following Debian install instruction. DietPi is not preventing anything. In many cases default config files are not created by deb packages or custom builds, so perhaps `/etc/netatalk/AppleVolumes.default` simply cam be created, if it does not exist.
________________
@ricardoandren 
Which kind of display do you use? HDMI or composite?
Are you able to log in SSH or is the boot process indeed hanging? In case try to be patient for 2 minutes or such, perhaps it's related to: https://github.com/MichaIng/DietPi/issues/2806

ricardoandren, 2019-06-28:
HDMI. I will show you a screenshot, it just sits idle after that, no green light blinking, but I can ssh into it. 

[image]

SimplyGardner, 2019-06-28:
@MichaIng 
There is definitely something on buster stopping the files from being added to ``/etc/netatalk``

Just tried on **stretch** (fresh install) and these are the files I should see:
[image]

On **buster**, there are missing files
[image]

Could you maybe try and see what you get? ``sudo apt-get install netatalk``

MichaIng, 2019-06-28:
@SimplyGardner 
Indeed in the Buster version of netatalk these files are not present by design: https://packages.debian.org/buster/amd64/netatalk/filelist
Check out some man pages or official docs about how it is handled with v3.X, as sad porribly these just need to be created now manually.

Ref: https://manpages.debian.org/testing/netatalk/netatalk.8.en.html

MichaIng, 2019-06-28:
@ricardoandren 
Ah yeah forgot about the two HDMI ports.

See here HDMI mode options: https://www.raspberrypi.org/documentation/configuration/config-txt/video.md
- So all config.txt entries, related to HDMI, can be applied to either port 0 or 1 while the default is port 0.

Ref2: https://www.raspberrypi.org/documentation/configuration/config-txt/conditional.md
- E.g. `hdmi_group:0=2`

Not sure now if/why port 1 is disabled by default or of there is a general switch to enable/disable one port.

DietPi side we need to add an option to dietpi-config on RPi4 to switch the used HDMI port or enable both. All HDMI related entries than need to be switched between `<setting>:1` <=> `<setting>:0` (while 0 is default), or doubled, depending on selection. Or we devide the whole Display menu into two sub menus to apply settings manually. `hdmi_hotplug_ignore:0=1` should allow to disable HDMI port 0 here.

There is a bunch of new settings as well, many only available for RPi4, e.g. enable/disable 4k output.

I am still wondering if there is some switch/setting to force HDMI output either through one or the other port, and if attached HDMI displays are auto-detected reliably otherwise (obviously not). Also if shared screen (one TTY on both screens) or one TTY for each screen can be mapped. The documentation stil lacks some details here.
__________
So for now, as DietPi v6.25 is already in Beta, we leave things as they are. HDMI port 0 should respond then according to chosen settings in dietpi.txt/dietpi-config. Switching/configuring port 1 is then something that we can add with v6.26, as this requires some more work and research and hopefully some extended documentation.

SimplyGardner, 2019-06-28:
> Indeed in the Buster version of netatalk these files are not present by design: https://packages.debian.org/buster/amd64/netatalk/filelist
> Check out some man pages or official docs about how it is [...]

@MichaIng 
Hmm, assuming Netatalk is dead? Strange how that is not included... these are core files for it work.

MichaIng, 2019-06-28:
@SimplyGardner 
Absolutely not dead, it's a whole new major version on Buster. Just the way to configure it obviously changed. Just compared a bid and it seems that on the Buster version you use afp.conf to configure mount points: https://manpages.debian.org/testing/netatalk/afp.conf.5.en.html
This was not available on Stretch.

Here the official Wiki: http://netatalk.sourceforge.net/wiki/index.php/Install_Netatalk_3.1.12_on_Debian_9_Stretch#Setting_Up
- The first part to build+install it can be skipped, but from "Setting up" on this might help.
- Note that while the Wiki states files to be in /usr/local/... APT installs place them in the directories without /usr/local, e.g. /etc and /usr/bin instead of /usr/local/bin etc.

meeki007, 2019-06-29:
> @MichaIng DietPi side we need to add an option to dietpi-config on RPi4 to switch the used HDMI port or enable both. 

I am making strides on bringing in hdmi auto detection to dietpi
Pi 4 is ordered an on the way. I will play with bringing these up mid july.

Current detection work has got me this. all the ones called DMT or CEA were scraped from the detected monitor connected and put in an array.

[image]

MichaIng, 2019-06-29:
Planned for implementation: https://github.com/MichaIng/DietPi/issues/2939
- For RPi4 it then further needs to be divided [Headless] [SDTV/Composite] [LCD Panel] [HDMI port 1] [HDMI port 2] [HDMI both ports] or something like that.
- Then show one resolution menu for each enabled HDMI port. It just need to be checked how to do `tvservice` check on a specific HDMI port.

MichaIng, 2019-06-29:
Additional topic for RPi4:
- `gpu_mem`_256/512/1024 do not apply to 2048/4096 memory RPi4 most likely.
- No new entries reported (in docs) yet. For now `gpu_mem` (without size suffix) should work.
- Doc to monitor: https://www.raspberrypi.org/documentation/configuration/config-txt/memory.md
  - `gpu_mem`: "For the Raspberry Pi 4, which is available in 1GB, 2GB and 4GB versions, the minimum and maximum value are the same as for a 1GB device." _(which does not imply that `gpu_mem_1024` is effective for all RPi4 models)_

teksindev, 2019-07-03:
Problem i have : after a while, the pi can't be reached by SSH but i see the PI it in the network. My mounted smb USB drive can't be reached anymore too. I tried to restart the pi, but impossible to access it.

The only solution is to reflash the sd card.

I have installed : emby, transmission and medusa

ricardoandren, 2019-07-03:
Thats the same problem I had except I didnt install emby transmission and medusa.

MichaIng, 2019-07-03:
@teksindev @ricardoandren 
This is both related to Raspberry Pi 4 with the testing image? You have access to local console via keyboard and monitor?

MattL0, 2019-07-04:
Just wondering. Can I download the buster image now? I mean, is the boot partition changed to get more space?

MichaIng, 2019-07-04:
@MattL0 
Jep the fresh Buster image of course has the larger boot partition already, only when upgrading the RPi firmware packages on an older image, one might run into issues.

SimplyGardner, 2019-07-04:
When is the official version of dietpi buster out?

MichaIng, 2019-07-04:
@SimplyGardner 
For Raspberry Pi we will switch to Buster very soon. I think the image above is already fine, it is still based on the latest Raspbian Lite release and now since DietPi v6.25 is release, it will be automatically updated to stable DietPi code on first boot.

Only limitation I know for non-RPi4: No Kodi available yet.
RPi4 works as well, but some dietpi-config settings are not yet RPi4 ready.

For x86_64 Buster images are already very stable (tested a long time in combination with DietPi testing in the past ~1.5 years), but we will officially set them to stable (and remove Stretch from download page) after official Debian Buster release (the next weeks).

For most other SBCs it will take a longer while I guess. firmware/drivers have to be Buster-ready and I expect this to take while, especially for older SBC models.

MichaIng, 2019-07-06:
@Fourdee 
VC4 modes related to this I guess: https://www.raspberrypi.org/forums/viewtopic.php?t=244596
Looks like master firmware version solves if (e.g. `rpi-update` for now): https://github.com/Hexxeh/rpi-firmware/commits/master
- Some VC4 related commits, including memory leak and composite mode fixes for non-RPi4 models.

Hehe now also the rpi-update notice makes it clear not NOT use rpi-update without specific reason: https://github.com/Hexxeh/rpi-firmware/commit/768ea94042bade6ea276c6cea358c9645cb5de6a

meeki007, 2019-07-06:
Thanks for keeping me (us, everyone) informed on these changes. The TO_DO_LIST grows :)

MichaIng, 2019-07-14:
New RPi4-capable Buster image uploaded: https://dietpi.com/downloads/images/DietPi_RPi-ARMv6-Buster.7z
- Based on new Raspbian Buster Lite from 10/07 and firmware upgrade with kernel 4.19.57. Hopefully some of the initial issues are solved by this.
- Based on DietPi stable code, which solves the dietpi-update menu bug and some other issues. I also manually fixed two known bug from stable code.

nboyan, 2019-07-15:
@MichaIng, Can I use apt update / upgrade only since I am with 6.25v.3 instead of pre-flash the sdcard once?

renarsvilnis, 2019-07-15:
@MichaIng If I use provided the experimental image, will it always stay on the "experimental" branch of updates? Or once it's stable It's just an update?

MichaIng, 2019-07-15:
@nboyan @renarsvilnis 
No need to reflash the image.

The experimental image was created with v6.25 dev or beta code, to have RPi4 definitions inside. But the target Git branch was manually reverted to master (stable). You can verify via:
```
grep 'DEV_GITBRANCH' /DietPi/dietpi.txt
```
Should how: `DEV_GITBRANCH=master`

If you are on v6.25.3 already, then you are already on latest stable release. If the login banner just says  `DietPi v6.25.3 : 14:19 - Mon 07/15/19` then this is as well a prove that you are on master (stable) code, otherwise it would say e.g. `DietPi v6.25.3 (beta) : 14:19 - Mon 07/15/19`.

If you are indeed not on master branch, run:
```
G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=master' /DietPi/dietpi.txt
```

If an update is offered (e.g. you are on v6.25.0/1/2), but the menu does not work (simply exits without update), then run `dietpi-update 1`. This includes apt update+upgrade.

If no update is offered, but you e.g. want to update the firmware, run as usual: `apt update && apt update` (or `apt full-upgrade`, although not required on RPi for firmware updates).

[2 more replies]

closed 2019-08-02 by MichaIng