MichaIng/DietPi #2936: Open Beta v6.25 | Please help testing and hardening the upcoming release

MichaIng, 2019-06-24:
RC version | v6.25.0
---------- | -------
Changelog | https://github.com/Fourdee/DietPi/blob/beta/CHANGELOG.txt
Code changes | https://github.com/Fourdee/DietPi/compare/master...beta
How to apply | https://github.com/Fourdee/DietPi/blob/master/BRANCH_SYSTEM.md
Release planned | End of June or beginning of July

#### When testing, especially have an eye on features mentioned in the change log above that you regularly use anyway.

______________
#### Closes issues:
#869 #1701 #2156 #2233 #2561 #2623 #2659 #2690 #2738 #2783 #2787 #2791 #2793 #2816 #2826 #2837 #2840 #2852 #2857 #2862 #2867 #2870 #2874 #2888 #2901 #2909 #2910 #2916 #2921 #2922
_______________
### Known issues
#### DietPi functionality
- DietPi-Config | Enabling WiFi + Ethernet adapters, both on different subnets, breaks WiFi connection in some cases: https://github.com/Fourdee/DietPi/issues/2103
  - Current workaround is to disable the adapter that is not in use (not for internet connection).
  - Otherwise a custom routing table is required.

#### SBC/device related
- RPi | LXAppearance (on LXDE desktop) hangs on dbus-launch: #1791
- RPi | Enabling OpenGL/VC4 causes sound card selection to be invalid: #2173
- Pine A64 | Bluetooth add-on module currently not supported: #528
- Odroid XU4 | Kodi freezes shortly on video playback: #2584
- Rock64 | 3.5mm A/V jack is currently not functional: #2522

#### Software title related
- Node-RED | Pre-installed modules cannot be updated via webUI: #2073
- Raspimjpeg | With Lighttpd, streaming mjpeg does not work: #1747
- WiFi Hotspot | Currently not working on some Odroid C2 and RPi devices: #1955 #2531

Fourdee, 2019-06-26:
@MichaIng 

Great job on this version :+1: 

Twitter REF: https://twitter.com/DietPi_/status/1143891065067462657

PeterLacknase, 2019-06-27:
Works like a charm.

MichaIng, 2019-07-01:
Beta v6.25.1 up btw: https://github.com/MichaIng/DietPi/pull/2940

MichaIng, 2019-07-01:
I introduced a little bug with v6.25.0 that breaks the dietpi-update menu. Instead of updating when hitting the confirmation button, it simply exists: https://github.com/MichaIng/DietPi/pull/2940#issuecomment-507367262
- Many thanks for reporting guys!

This is resolved with v6.25.2, please update via `dietpi-update 1` until then, which has the same end result just without menu and confirmation prompt.

teksindev, 2019-07-02:
hello, how to update please ?

i have this message

  OK  ] DietPi-Update | No update required, your DietPi installation is already up to date:
[ INFO ] DietPi-Update | Current version : v6.24.1337
[ INFO ] DietPi-Update | Latest version  : v6.24.1

MichaIng, 2019-07-02:
@teksindev 
See https://github.com/MichaIng/DietPi/blob/master/BRANCH_SYSTEM.md

MichaIng, 2019-07-02:
Beta 6.25.2 is up: https://github.com/MichaIng/DietPi/pull/2948
Next will be release, probably tomorrow, if no urgent reports are done.

MichaIng, 2019-07-03:
Released: https://github.com/MichaIng/DietPi/pull/2950

closed 2019-07-03 by MichaIng