MichaIng/DietPi #2937: Dietpi software password error on buster

MattL0, 2019-06-25:
### Creating a bug report/issue

#### Required Information
- DietPi version | 

#!/bin/bash
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=24
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version |

10.0

- Kernel version | 

Linux DietPi-Salon 4.19.50-v7+ #896 SMP Thu Jun 20 16:11:44 BST 2019 armv7l GNU/Linux

- SBC device |  RPI3 and RPI3b+

#### Steps to reproduce

1. upgrade buster from a stretch install
2. load dietpi-software

#### Expected behaviour

- Should be able to retrieve password?

#### Actual behaviour

- see picture 
[image]

MichaIng, 2019-06-25:
@MattL0 
Ah you upgraded from Stretch, then this makes sense.

Okay since I don't want to print the encryption keys and stuff here (although it's open source, if one wants to find it 😉), the easiest is:
```
rm /var/lib/dietpi/dietpi-software/.GLOBAL_PW.bin
G_CONFIG_INJECT 'AUTO_SETUP_GLOBAL_PASSWORD=' 'AUTO_SETUP_GLOBAL_PASSWORD=yourPassword' /DietPi/dietpi.txt
```
- Choose yourPassword freely of course.
- Next time DietPi-Software will re-encrypt this password silently.

The issue was that on Buster a new key derivation method is available and if not used, ugly warnings are printed. So some DietPi versions ago we moved Buster systems passwords to the new encryption method. But if you upgrade a system from Stretch to Buster, this is not done, hence DietPi-Software tries to use a key derivation method that does not fit the one of the encrypted password => failure.

I will add the ondemand encryption method upgrade to every DietPi-Software and more current DietPi-Update runs. This is actually not a big deal and allows Buster distro-upgrades then.

MattL0, 2019-06-25:
Works perfect *

MichaIng, 2019-06-25:
@MattL0 
Thanks for feedback. I will reopen the issue until I implemented the automatic encryption transition into code.

PeterLacknase, 2019-06-26:
@MattL0 

Out of curiosity:

> Steps to reproduce
> 1. upgrade buster from a stretch install
> 2. load dietpi-software

What where the exact steps (i.e. commands) you used? Did you change anything else apart from sources.list?

MattL0, 2019-06-26:
No other step  aappart from changing /etc/apt/sources.list.d/raspi.list with buster instead of stretch.

also did 

apt update 
apt full-upgrade

MichaIng, 2019-06-27:
Finished: DietPi-Software will now automatically attempt do decrypt the password with the Stretch method, if it fails on Buster. If this succeeds, it will automatically re-encrypt the password with the Buster method.

closed 2019-07-03 by MichaIng