MichaIng/DietPi #2939: DietPi-Config | Enhance display menu

MichaIng, 2019-06-29:
Separate headless mode switch which, optionally with more options to this: https://github.com/MichaIng/DietPi/issues/2442

On RPi:
- Separate switch for SDTV, HDMI or LCD panel.
- When LCD panel is selected, hide resolution menu, as fixed matching values are applied automatically.
- When SDTV is selected, hide resolution menu, show SDTV mode menu instead.
- OpenGL can be toggled independently from other settings?
- List only supported display resolution menu based on `tvservice` output/auto detection: https://github.com/MichaIng/DietPi/issues/2938#issuecomment-506932017
  - If this fails, show old menu?

meeki007, 2019-06-29:
Good day.  Working on optimizations for building the array today.  Its way too slow.

Calling on tvservice 24 times is not the answer :)  I just wanted to get the basic idea working first

[image]

MichaIng, 2019-06-29:
@meeki007 
Idea: Write `tvservice` output into a tmp file or variable once, then scrape it for entries when generating the menu.

meeki007, 2019-06-29:
@MichaIng 

Oh yep im on it. Do keep the suggestions coming. My brain is just starting to get back into the swing of bash. 

One method you do not see used much but I find very fast is using BASH built in parameter expansion.
most people that see it will pause so im kinda verbose when using it :) 

```
#Save output of tvservice -s to var | NOTE: tvservice will output a default resolution even if hdmi is Disconnected or unable to pull information from connected device
value_of_tvservice_s=$(tvservice -s)

#Transformation of the output of tvservice -s to get detected or default X,Y resolution
#get X
tvservice_framebuffer_x=${value_of_tvservice_s%x*}  # retain the part before last character x
tvservice_framebuffer_x=${tvservice_framebuffer_x##*,}  # retain the part after the last character ,
tvservice_framebuffer_x=${tvservice_framebuffer_x:1}   # Remove the first character of blank space
```

MichaIng, 2019-06-29:
@meeki007 
> One method you do not see used much but I find very fast is using BASH built in parameter expansion.
most people that see it will pause so im kinda verbose when using it :)

Jep, definitely the preferred method over external binaries (sed/grep/(m)awk) where possible (e.g. no extended regex required).

meeki007, 2019-06-29:
yep we all got spoiled with the speed of tech. Then you go and write something on a small SBC and your brain says, "2 second load ?????" "this is a Pi 3 B+"" wonder how fast this is going to be on a pi zero "

Fix now and optimize before you **keyhole** yourself in a breaking change later.

meeki007, 2019-06-30:
Just got the DMT to Main menu array finished
feel free to test it. rename to .sh and set executable. it will print out the array
Man its a verbose sucker but boy its fast.

[DMTtoMainMenu.txt](https://github.com/MichaIng/DietPi/files/3342410/DMTtoMainMenu.txt)

```
#!/bin/bash

#Save output of tvservice -m DMT to array by line
readarray -t output_of_tvservice_m_DMT_array <<< "$(tvservice -m DMT)"
#Save output of tvservice -m CEA to array by line
readarray -t output_of_tvservice_m_CEA_array <<< "$(tvservice -m CEA)"

#Check if hdmi monitor/tv is detected
#Monitor?  Get value DMT Modes value of zero if none are supported else any other value than zero it is detecting suported modes
DMT_detected=${output_of_tvservice_m_DMT_array[0]:14:1}   #first 1 characters after removing the first 14
#TV? Get value CEA Modes value of zero if none are supported else any other value than zero it is detecting suported modes
CEA_detected=${output_of_tvservice_m_CEA_array[0]:14:1}   #first 1 characters after removing the first 14 

###########################

if [[ "$DMT_detected" != '0' ]] || [[ "$CEA_detected" != '0' ]]; then
    G_WHIP_MENU_ARRAY=()
    #Add DMT & resoulutions to main menu array
    if [[ "$DMT_detected" != '0' ]]; then
      #Usable info starts at line 1
      DMT_LineNumber=1
      #Check when we reach the end of the array
      DMT_Nullcheck=${output_of_tvservice_m_DMT_array[$DMT_LineNumber]}

      
      
      until [ ! "$DMT_Nullcheck" ] || [ $DMT_LineNumber -gt 1000 ] ; do #make sure it stops even if array is huge

          #Transformation - output_of_tvservice_m_DMT_array[]
          #Sore value for check if perfered
          DMT_perfered=${output_of_tvservice_m_DMT_array[$DMT_LineNumber]%)*} #retain the part before last character )
          DMT_perfered=${DMT_perfered##*(}  # retain the part after the character (
          #Sore value for check if native
          DMT_native=${output_of_tvservice_m_DMT_array[$DMT_LineNumber]%)*} #retain the part before last character )
          DMT_native=${DMT_native##*(}  # retain the part after the character (

          #Perfered?
          if [[ "$DMT_perfered" == 'prefer' ]]; then
              
              #format to get just the DMT mode #
              DMT_perfered_mode_number=${output_of_tvservice_m_DMT_array[$DMT_LineNumber]:16} # removing the first 16 characters
              DMT_perfered_mode_number=${DMT_perfered_mode_number%%:*}  # retain the part before character :
              #format to get just the DMT resolution information
              DMT_perfered_resoulution=${output_of_tvservice_m_DMT_array[$DMT_LineNumber]#*:} # retain everything after first :
              DMT_perfered_resoulution=${DMT_perfered_resoulution##?} # Remove the fist space in the string          

              #output to Main Menu array
              G_WHIP_MENU_ARRAY+=('Detected DMT '"$DMT_perfered_mode_number"' Preferd' ': '"$DMT_perfered_resoulution")

          #Native?
          elif [[ "$DMT_native" == 'native' ]]; then
            
              #format to get just the DMT mode #
              DMT_native_mode_number=${output_of_tvservice_m_DMT_array[$DMT_LineNumber]:16} # removing the first 16 characters
              DMT_native_mode_number=${DMT_native_mode_number%%:*}  # retain the part before character :
              #format to get just the DMT resolution information
              DMT_native_resoulution=${output_of_tvservice_m_DMT_array[$DMT_LineNumber]#*:} # retain everything after first :
              DMT_native_resoulution=${DMT_native_resoulution##?} # Remove the fist space in the string          

              #output to Main Menu array
              G_WHIP_MENU_ARRAY+=('Detected DMT '"$DMT_native_mode_number"' native' ': '"$DMT_native_resoulution")

          #all other modes and resolutions
          else 

            #format to get just the DMT mode #
            DMT_mode_number=${output_of_tvservice_m_DMT_array[$DMT_LineNumber]:16} # removing the first 16 characters
            DMT_mode_number=${DMT_mode_number%%:*}  # retain the part before character :
            #format to get just the DMT resolution information
            DMT_resoulution=${output_of_tvservice_m_DMT_array[$DMT_LineNumber]#*:} # retain everything after first :
            DMT_resoulution=${DMT_resoulution##?} # Remove the fist space in the string          

            #output to Main Menu array
            G_WHIP_MENU_ARRAY+=('Detected DMT '"$DMT_mode_number" ': '"$DMT_resoulution")
            
          fi
          
          let DMT_LineNumber=DMT_LineNumber+1
          DMT_Nullcheck=${output_of_tvservice_m_DMT_array[$DMT_LineNumber]}
      done
      
      
    fi
fi

echo "G_WHIP_MENU_ARRAY by elements"
echo " "

#To print all array elements on separate lines, you can use:
printf -- "%s\n" "${G_WHIP_MENU_ARRAY[@]}"
```

meeki007, 2019-06-30:
Its working !!!
I need to:
add a dialog stating changes will take affect after reboot
some kind of response or validation that the user has made the selection

Besides that i need testing to make sure it does not break anything.

messed up pull request .... too tired but here is a link to changes of dietpi-config
https://github.com/meeki007/DietPi/pull/1/files#diff-d92a6ee04e02fd2a2dc23d5bec3e6a98

here is a copy of file

[dietpi-config.txt](https://github.com/MichaIng/DietPi/files/3342503/dietpi-config.txt)

meeki007, 2019-06-30:
On RPi:

NOT DONE - Separate switch for SDTV, HDMI or LCD panel.

KINDA - When LCD panel is selected, hide resolution menu, as fixed matching values are applied automatically.
Right now it removes most old menu options

NOT DONE - When SDTV is selected, hide resolution menu, show SDTV mode menu instead.

YES - OpenGL can be toggled independently from other settings?

YES but AUDIO NO FOR NOW -List only supported display resolution menu based on tvservice output/auto detection:
If this fails, show old menu?

I plan on moving this direction. I wanted to get the base working first and bug check before I go down the user interface path. 

For most users you don't want to have to dig around with settings. The idea is to make this easy on the top but if you need to dig for eclectic settings the user can
Submitting this plan/direction for approval:

main menu dietpi-config: Keep all the choices the same "Display Options" is what you use still use

Display Options menu: Keep all the choices the same "Change Resolution" is what you still use

Change Resolution menu: This is what I would like to change.
Ruff example
[image]

The HDMI Not Detected Changes to Detected if it detects resolutions

Also need to add AUDIO over HDMI override for LCD screens that are detected
Im thinking about putting that here 
[image]

Have it change to Disable for the screen if its enabled

None of this is set in stone. I kinda dont like the options it needs somthing im missing. I need your input to polish this a bit.

meeki007, 2019-07-02:
@MichaIng 

Problem - DMT and Connecting a DVI adaptor to a hdmi tv/monitor may disable sound

Solution - Force enabling of audio through the hdmi port.
hdmi_force_edid_audio
hdmi_drive

Question - should enabling audio by default be the standard option?
Then have the user disable it? see the pictures above for reference for menu direction.

MichaIng, 2019-07-02:
@meeki007 
Thoughts I have about the sound card selection:
- Detect sound cards by `aplay -l` output.
- RPis will have a separate menu to "enable" an additional sound card. There all the known dtoverlays will be inside + enforcing HDMI sound.
  - I am not 100% sure if running the dtoverlay command reliably makes the sound card detectable by aplay/asound, but in case of HDMI then a reboot needs to be offered.
  - After reboot then the upper auto-detection menu should list the additional sound card as well.

Currently it is a bid unreliable since the menu relies on every sound card having always the same card and device index. But his can change, e.g. if you enable VC4, then this will become card0:device0, so the selection menu then is wrong. Fourdee added a workaround for this particular issue for v6.25 but generally auto-detection is the most reliable way to go. Only downside is that most external sound cards then need a two stage selection.

meeki007, 2019-07-14:
@MichaIng 

Progress update
I would like for what ever i write to work on Stretch and Buster

The thought process:
Pi-4
```
root@DietPi:~# tvservice -l
2 attached device(s), display ID's are : 
Display Number 2, type HDMI 0
Display Number 7, type HDMI 1
```
Pi-3,2,1
```
root@DietPi:~# tvservice -l
tvservice: unrecognized option '-l'
Usage: tvservice [OPTION]...
  -p, --preferred                   Power on HDMI with preferred settings
  -e, --explicit="GROUP MODE DRIVE" Power on HDMI with explicit GROUP (CEA, DMT, CEA_3D_SBS, CEA_3D_TB, CEA_3D_FP, CEA_3D_FS)
                                      MODE (see --modes) and DRIVE (HDMI, DVI)
  -t, --ntsc                        Use NTSC frequency for HDMI mode (e.g. 59.94Hz rather than 60Hz)
  -c, --sdtvon="MODE ASPECT [P]"    Power on SDTV with MODE (PAL or NTSC) and ASPECT (4:3 14:9 or 16:9) Add P for progressive
  -o, --off                         Power off the display
  -m, --modes=GROUP                 Get supported modes for GROUP (CEA, DMT)
  -M, --monitor                     Monitor HDMI events
  -s, --status                      Get HDMI status
  -a, --audio                       Get supported audio information
  -d, --dumpedid <filename>         Dump EDID information to file
  -j, --json                        Use JSON format for --modes output
  -n, --name                        Print the device ID from EDID
  -h, --help                        Print this information
```

Writing a Check at the start to decide if it will run menu system for pi-4 or other pi's

I'm Going on 2 week vacation tomorrow :) Girl said I can do what ever I want with code.
Drive mon,tue   
at location Wed on ill be banging away on this code and bringing the pi with me :)

Ive set a goal to have a working example for you by friday :) 
Expect some more code snippets soon. Almost got the first part 1 of 5 done

meeki007, 2019-07-15:
@MichaIng 

That did it!!!!!!!!!! I'm scrapping my current code and starting over

need a new TARGETMENUID i think......... too tired to continue tonight. Just now getting my head wrapped around that section of the menu system.
I'm tired of writing the same stuff or calling function to deal with the menu system and a ton of if than elif's to make it all happen. when a new menu screen would solve it just fine.

more to come Wed night.

MichaIng, 2019-07-15:
@meeki007 
New TARGETMENUID should not be required for a "simple" resolution selection menu. This is only required if you create a whole new submenu with multiple options. I would just show two identical resolution menus for HDMI 0 and HDMI 1 on RPi4. To check: `if (( $G_HW_MODEL == 4 )); then this_is_RPi4...`
On the other hand, humm one should be able to toggle SDTV and such. Another question: If you have two monitors with audio capability attached and both with CEA modes enabled, how does the system decide where to send the HDMI audio signal to? Or does it use both, or HDMI 0 if possible, else HDMI 1? Or are both listed as different sound devices? Could be important for the audio options menu, to allow selecting the target HDMI port on RPi4 in case both are generally available as `hw:0,0`.

However enjoy hour vacation and don't do too much coding there. You know somethings girls allow things on the one hand but want you to decide against it, at least partly, on the other side 😉.

meeki007, 2019-07-20:
@MichaIng 

> If you have two monitors with audio capability attached and both with CEA modes enabled, how does the system decide where to send the HDMI audio signal to?

you specifiy the device you want to use.

example:

```
root@DietPi:~# tvservice -s -v 7 -a
state 0xa [HDMI DMT (28) RGB full 16:10], 1280x800 @ 60.00Hz, progressive
     PCM supported: Max channels: 2, Max samplerate:  48kHz, Max samplesize 16

so if we have audio forced on hdmi it must send it out both channels ????? 
I'm on vacation with on monitor can't test this.

However I have a nice menu screen that is coming along nicely........ hang in there I have not forgot DietPI,

meeki007, 2019-07-20:
@MichaIng 

Sorry I've been down a rabbit hole with https://github.com/MichaIng/DietPi/blob/master/dietpi/func/dietpi-set_hardware

as HDMI is set **not** using framebuffer
so if the user in my current menu system selects to enable vc4 it will change his/her resolution to 1080p and overide the HDMI autodetected resolution they selected.

I'm moving on from this and leaving it broken for now.

meeki007, 2019-07-20:
@MichaIng 

https://www.amazon.com/gp/product/B06W2JXLSK/ref=ox_sc_act_title_2?smid=A3TI1DCSJ1BL83&psc=1

composite cable in the mail now.

needed for testing of new video options menu

MichaIng, 2019-09-04:
@meeki007 
How far are you with menu coding? If you can, please open a PR, and I will help implementing what is missing and polish in case.

We need to get v6.26 beta up soon 🙂.

MichaIng, 2019-09-16:
Added SDTV support for RPi4: https://github.com/MichaIng/DietPi/commit/9488ce159fa5ebde498e4d2c51e650bec4b21151
Changelog: https://github.com/MichaIng/DietPi/commit/be075a3605969cdfecd02e183ae90353c71b575e

Further RPi4 specific settings and HDMI port 1 support are postponed to v6.27, as we need to release v6.26 now.

MichaIng, 2025-02-09:
We added a new tool to control console modes and rotation via DRM/KMS sysfs and `video` kernel command-line parameter: #7390
`tvservice` and the old `config.txt` display options have been deprecated resp. removed with the new RPi kernel, hence I'll close this issue and in case implement still relevant features into `dietpi-display`.

closed 2025-02-09 by MichaIng