MichaIng/DietPi #2944: Failure when launching dietpi-software

JJGO, 2019-06-30:
### Creating a bug report/issue

#### Required Information
- DietPi version | 6.22.3
- Distro version | stretch
- Kernel version | 4.19.42-v7+
- SBC device | RPi 3b+

Recently I have run into issues when executing `dietpi-software`, the connection test times out and I get the following

    DietPi-Software: Connection test: http://raspbian.raspberrypi.org/raspbian              
     - Exit code: 4                                                                         
     - DietPi version: v6.22.3 (Fourdee/master) | HW_MODEL:3 | HW_ARCH:2 | DISTRO:4         
     - Image creator: DietPi Core Team                                                      
     - Pre-image: Raspbian Lite                                                             
                                                                                            
    Log file contents:                                                                      
    Spider mode enabled. Check if remote file exists.                                       
    --2019-06-30 20:40:32--  http://raspbian.raspberrypi.org/raspbian                       
    Resolving raspbian.raspberrypi.org (raspbian.raspberrypi.org)... failed: Connection     
    timed out.                                                                              
    wget: unable to resolve host address ‘raspbian.raspberrypi.org’       

However the server has a valid network connection and it is reachable from the outside (all my webservices are running without issue). I have tried rebooting and I have checked that `wget  raspbian.raspberrypi.org/raspbian` works as expected when run on its own. I use cloudfare DNS (1.1.1.1) but I have an identical rpi with dietpi with that DNS and it still works without issue.

MichaIng, 2019-06-30:
@JJGO 
Many thanks for your report.

DNS resolving takes very long and times out our connection test.
Can you please do:
```
time wget --spider http://raspbian.raspberrypi.org/raspbian
```

Custom timeouts are possible since v6.23 but to allow update you need to adjust the hard coded timeout first:
```
sed -i 's/local timeout=\$\{timeout:-5\}/local timeout=\$\{timeout:-20\}/' /DietPi/dietpi/func/dietpi-globals
G_CHECK_URL http://raspbian.raspberrypi.org/raspbian
```
- Check if the second command then works with 20 seconds timeout.

JJGO, 2019-06-30:
Output of `time wget --spider http://raspbian.raspberrypi.org/raspbian`

    Spider mode enabled. Check if remote file exists.
    --2019-06-30 22:12:24--  http://raspbian.raspberrypi.org/raspbian
    Resolving raspbian.raspberrypi.org (raspbian.raspberrypi.org)... 93.93.128.193, 2a00:1098:0:80:1000:75:0:3
    Connecting to raspbian.raspberrypi.org (raspbian.raspberrypi.org)|93.93.128.193|:80... connected.
    HTTP request sent, awaiting response... 301 Moved Permanently
    Location: http://raspbian.raspberrypi.org/raspbian/ [following]
    Spider mode enabled. Check if remote file exists.
    --2019-06-30 22:12:29--  http://raspbian.raspberrypi.org/raspbian/
    Connecting to raspbian.raspberrypi.org (raspbian.raspberrypi.org)|93.93.128.193|:80... connected.
    HTTP request sent, awaiting response... 200 OK
    Length: unspecified [text/html]
    Remote file exists and could contain further links,
    but recursion is disabled -- not retrieving.

    real    0m5.172s
    user    0m0.008s
    sys     0m0.020s

Output after `haveged` and `sleep`

    Spider mode enabled. Check if remote file exists.
    --2019-06-30 22:14:22--  http://raspbian.raspberrypi.org/raspbian
    Resolving raspbian.raspberrypi.org (raspbian.raspberrypi.org)... 93.93.128.193, 2a00:1098:0:80:1000:75:0:3
    Connecting to raspbian.raspberrypi.org (raspbian.raspberrypi.org)|93.93.128.193|:80... connected.
    HTTP request sent, awaiting response... 301 Moved Permanently
    Location: http://raspbian.raspberrypi.org/raspbian/ [following]
    Spider mode enabled. Check if remote file exists.
    --2019-06-30 22:14:27--  http://raspbian.raspberrypi.org/raspbian/
    Connecting to raspbian.raspberrypi.org (raspbian.raspberrypi.org)|93.93.128.193|:80... connected.
    HTTP request sent, awaiting response... 200 OK
    Length: unspecified [text/html]
    Remote file exists and could contain further links,
    but recursion is disabled -- not retrieving.

    real    0m5.179s
    user    0m0.020s
    sys     0m0.010s

Since you mention the low entropy, I do monitor the size of the entropy pool [here](https://imgur.com/a/inSBoeS) is a graph covering the last 24h. As of now `cat /proc/sys/kernel/random/entropy_avail` returns 1915 but wget takes a similar amount of time

JJGO, 2019-06-30:
After changing the timeout from 5 to 20 in the globals file, `dietpi-software` now runs correctly. the `G_CHECK_URL` command just freezes and hangs there without any output.

MichaIng, 2019-06-30:
@JJGO 
Okay so not entropy related. Slightly more than 5 seconds for name resolving, totally unacceptable IMO.
You could play around between using your router (which forwards the request to some local DNS server, provided by your ISP in most cases) or Google DNS 8.8.8.8 and see if this works better. Although CloudFlare should do fine as well, actually...

You have usual cable based or DSL intern access or any special network setup? In one reported case it was up to 20 seconds because of satellite internet.
WiFi should have only minor effect, my print above was from a VM with bridged network via host WiFi adapter, so a quite bad local network setup actually 😉.

If you have IPv6 enabled (seems like from our output) you can try to disable it (also dietpi-config > Network Options: Adapters).

>  the G_CHECK_URL command just freezes and hangs there without any output.

But `dietpi-software` works well? It just calls `G_CHECK_URL` as well. You reloaded globals to your bash session after change? `. /DietPi/dietpi/func/dietpi-globals` (sourcing, so space between dot and path)

Okay however good that it works now. As said with v6.23 we added new options to dietpi-config > Network Options: Adapters to adjust the timeout and attempts. We just need to define some for non-interactive calls, otherwise wget can theoretically run forever.

MichaIng, 2019-06-30:
Further debugging steps:
- Check if curl tasks as well so long:
```
time curl -IL http://raspbian.raspberrypi.org/raspbian
```
- Check if raw libc name resolving takes as well so long:
```
time getent hosts raspbian.raspberrypi.org
```
- And when forcing IPv4?
```
time getent ahostsv4 raspbian.raspberrypi.org
```

JJGO, 2019-07-10:
I ended up figuring out the cause. After changing DNS from Cloudfare to Google (1.1.1.1 to 8.8.8.8) curl started to take the expected amount of time. I am unsure why this might be the case, my hypothesis is that my ISP is involved.

MichaIng, 2019-07-10:
@JJGO 
Strange, but yeah, last resort is Google DNS which AFAIK has most servers around the world.

But as said, best is IMO to use the default DNS that is provided by your ISP and configured in your router. That should always work, is applied via DHCP anyway (to other devices in your network) etc. In very most cases you can use the local router IP as DNS nameserver, which is forwarded then. And the router in most cases caches entries, so it should be fastest method as well for regularly accessed hosts.

JJGO, 2019-07-11:
Yes, I do that by default, however this ISP uses its DNS to block some webpages, that is why I am using a third party one.

MichaIng, 2019-07-11:
@JJGO 
Ah okay makes sense then.

Okay I mark this issue as closed then. Feel free to reopen if required.

closed 2019-07-11 by MichaIng