MichaIng/DietPi #2945: Mariadb preventing OS upgrade

JJGO, 2019-07-01:
#### Required Information
- DietPi version | 6.22.3
- Distro version | stretch
- Kernel version | 4.19.42-v7+
- SBC device | RPi 3b+

When running `dietpi-update` the process aborts due to some misconfiguration of `mariadb`. Output below

    [  OK  ] DietPi-Update | G_AGUP                                                 
    [ INFO ] DietPi-Update | APT upgrade, please wait...                            
    Setting up mariadb-server-10.1 (10.1.38-0+deb9u1) ...                           
    dpkg: error processing package mariadb-server-10.1 (--configure):                subprocess installed post-installation script returned error exit status 1     dpkg: dependency problems prevent configuration of mariadb-server:              
     mariadb-server depends on mariadb-server-10.1 (>= 10.1.38-0+deb9u1); however:  
      Package mariadb-server-10.1 is not configured yet.                            
                                                                                    
    dpkg: error processing package mariadb-server (--configure):                    
     dependency problems - leaving unconfigured                                     
    Errors were encountered while processing:                                       
     mariadb-server-10.1                                                            
     mariadb-server                                                                 
    E: Sub-process /usr/bin/dpkg returned an error code (1)  

I have tried applying some fixes like [this](https://serverfault.com/questions/736419/cant-install-mariadb-dependency-error), [this](https://serverfault.com/questions/915158/how-to-forcefully-remove-mysql-and-mariadb-with-unmet-dependencies-when-apt-get) and [this](https://unix.stackexchange.com/questions/249530/mariadb-dependency-problems-leaving-unconfigured) but none of those worked.

Is there any precedence of this issue?

MichaIng, 2019-07-01:
@JJGO 
Many thanks for your report.

Lets see, link 1 does not apply since DietPi already installs die mariadb-server package. Generally I would be carefull with purging those packages since possibly existing databases will be purged what you definitly don't want. So the other two fix attempts are no good "solution" as well, more a hammer method, break down and make new...

I bet the MariaDB service fails for some reason, which makes the package configuration fail.
Please paste: `journactl -u mariadb-server`

JJGO, 2019-07-01:
That outputs 

    -- No entries --

It looks like no database is running by grepping through `ps aux` but doing `apt install mariadb-server` renders the same error

MichaIng, 2019-07-01:
@JJGO 
Ah sorry, wrong service name, it's just: `journactl -u mariadb`

JJGO, 2019-07-01:
That also returns `-- No entries --`

MichaIng, 2019-07-01:
@JJGO 
Okay strange, what does `systemctl restart mariadb` say?

JJGO, 2019-07-01:
`systemctl` returns `Failed to restart mariadb.service: Unit mariadb.service is masked.`

MichaIng, 2019-07-01:
@JJGO 
> Unit mariadb.service is masked

Lol, why that...

```
systemctl unmask mariadb
systemctl start mariadb
```

JJGO, 2019-07-01:
Now the `journalctl` command outputs 

    Jul 01 17:44:12  systemd[1]: Starting MariaDB 10.1.38 database server...
    Jul 01 17:44:13  mysqld[9094]: 2019-07-01 17:44:13 1995811632 [Note] /usr/sbin/mysqld (mysqld 10.1.38-MariaDB-0+deb9u1) starting as process 9094 ...
    Jul 01 17:44:16  systemd[1]: mariadb.service: Main process exited, code=exited, status=1/FAILURE
    Jul 01 17:44:16  systemd[1]: Failed to start MariaDB 10.1.38 database server.
    Jul 01 17:44:16  systemd[1]: mariadb.service: Unit entered failed state.
    Jul 01 17:44:16  systemd[1]: mariadb.service: Failed with result 'exit-code'.

The log at `/var/log/mysql/error.log` contains the following 

    2019-07-01 17:44:13 1995811632 [Note] InnoDB: innodb_empty_free_list_algorithm has been changed to legacy because of small buffer pool size. In order to use backoff, increase buffer pool at least up to 20MB.

    2019-07-01 17:44:13 1995811632 [Note] InnoDB: Using mutexes to ref count buffer pool pages
    2019-07-01 17:44:13 1995811632 [Note] InnoDB: The InnoDB memory heap is disabled
    2019-07-01 17:44:13 1995811632 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
    2019-07-01 17:44:13 1995811632 [Note] InnoDB: GCC builtin __atomic_thread_fence() is used for memory barrier
    2019-07-01 17:44:13 1995811632 [Note] InnoDB: Compressed tables use zlib 1.2.8
    2019-07-01 17:44:13 1995811632 [Note] InnoDB: Using Linux native AIO
    2019-07-01 17:44:13 1995811632 [Note] InnoDB: Using generic crc32 instructions
    2019-07-01 17:44:13 1995811632 [Note] InnoDB: Initializing buffer pool, size = 128.0M
    2019-07-01 17:44:13 1995811632 [Note] InnoDB: Completed initialization of buffer pool
    2019-07-01 17:44:13 1995811632 [Note] InnoDB: Highest supported file format is Barracuda.
    2019-07-01 17:44:13 1995811632 [Note] InnoDB: 128 rollback segment(s) are active.
    2019-07-01 17:44:13 1995811632 [Note] InnoDB: Waiting for purge to start
    2019-07-01 17:44:13 1995811632 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.42-84.2 started; log sequence number 203892126
    2019-07-01 17:44:14 1384117056 [Note] InnoDB: Dumping buffer pool(s) not yet started
    2019-07-01 17:44:14 1995811632 [Note] Plugin 'FEEDBACK' is disabled.
    2019-07-01 17:44:14 1995811632 [Note] Recovering after a crash using tc.log
    2019-07-01 17:44:14 1995811632 [ERROR] Bad magic header in tc log
    2019-07-01 17:44:14 1995811632 [ERROR] Crash recovery failed. Either correct the problem (if it's, for example, out of memory error) and restart, or delete tc log and start mysqld with --tc-heuristic-recover={commit|rollback}
    2019-07-01 17:44:14 1995811632 [ERROR] Can't init tc log
    2019-07-01 17:44:14 1995811632 [ERROR] Aborting

MichaIng, 2019-07-01:
@JJGO 
Try:
```
rm /var/lib/mysql/tc.log
systemctl restart mariadb
```

MichaIng, 2019-07-19:
@JJGO
Did the above solve your issue?

JJGO, 2019-07-21:
Yes, that did solve the issue

closed 2019-07-21 by JJGO