MichaIng/DietPi #2949: Installing Xserver FAILED

ludovick, 2019-07-03:
#### Details:
- Date           | Wed  3 Jul 11:26:36 CEST 2019
- Bug report     | N/A
- DietPi version | v6.24.1 (MichaIng/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 2 Model B (armv7l) (index=2)
- Kernel version | #1219 SMP Tue May 14 21:20:58 BST 2019
- Distro         | stretch (index=4)
- Command        | Connection test: https://raw.githubusercontent.com/MichaIng/DietPi/master/.conf/dps_6/xcompmgr.desktop
- Exit code      | 4
- Software title | DietPi-Software

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
Spider mode enabled. Check if remote file exists.
--2019-07-03 11:26:33--  https://raw.githubusercontent.com/MichaIng/DietPi/master/.conf/dps_6/xcompmgr.desktop
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 192.168.0.8
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|192.168.0.8|:443... failed: Connection refused.
```

MichaIng, 2019-07-03:
@ludovick 
Many thanks for your report. Humm something is wrong with your DNS server:
> Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 192.168.0.8

It resolves `raw.githubusercontent.com` to an IP in your local network. Which DNS server do you use?
`cat /etc/resolv.conf` (or in case you use Pi-hole, which upstream DNS server?)

ludovick, 2019-07-03:
c'est résolu j'ai ajouté raw.githubusercontent.com et github.com dans la liste blanche de Pi-hole
merci beaucoup

it's resolved i added raw.githubusercontent.com and github.com in the Pi-hole whitelist
thank you so much

MichaIng, 2019-07-03:
@ludovick 
I am still confused where these blocklist entries come from. Did you add/enable any further blocklist that is not selectable during the Pi-hole installer?

ludovick, 2019-07-03:
only this list :

https://raw.githubusercontent.com/StevenBlack/hosts/master/hosts |  
  | http://sysctl.org/cameleon/hosts |  
  | https://s3.amazonaws.com/lists.disconnect.me/simple_tracking.txt |  
  | https://s3.amazonaws.com/lists.disconnect.me/simple_ad.txt |  
  | https://v.firebog.net/hosts/lists.php?type=tick

MichaIng, 2019-07-05:
@ludovick 
Hmm just checked all files (including all linked files from https://v.firebog.net/hosts/lists.php?type=tick) but could not find the related github.com entry. There were some github.com subdomains in one of them, but not the ones the cause the download issue. However perhaps I've overseen something.

Regardless workaround (add the whitelist) works well, so I mark this as closed.

closed 2019-07-05 by MichaIng