MichaIng/DietPi #2956: mpd won't upgrade

maartenlangeveld, 2019-07-04:
- DietPi version | 6.25 RC3
- Distro version | stretch
- Kernel version | Linux DietPi 4.19.42+ #1219 Tue May 14 21:16:38 BST 2019 armv6l GNU/Linux
- SBC device | RPi Zero W (armv6l)

> root@DietPi:/# apt upgrade
> Reading package lists... Done
> Building dependency tree
> Reading state information... Done
> Calculating upgrade... Done
> The following packages have been kept back:
>   mpd
> 0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.

> root@DietPi:/# apt upgrade mpd
> Reading package lists... Done
> Building dependency tree
> Reading state information... Done
> Calculating upgrade... Done
> Some packages could not be installed. This may mean that you have
> requested an impossible situation or if you are using the unstable
> distribution that some required packages have not yet been created
> or been moved out of Incoming.
> The following information may help to resolve the situation:
> 
> The following packages have unmet dependencies:
>  mpd : PreDepends: init-system-helpers (>= 1.54~) but 1.48 is to be installed
>        Depends: libadplug-2.2.1-0v5 (>= 2.2.1-0) but it is not going to be installed
>        Depends: libao4 (>= 1.1.0) but it is not going to be installed
>        Depends: libaudiofile1 (>= 0.3.6) but it is not going to be installed
>        Depends: libavcodec58 (>= 7:4.0)
>        Depends: libavformat58 (>= 7:4.1) but it is not going to be installed
>        Depends: libavutil56 (>= 7:4.0) but it is not going to be installed
>        Depends: libcdio-cdda2 (>= 10.2+0.94+2) but it is not going to be installed
>        Depends: libcdio-paranoia2 (>= 10.2+0.94+2) but it is not going to be installed
>        Depends: libcdio18 (>= 2.0.0) but it is not going to be installed
>        Depends: libfluidsynth1 (>= 1.1.6-4~) but it is not going to be installed
>        Depends: libgcrypt20 (>= 1.8.0) but 1.7.6-2+deb9u3 is to be installed
>        Depends: libicu63 (>= 63.1-1~) but it is not going to be installed
>        Depends: libiso9660-11 (>= 2.0.0) but it is not going to be installed
>        Depends: libixml10 (>= 1:1.8.0) but it is not going to be installed
>        Depends: libmikmod3 (>= 3.3.3) but it is not going to be installed
>        Depends: libmms0 (>= 0.4) but it is not going to be installed
>        Depends: libmodplug1 (>= 1:0.8.8.5) but it is not going to be installed
>        Depends: libmp3lame0 (>= 3.100) but 3.99.5+repack1-9 is to be installed
>        Depends: libmpcdec6 (>= 1:0.1~r435) but it is not going to be installed
>        Depends: libmpdclient2 (>= 2.16) but 2.9-1 is to be installed
>        Depends: libnfs12 (>= 1.9.7) but it is not going to be installed
>        Depends: libshout3 (>= 2.4.1) but 2.3.1-3 is to be installed
>        Depends: libsidplayfp4 but it is not going to be installed
>        Depends: libstdc++6 (>= 7) but 6.3.0-18+rpi1+deb9u1 is to be installed
>        Depends: libupnp13 (>= 1:1.8.3) but it is not going to be installed
>        Depends: libwildmidi2 (>= 0.2.3) but it is not going to be installed
>        Depends: libzzip-0-13 (>= 0.13.56) but it is not going to be installed
>        Depends: libjs-sphinxdoc (>= 1.0) but it is not going to be installed
> E: Broken packages

MichaIng, 2019-07-04:
@maartenlangeveld 
Many thanks for your report.

Note that it is not intended to upgrade MPD to the version from the Buster branch on Stretch systems.
- ARMv6 systems have the Buster branch added when installing PHP to allow PHP7.3 install, which is more and more required for web applications, respectively PHP7.0 support is dropped more and more.
- But we only install the packages from Buster that are strongly required for PHP7.3 and to resolve any related dependency issues.
- We already ship an updated own build of MPD on Stretch with, v0.20.23, where on Stretch repo only v0.19.21 is available. On Buster the newer v0.21.5 is available, but as you can see, installing this would require to upgrade several other packages as well, finally the whole system would become Buster easily, which is not intended.

`apt upgrade` itself (without explicitly adding "mpd") should not offer or attempt the MPD update. Check `apt policy mpd` to assure that v0.21.5 is not the install candidate.
__________
But since Raspbian officially release Buster for RPi4 support, the above issue is solved for fresh installs. Fourdee already created a testing image, for RPi4 is still needs some tuning and added features since many things have changed firmware-wise, but it should work quite well on other RPi models: https://github.com/MichaIng/DietPi/issues/2935#issuecomment-506389899
- Kodi is still missing on the Raspberry Pi Buster branch and possibly some other packages, as Raspbian/RPi Foundation are as well still tuning this release.

maartenlangeveld, 2019-07-04:
@MichaIng 
Thanks, now I understand. I will close this 'issue'.

closed 2019-07-04 by maartenlangeveld