MichaIng/DietPi #2957: error when upgrading to 6.25 (O!MPD)

maartenlangeveld, 2019-07-04:
- DietPi version | 6.25 RC3
- Distro version | stretch
- Kernel version | Linux DietPi 4.19.42+ #1219 Tue May 14 21:16:38 BST 2019 armv6l GNU/Linux
- SBC device\| RPi Zero W (armv6l)

```
DietPi-Software | RPi Zero W (armv6l) | IP: 192.168.168.10

          ┌─────────────────────────────────────────────────┤ DietPi-Software ├──────────────────────────────────────────────────┐
          │                                                                                                                      │
          │ G_THREAD ERROR:                                                                                                      │
          │  - Command = curl -L https://yt-dl.org/downloads/latest/youtube-dl -o /usr/local/bin/youtube-dl                      │
          │  - Exit code = 35                                                                                                    │
          │                                                                                                                      │
          │   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current                                      │
          │                                  Dload  Upload   Total   Spent    Left  Speed                                        │
          │ ^M  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0^M  0     0    0     0    0     0    │
          │ 0 --:--:--  0:00:11 --:--:--     0                                                                                   │
          │ curl: (35) OpenSSL SSL_connect: SSL_ERROR_SYSCALL in connection to yt-dl.org:443                                     │
          │                                                                                                                      │
          │                                                        <Ok>                                                          │
          │                                                                                                                      │
          └──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┘
```

MichaIng, 2019-07-04:
@maartenlangeveld 
Many thanks for your report.
Strange, works well here.

Can you please try and paste:
```
cd /tmp
wget https://yt-dl.org/downloads/latest/youtube-dl
```
- I found that this can occur when IPv6 is attempted. In this case you could try to disable IPv6: dietpi-software > Network Settings: Adapters > IPv6
- Also wrong configured proxy settings can cause this, did you setup a proxy?

And last but not least it might be a OpenSSL/curl internal issue since libssl and curl are upgraded to the Buster version (required to satisfy dependencies for PHP7.3) but OpenSSL is not. However no reported issues about this since introduced with v6.23.4/24.

MichaIng, 2019-07-04:
@maartenlangeveld 
Great the manual download worked. Indeed strange it failed first, perhaps a temporary issue with the yt-dl server. However please report back if you face `SSL_ERROR_SYSCALL` another time.

closed 2019-07-04 by MichaIng