MichaIng/DietPi #2958: Install mpc along with mpd

maartenlangeveld, 2019-07-04:
Sometimes it's practical to use mpc, e.g. when one of the mpd clients (mympd, O!mpd etc.) fails for some reasons, can mpc be installed alongside mpd ?

maartenlangeveld, 2019-07-04:
@MichaIng 
Have installed mpc using apt, was only suggestion to automatically install when installing mpd.

MichaIng, 2019-07-04:
@maartenlangeveld 
Hmm yeah not sure, as most users seem to not require it, at least I couldn't find some related request on quick check. I would keep the install slim until further requests appear.

You could add it to https://feathub.com/MichaIng/DietPi/ and we'll see how much votes we get.

maartenlangeveld, 2019-07-04:
Seems to work well as root, but have not tried all possibilities. I needed mpc since I use also upmpdcli (DietPi) with Linn Kazoo (on Windows PC) as alternative to O!MPD and when playing radio Linn Kazoo sets 'single: on'... When returning tot O!MPD it wouldn't play an whole album until I discovered that mpd single flag was set to 'on'. Needed mpc to set it back to off.

So not important and you're right that you want to keep DietPi install slim. I will close since not important at all. Thanks.

BTW upmpdcli works as a charm, can recommend.

closed 2019-07-04 by maartenlangeveld