MichaIng/DietPi #2959: Update to 6.24.3 is not working, because gitea-1.8-linux-arm-7 is not available

maschiw, 2019-07-04:
#### Details:
- Date           | Thu  4 Jul 11:50:44 BST 2019
- Bug report     | N/A
- DietPi version | v6.24.3 (MichaIng/master)
- Img creator    | DietPi Core Team
- Pre-image      | Meveric
- SBC device     | Odroid XU3/XU4/HC1/HC2 (armv7l) (index=11)
- Kernel version | #1 SMP PREEMPT Thu Aug 23 05:59:33 UTC 2018
- Distro         | stretch (index=4)
- Command        | Connection test: https://dl.gitea.io/gitea/1.8/gitea-1.8-linux-arm-7
- Exit code      | 8
- Software title | DietPi-Software

#### Steps to reproduce:
I just called dietpi-update

#### Expected behaviour:
Update the system

#### Actual behaviour:
The check if https://dl.gitea.io/gitea/1.8/gitea-1.8-linux-arm-7 exists is failing,
because it is not there
The file gitea-1.8-linux-arm-7 is not listed at https://dl.gitea.io/gitea/1.8/

#### Additional logs:
```
Log file contents:
Spider mode enabled. Check if remote file exists.
--2019-07-04 11:50:18--  https://dl.gitea.io/gitea/1.8/gitea-1.8-linux-arm-7
Resolving dl.gitea.io (dl.gitea.io)... 104.27.142.155, 104.27.143.155, 2606:4700:30::681b:8f9b, ...
Connecting to dl.gitea.io (dl.gitea.io)|104.27.142.155|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
Remote file does not exist -- broken link!!!

MichaIng, 2019-07-04:
@maschiw 
Many thanks for your report.

You are right, somehow the ARMv7 binary is missing in all v1.8 version, ARMv5/6/64 i386/x64_64 are all there. No idea what happened there, if it's a temporary upload/sync issue or wanted: https://dl.gitea.io/gitea/

Will check release notes and opened issues/reports about this.

MichaIng, 2019-07-04:
Found: https://github.com/go-gitea/gitea/issues/7222
- They have an unresolved issue with ARMv7 builds.
- But the ARMv6 binaries are backwards compatible, so can be used on ARMv7 systems as well.
- Here is the initial bug report for v1.8 and ARMv7: https://github.com/go-gitea/gitea/issues/6700

MichaIng, 2019-07-04:
Workaround:
```
sed -i 's/arm-7/arm-6/' /DietPi/dietpi/dietpi-software
```
- Added for v6.26 with: https://github.com/MichaIng/DietPi/commit/a7eb923f03ac3848791049830b962cc0939727ce
- Changelog: https://github.com/MichaIng/DietPi/commit/88318e3001b3009f80b619a3001550e2c8980ea4

maschiw, 2019-07-05:
Thanks for your workaround. I implemented it and started dietpi-update again.
I expected a new update run. But dietpi-update said "No update required" and "Current version: v6.25.3".
Have I to use a special dietpi-update parameter like "dietpi-update -1"?
Or is it good to have a backup?

closed 2019-07-05 by maschiw

MichaIng, 2019-07-05:
@maschiw 
`dietpi-update -1` will rerun the whole update. Since only the Gitea reinstall failed, reinstalling it is sufficient: `dietpi-software reinstall 165`

MichaIng, 2019-07-06:
Okay I merged the hotfix into the v6.25 patch file. Just verified that it is official aim/recommendation to use the ARMv6 binaries on ARMv7 from v1.8 on.

I mark this as closed.