MichaIng/DietPi #2960: Unable to update to v6.25.3 from v6.24.1337

adamgibbo, 2019-07-04:
### Creating a bug report/issue

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=24
G_DIETPI_VERSION_RC=1337
G_GITBRANCH='master'
G_GITOWNER='MichaIng'`
- Distro version | `Buster`
- Kernel version | `Linux DietPi 4.19.56-v7+ #1242 SMP Wed Jun 26 17:31:47 BST 2019 armv7l GNU/Linux`
- SBC device | `RPi 3 Model B+ (armv7l)`

#### Steps to reproduce

1. sudo dietpi-update
2. 
[image]
3. Can not proceed any further when selecting 'ok' just flashes and returns to the menu
4 . Following text is then showing when exiting:
[image]

Any assistance would be appreciated.
Thanks

MichaIng, 2019-07-04:
@adamgibbo 
Many thanks for your report.

Jep known issue with the dev code that the RPi4 testing image is based on. Please update by skipping the manu: `dietpi-update 1`
- It is resolved with v6.25

adamgibbo, 2019-07-04:
@MichaIng `dietpi-update 1` has worked.
Many thanks

MichaIng, 2019-07-04:
I will reopen an Pin, since all RPi4 and beta testers are affected.

closed 2019-09-02 by MichaIng