MichaIng/DietPi #2962: hostapd failed to start

fossilaudio, 2019-07-05:
#### Details:
- Date           | Fri  5 Jul 10:53:16 PDT 2019
- Bug report     | N/A
- DietPi version | v6.25.3 (MichaIng/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B+ (armv7l) (index=3)
- Kernel version | #1219 SMP Tue May 14 21:20:58 BST 2019
- Distro         | stretch (index=4)
- Command        | systemctl start hostapd
- Exit code      | 1
- Software title | DietPi-Config

#### Steps to reproduce:

1. Fresh install.
2. Install #60 from DietPi-Software (WiFi HotSpot)

#### Expected behaviour:

-

#### Actual behaviour:

- ...

#### Extra details:

- partially resolved with "dietpi-services unmask hostapd"
- Unmasking service resurfaces issue logged at https://github.com/MichaIng/DietPi/issues/2894 

#### Additional logs:
```
Log file contents:
Failed to start hostapd.service: Unit hostapd.service is masked.
```

MichaIng, 2019-07-07:
@kcasner 
Many thanks for your report.

Strange there was another case reported where hostapd.service was masked. No idea what does this, perhaps another package/installer that is simply incompatible, not sure.

Did the `G_AGI --reinstall hostapd` resolved it for you as well?

kcasner, 2019-07-07:
You’re welcome. Yes. Everything is working once those two commands are applied.

symbios24, 2019-07-21:
Hi, the command to reinstall did not work for me(Rpi3B) wifi hotspot is not active
failed to start hostapd.service: Unit hostapd.service is masked

MichaIng, 2019-07-22:
@symbios24 
Jep, unmask is not yet implemented in current dietpi-software code. Will add it for v6.26. I am still wondering what places the mask, as you are not the only one who faced it 🤔.

MichaIng, 2019-09-16:
Solved via: https://github.com/MichaIng/DietPi/commit/cdcb87a10c355d3f09b3099737321bc39161f989

closed 2019-10-15 by MichaIng