MichaIng/DietPi #2964: dietpi-software command fails

soloza, 2019-07-07:
#### Details:
- Date           | Sun  7 Jul 07:25:16 BST 2019
- Bug report     | N/A
- DietPi version | v6.25.3 (MichaIng/master)
- Img creator    | DietPi Core Team
- Pre-image      | Debian mini.iso
- SBC device     | Native PC (x86_64) (index=21)
- Kernel version | #1 SMP Debian 4.9.144-3.1 (2019-02-19)
- Distro         | stretch (index=4)
- Command        | Connection test: http://security.debian.org
- Exit code      | 127
- Software title | DietPi-Software

#### Steps to reproduce:

1. ... run dietpi-software
2. ...

#### Expected behaviour:

- ... want to configure software i.e.uninstall app

#### Actual behaviour:

- ... 

#### Extra details:

- ... Internet connection is working 100%, have changed APT sources
[image]

#### Additional logs:
```
Log file contents:
/DietPi/dietpi/func/dietpi-globals: line 1379: wget: command not found

MichaIng, 2019-07-07:
@soloza 
You or something removed `wget`, which is an essential download utility for DietPi. To resolve: `G_AGI wget`

soloza, 2019-07-07:
Hi.
Very strange as server has been running a week and have not changed anything.

ran
root@DietPi:~# G_AGI wget
[ INFO ] APT installation for: wget, please wait...
[  OK  ] G_AGI wget
root@DietPi:~#

But I get the exact sane error.

Also tried.

root@DietPi:~# apt-get install wget
Reading package lists... Done
Building dependency tree
Reading state information... Done
wget is already the newest version (1.18-5+deb9u3).
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.

Also:

root@DietPi:~# time wget --spider http://ftp.debian.org/debian/
-bash: wget: command not found

real    0m0.001s
user    0m0.000s
sys     0m0.000s
root@DietPi:~#

MichaIng, 2019-07-07:
@soloza 
Ah good it is installed. Strange that the command is not found. Please try and paste:
```
which -a wget
hash -d
wget
```

soloza, 2019-07-07:
root@DietPi:~# which -a wget
root@DietPi:~# hash -d
hits    command
   3    /usr/bin/which
   3    /bin/rm
   1    /usr/bin/mesg
   6    /usr/bin/apt-get
   2    /usr/bin/clear
root@DietPi:~# wget
-bash: wget: command not found
root@DietPi:~#

soloza, 2019-07-07:
[image]

MichaIng, 2019-07-07:
@soloza 
Very strange, please paste: `echo $PATH`

In case for whatever reason the binary was removed (manually), so if the $PATH variable is correct, a reinstall might solve it: `G_AGI --reinstall wget`

soloza, 2019-07-07:
"G_AGI --reinstall wget" allowed me to get into dietpi-software again.

Thank you very much!

Saved me from re-installing the OS :)

MichaIng, 2019-07-07:
@soloza 
Whatever removed the binary, good that it works now.

I mark this issue as closed, feel free to reopen if required.

closed 2019-07-07 by MichaIng