MichaIng/DietPi #2966: dietpi-config does not change i2c_baudrate correctly

flashspys, 2019-07-07:
### Creating a bug report/issue

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=25
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'`
- Distro version | `stretch`
- Kernel version | `Linux link 4.19.42-v7+ #1219 SMP Tue May 14 21:20:58 BST 2019 armv7l GNU/Linux`
- SBC device | `RPi 3 Model B (armv7l)`
- Power supply used | 5V 2A
- SDcard used | Samsung 32GB

#### Steps to reproduce

1. Set the i2c_arm_baudrate via `dietpi-config` to e.g. 10kHz

#### Expected behaviour

- As described in e.g. [the official doc](https://www.raspberrypi.org/documentation/configuration/device-tree.md) or in various other sources, there should be an entry in config.txt like `dtparam=i2c_arm_baudrate=10000`

#### Actual behaviour

- The entry is wrong: `i2c_arm_baudrate=10000`. The `dtparam=` is missing. Because of this, dietpi-config does not change the baudrate effectively.

I changed the entry in /DietPi/config.txt manually and then the baudrate changed.

MichaIng, 2019-07-07:
@flashspys 
Many thanks for your report. I mark this for investigation.

MichaIng, 2019-07-07:
Resolved for v6.26: https://github.com/MichaIng/DietPi/pull/2967

flashspys, 2019-07-07:
I just switched to the dev branch, works as expected 👍

closed 2019-07-07 by flashspys

MichaIng, 2019-07-07:
@flashspys 
Many thanks for testing!