MichaIng/DietPi #2968: dietpi-banner broken after upgrade to v6.25.3

Akito13, 2019-07-07:
### Creating a bug report/issue

#### Required Information
- DietPi version: 6.25.3
- Distro version: stretch
- Kernel version: 4.19.42-v7+
- SBC device: RPi 3 Model B+ (armv7l)
- Power supply used: 5V, true 3A
- SDcard used: Transcend 400x Premium

#### Additional Information (if applicable)
- Software title: dietpi-banner

#### Steps to reproduce

Update to v6.25.3 according to the book.

#### Expected behaviour

dietpi-banner should be working.

#### Actual behaviour

```
 - LAN IP : 1.1.1.1 (eth0)
/tmp/.dietpi_motd: line 9: unexpected EOF while looking for matching `''
/tmp/.dietpi_motd: line 12: syntax error: unexpected end of file
 - MOTD :
```

#### Extra details

```
$ cat /etc/motd
The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
```
```
$ cat /tmp/.dietpi_motd
# MOTD
if (( $G_HW_MODEL < 9 )); then

        motd='WARNING: On Raspberry Pi the new available firmware upgrade, that adds support for RPi 4, uses nearly all space in /boot. Please check your /boot directory and in case remove junk from it.
          See this GitHub issue about how a clean /boot directory should look like (including new RPi 4 files): https://github.com/MichaIng/DietPi/issues/2935'

else

        motd='v6.25 release notes: https://t.co/A6qQuG9fN2?amp=1

fi
```

MichaIng, 2019-07-07:
@Akito13 
Many thanks for your report. Jep little syntax error in the MOTD, I just fixed it. Please run to resolve:
```
rm /tmp/.dietpi_motd
/DietPi/dietpi/func/dietpi-banner 1
```

Akito13, 2019-07-07:
@MichaIng Kind of works, but the spacing is terrible:
```
WARNING: On Raspberry Pi the new available firmware upgrade, that adds support for RPi 4, uses nearly all space in /boot. Please check your /boot directory and in case remove junk from it.
          See this GitHub issue about how a clean /boot directory should look like (including new RPi 4 files): https://github.com/MichaIng/DietPi/issues/2935
```
Seeing it in this comment doesn't make it look bad. Maybe it relates to the CLI-window size, as it looks very bad in it.

MichaIng, 2019-07-07:
@Akito13 
Yeah you are right, I will shorten the warning since all info is provided in the link already.

Okay should be better now. Also I removed the warning for users which already did the kernel upgrade. Rerun the above two commands to apply.
For others I will leave this warning for a while, since I think a full `/boot` partition, that in cases can be caused by the latest firmware upgrade, can cause failing boot.

Akito13, 2019-07-07:
> 
> 
> @Akito13
Also I removed the warning for users which already did the kernel upgrade. 

I think I missed something, because I already have my system fully upgraded and I still see this warning.

That said, the main `dietpi-banner` issue is resolved.

closed 2019-07-07 by Akito13

MichaIng, 2019-07-07:
@Akito13 
> I think I missed something, because I already have my system fully upgraded and I still see this warning.

Redo:
```
rm /tmp/.dietpi_motd
/DietPi/dietpi/func/dietpi-banner 1
```
And the warning still shows up?

It checks for the following file: `/boot/start4.elf`
This is present after the latest firmware upgrade on RPi.

Akito13, 2019-07-07:
> /DietPi/dietpi/func/dietpi-banner 1

I ran `apt upgrade`, applied the solution several times and the warning still shows up.
I neither have `/boot/start4.elf`, nor `bcm2711-rpi-4-b.dtb`, though.

MichaIng, 2019-07-07:
@Akito13 
Strange, could you paste: `ls -lAh /boot`

Akito13, 2019-07-07:
> 
> 
> @Akito13
>#### EDIT: Ah now I see the reason. The firmware upgrade is only available on the Buster repo, not on the Stretch repo (anymore).

I see now, I was confused by this, initially. So I guess this issue is solved now as well?

MichaIng, 2019-07-07:
@Akito13 
Jep then everything is solved. Good find btw, must have confused other users as well.