MichaIng/DietPi #2969: Failed to update DietPi from v6.24.1 to v6.25.3

notwork-2000, 2019-07-08:
#### Details:
- Date           | Mon  8 Jul 01:56:25 BST 2019
- Bug report     | N/A
- DietPi version | v6.24.1 (Fourdee/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B+ (armv7l) (index=3)
- Kernel version | #1219 SMP Tue May 14 21:20:58 BST 2019
- Distro         | stretch (index=4)
- Command        | G_AGUG
- Exit code      | 100
- Software title | DietPi-Update

#### Steps to reproduce:

1. dietpi-update

#### Expected behaviour:

- dietpi should update from v6.24.1 to v6.25.3 

#### Actual behaviour:

- update process fails, details below.

#### Extra details:

-  DietPi-Pre-patch
─────────────────────────────────────────────────────
 Mode: Applying pre-patches

[  OK  ] Successfully applied pre-patches

[ INFO ] DietPi-Update | APT update, please wait...
Get:1 http://apt.sonarr.tv develop InRelease [13.3 kB]
Get:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease [15.0 kB]
Ign:3 https://downloads.plex.tv/repo/deb public InRelease
Hit:4 https://downloads.plex.tv/repo/deb public Release
Ign:1 http://apt.sonarr.tv develop InRelease
Hit:5 https://download.mono-project.com/repo/debian raspbianstretch InRelease
Hit:6 https://archive.raspberrypi.org/debian stretch InRelease
Fetched 28.3 kB in 2s (12.0 kB/s)
Reading package lists...
W: GPG error: http://apt.sonarr.tv develop InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY EBFF6B99D9B78493
W: The repository 'http://apt.sonarr.tv develop InRelease' is not signed.
[  OK  ] DietPi-Update | G_AGUP
[ INFO ] DietPi-Update | APT upgrade, please wait...
WARNING: The following packages cannot be authenticated!
  nzbdrone
E: There were unauthenticated packages and -y was used without --allow-unauthenticated
[FAILED] DietPi-Update | G_AGUG

#### Additional logs:
```
Log file contents:
WARNING: The following packages cannot be authenticated!
  nzbdrone
E: There were unauthenticated packages and -y was used without --allow-unauthenticated
```

MichaIng, 2019-07-08:
@notwork-2000 
Many thanks for your report. Looks similar to: https://dietpi.com/phpbb/viewtopic.php?f=11&t=6082
Not sure if it is related to the Debian Buster release yesterday (06/07/2019) where the APT repo suites changed (Buster: testing => stable and Stretch: stable => oldstable). Perhaps repo keys got reset because of this.

Please try:
```
apt-key adv --keyserver keyserver.ubuntu.com --recv-keys FDA5DFFC
apt update
```

notwork-2000, 2019-07-08:
Hello, thanks for the lightning fast response.  Huge fan of dietpi!  Thanks!!!
I tried looking for a similar issue but didn't run across that one, my apologies.

The apt-key command (ran as root) failed with the following output:

apt-key adv --keyserver keyserver.ubuntu.com --recv-keys FDA5DFFC

Executing: /tmp/apt-key-gpghome.fIfuCrYbDZ/gpg.1.sh --keyserver keyserver.ubuntu.com --recv-keys FDA5DFFC
gpg: keyserver receive failed: Server indicated a failure

apt update

Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Get:2 http://apt.sonarr.tv develop InRelease [13.3 kB]
Ign:3 https://downloads.plex.tv/repo/deb public InRelease
Hit:4 https://downloads.plex.tv/repo/deb public Release
Hit:5 https://download.mono-project.com/repo/debian raspbianstretch InRelease
Err:2 http://apt.sonarr.tv develop InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY EBFF6B99D9B78493
Hit:6 https://archive.raspberrypi.org/debian stretch InRelease
Reading package lists... Done
W: GPG error: http://apt.sonarr.tv develop InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY EBFF6B99D9B78493
E: The repository 'http://apt.sonarr.tv develop InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

notwork-2000, 2019-07-08:
FYI, with a -v flag:

root@DietPi:~# apt-key adv -v --keyserver keyserver.ubuntu.com --recv-keys FDA5DFFC
Executing: /tmp/apt-key-gpghome.zbHt7qEOsg/gpg.1.sh -v --keyserver keyserver.ubuntu.com --recv-keys FDA5DFFC
gpg: no running Dirmngr - starting '/usr/bin/dirmngr'
gpg: waiting for the dirmngr to come up ... (5s)
gpg: connection to the dirmngr established
gpg: keyserver receive failed: Server indicated a failure

MichaIng, 2019-07-08:
@notwork-2000 
Strange, works well here:
```
root@VM-Stretch:~# apt-key adv --keyserver keyserver.ubuntu.com --recv-keys FDA5DFFC
Executing: /tmp/apt-key-gpghome.88GZB5vOca/gpg.1.sh --keyserver keyserver.ubuntu.com --recv-keys FDA5DFFC
gpg: key 249A1A0DFDA5DFFC: public key "Totally Legit Signing Key <mallory@example.org>" imported
gpg: key EBFF6B99D9B78493: public key "NzbDrone <contact@nzbdrone.com>" imported
gpg: Total number processed: 2
gpg:               imported: 2
```

> Server indicated a failure

Perhaps a temporary server issue, please try again. Also you can test if the server is reachable in general for you and how long it takes: `time curl -IL keyserver.ubuntu.com`

MichaIng, 2019-07-08:
@notwork-2000 
Just checked an indeed the APT repo key has changed. Not sure why in my case the old still worked but please try instead:
```
apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 0xA236C58F409091A18ACA53CBEBFF6B99D9B78493
```

notwork-2000, 2019-07-08:
Must be something with my dns or firewall setup.  Acted the same until I changed the keyserver url:

```
apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys FDA5DFFC
Executing: /tmp/apt-key-gpghome.Z72fVgWvDN/gpg.1.sh --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys FDA5DFFC
gpg: key 249A1A0DFDA5DFFC: public key "Totally Legit Signing Key <mallory@example.org>" imported
gpg: key EBFF6B99D9B78493: public key "NzbDrone <contact@nzbdrone.com>" imported
gpg: Total number processed: 2
gpg:               imported: 2

apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys 0xA236C58F409091A18ACA53CBEBFF6B99D9B78493
Executing: /tmp/apt-key-gpghome.e16nvBlB3D/gpg.1.sh --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys 0xA236C58F409091A18ACA53CBEBFF6B99D9B78493
gpg: key EBFF6B99D9B78493: "NzbDrone <contact@nzbdrone.com>" not changed
gpg: Total number processed: 1
gpg:              unchanged: 1
```
Now dietpi-update is happy and proceeds.  Thanks again for your help!!!

closed 2019-07-08 by notwork-2000

MichaIng, 2019-07-08:
@notwork-2000 
Okay great. I updated dietpi-software to use the safer long version key, however this obviously was not the issue.