MichaIng/DietPi #2970: v6.25.3 raspbian repo changed 'Suite' from testing to 'stable'

wozeparrot, 2019-07-08:
#### Details:
- Date           | Mon  8 Jul 02:33:49 BST 2019
- Bug report     | N/A
- DietPi version | v6.25.3 (MichaIng/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 2 Model B (armv7l) (index=2)
- Kernel version | #896 SMP Thu Jun 20 16:11:44 BST 2019
- Distro         | buster (index=5)
- Command        | G_AGUP
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
Get:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Hit:2 https://archive.raspberrypi.org/debian buster InRelease
Reading package lists...
E: Repository 'http://raspbian.raspberrypi.org/raspbian buster InRelease' changed its 'Suite' value from 'testing' to 'stable'
```

MichaIng, 2019-07-08:
@wozeparrot 
Jep this is due to official Debian Buster release on 06/07 (yesterday). I also faced this. You need to run `apt-get update` interactively once and accept this suite change.

wozeparrot, 2019-07-08:
How would I do that from an install?

MichaIng, 2019-07-08:
@wozeparrot 
As said, run the following command once from command line/terminal and confirm if being asked:
```
apt-get update
```

wozeparrot, 2019-07-08:
Just an update, running `apt-get update` does not help. You need to run `apt update`.

MichaIng, 2019-07-08:
@wozeparrot 
Ah `apt-get update` didn't ask for confirming the suite change? Can't remember currently which one I called, apt update should be just a wrapper for apt-get update with prints slightly more output, e.g. how much packages can be updated.

wozeparrot, 2019-07-08:
No, apt-get update is not interactive, while apt update is.

MichaIng, 2019-07-08:
`apt-get update` is interactive as well, actually, but perhaps the suite accepting prompt is just part of the `apt update` wrapper. Thanks for this info!

closed 2019-07-08 by MichaIng

Trunkzeh, 2019-07-09:
For clarity; for those running the experimental Buster image (e.g. the raspberry dietpi buster img) is when you start getting this error is to open a second ssh session and run apt update and then accept the suite change. Then you can close this session and resume the dietpi install :-)

MichaIng, 2019-07-10:
@Trunkzeh 
Good point. I will also create new Buster images. VirtualBox is already done and Buster the new default/stable.