MichaIng/DietPi #2971: [v6.25.3] Webmin - Failed to fetch apt-show-version_0.22.7_all.deb

tindecken, 2019-07-08:
#### Details:
- Date           | Sun  7 Jul 20:51:54 -07 2019
- Bug report     | N/A
- DietPi version | v6.25.3 (MichaIng/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B+ (armv7l) (index=3)
- Kernel version | #1219 SMP Tue May 14 21:20:58 BST 2019
- Distro         | stretch (index=4)
- Command        | G_AGF
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
E: Failed to fetch https://mirror.freedif.org/Raspbian/raspbian/pool/main/a/apt-show-versions/apt-show-versions_0.22.7_all.deb  Operation timed out after 120955 milliseconds with 0 out of 0 bytes received
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

[image]

MichaIng, 2019-07-08:
@tindecken 
> Operation timed out after 120955 milliseconds

Hmm, could you try it manually: `apt-get -f install`

tindecken, 2019-07-08:
I tried successfully 👍 
```
root@Tindecken:~# sudo apt-get -f install
Reading package lists... Done
Building dependency tree
Reading state information... Done
Correcting dependencies... Done
The following additional packages will be installed:
  apt-show-versions libapt-pkg-perl libauthen-pam-perl libio-pty-perl libnet-ssleay-perl perl-openssl-defaults
The following NEW packages will be installed:
  apt-show-versions libapt-pkg-perl libauthen-pam-perl libio-pty-perl libnet-ssleay-perl perl-openssl-defaults
0 upgraded, 6 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
Need to get 435 kB of archives.
After this operation, 1,662 kB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Get:1 https://mirror.freedif.org/Raspbian/raspbian stretch/main armhf perl-openssl-defaults armhf 3 [6,782 B]
Get:2 https://mirror.freedif.org/Raspbian/raspbian stretch/main armhf libnet-ssleay-perl armhf 1.80-1 [270 kB]
Get:3 https://mirror.freedif.org/Raspbian/raspbian stretch/main armhf libauthen-pam-perl armhf 0.16-3+b3 [25.9 kB]
Get:4 https://mirror.freedif.org/Raspbian/raspbian stretch/main armhf libio-pty-perl armhf 1:1.08-1.1+b2 [33.0 kB]
Get:5 https://mirror.freedif.org/Raspbian/raspbian stretch/main armhf libapt-pkg-perl armhf 0.1.32 [64.8 kB]
Get:6 https://mirror.freedif.org/Raspbian/raspbian stretch/main armhf apt-show-versions all 0.22.7 [34.3 kB]
Fetched 435 kB in 4s (94.9 kB/s)
Selecting previously unselected package perl-openssl-defaults:armhf.
(Reading database ... 57893 files and directories currently installed.)
Preparing to unpack .../0-perl-openssl-defaults_3_armhf.deb ...
Unpacking perl-openssl-defaults:armhf (3) ...
Selecting previously unselected package libnet-ssleay-perl.
Preparing to unpack .../1-libnet-ssleay-perl_1.80-1_armhf.deb ...
Unpacking libnet-ssleay-perl (1.80-1) ...
Selecting previously unselected package libauthen-pam-perl.
Preparing to unpack .../2-libauthen-pam-perl_0.16-3+b3_armhf.deb ...
Unpacking libauthen-pam-perl (0.16-3+b3) ...
Selecting previously unselected package libio-pty-perl.
Preparing to unpack .../3-libio-pty-perl_1%3a1.08-1.1+b2_armhf.deb ...
Unpacking libio-pty-perl (1:1.08-1.1+b2) ...
Selecting previously unselected package libapt-pkg-perl.
Preparing to unpack .../4-libapt-pkg-perl_0.1.32_armhf.deb ...
Unpacking libapt-pkg-perl (0.1.32) ...
Selecting previously unselected package apt-show-versions.
Preparing to unpack .../5-apt-show-versions_0.22.7_all.deb ...
Unpacking apt-show-versions (0.22.7) ...
Setting up libauthen-pam-perl (0.16-3+b3) ...
Setting up libapt-pkg-perl (0.1.32) ...
Setting up apt-show-versions (0.22.7) ...
** initializing cache. This may take a while **
Setting up libio-pty-perl (1:1.08-1.1+b2) ...
Setting up perl-openssl-defaults:armhf (3) ...
Setting up libnet-ssleay-perl (1.80-1) ...
Setting up webmin (1.920) ...
Webmin install complete. You can now login to https://Tindecken:10000/
as root with your root password, or as any user who can use sudo
to run commands as root.
Processing triggers for systemd (232-25+deb9u11) ...
root@Tindecken:~#
```

MichaIng, 2019-07-08:
@tindecken 
Okay great. Perhaps the first attempt failed due to some temporary APT mirror issue related to the Debian Buster release.

Rerun dietpi-software to apply all desired steps: `dietpi-software install 115`

I mark this as closed then. Feel free to reopen if required.

closed 2019-07-08 by MichaIng