MichaIng/DietPi #2973: APT | Update/install issues due to Debian Buster release

MichaIng, 2019-07-08:
#### Debian Buster has been officially released on 06/07/2019: https://www.debian.org/News/2019/20190706
- **Congratulations** to the Debian team and the mostly smooth usage and transition, at least during my tests.
- This allows us to move our Buster images from experimental to stable.

#### APT update issues due to the repository suite transition
- To compare: https://cdn-aws.deb.debian.org/debian/dists/
  - Jessie has been moved from `oldstable` to `oldoldstable`.
  - Stretch has been moved from `stable` to `oldstable`.
  - Buster has been moved from `testing` to `stable`.
  - And `testing` now links to the future Debian release, version 11, code name `bullseye`.
- At least on Buster it was in cases required to accept the above suite change once interactively before dietpi-update/software was able to call APT update interactively: https://github.com/MichaIng/DietPi/issues/2970
  - Run `apt-get update` and accepts the suite change.

#### Lost 3rd party repository keys
- It was observed that APT keys were lost for 3rd party repository. You might need to reapply them, before `apt-get update` runs through without error:
  - PHP (**on Jessie and Stretch only!**): `curl -sSL 'https://packages.sury.org/php/apt.gpg' | apt-key add -`
  - Mopidy: `curl -sSL 'https://apt.mopidy.com/mopidy.gpg' | apt-key add -`
  - InfluxDB: `curl -sSL 'https://repos.influxdata.com/influxdb.key' | apt-key add -`
  - Grafana: `curl -sSL 'https://packages.grafana.com/gpg.key' | apt-key add -`
  - Plex: `curl -sSL 'https://downloads.plex.tv/plex-keys/PlexSign.key' | apt-key add -`
  - CloudPrint: `curl -sSL 'https://davesteele.github.io/key-366150CE.pub.txt' | apt-key add -`
  - Sonarr: `apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 0xA236C58F409091A18ACA53CBEBFF6B99D9B78493`
  - Raspotify: `curl -sSL 'https://dtcooper.github.io/raspotify/key.asc' | apt-key add -`
  - Mono: `apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys 3FA7E0328081BFF6A14DA29AA6A19B38D3D831EF`

#### Temporary transition issues
- Shortly after release I experienced some APT update issues that could be finally solved by running the following 2-3 times. I guess it was a temporary server or cache issue or occurred while APT mirror synchronisation. It should not occur anymore but I list it, just in case: `apt-get clean; apt-get update`

symbios24, 2019-07-14:
it still does occur first run update failed it prompts for a retry or to choose another update mirror
will you include the keys to the next release?

MichaIng, 2019-07-14:
@symbios24 
You mean the RPi4 testing image right? Yeah this created with beta code that has this update error and was created before official Debian Buster release, thus suffers the above mentioned APT suite change issues (testing => stable).

I will recreate the RPi Buster image later this day. Also to include the recent firmware update which most likely solves some video related issues.

To update DietPi, run: `dietpi-update 1`
Ref: #2960

symbios24, 2019-07-14:
for now i had tried it only i a Rpi 3B 
ok thank you

MichaIng, 2019-07-14:
@symbios24 
The issues are the same, regardless of RPi model.

MichaIng, 2019-07-14:
@symbios24 
I just uploaded an updated RPi Buster image. The dietpi-update bug is solved by this and hopefully some firmware related bugs since it includes an RPi firmware upgrade as well.

matteos1, 2019-07-23:
how do I update diet-pi to the latest buster version?

MichaIng, 2019-07-24:
@matteos1 
Distro upgrading is generally not supported our side. The risk of issues, especially follow-up errors that one does not recognise first, is simply too high.

However you could try it on your own risk:
```
sed -i 's/stretch/buster/g' /etc/apt/sources.list
sed -i 's/stretch/buster/g' /etc/apt/sources.list.d/raspi.list # Assuming this is RPi
G_AGUP
apt upgrade
apt full-upgrade
```

But the cleaner and safer approach is to freshly flash the Buster image and copy back required data and settings manually from a `dietpi-backup` if the initial system.

MichaIng, 2019-08-02:
Okay I mark this issue as closed now, I guess all Buster images have been gone through the suite transision.
I found the apt-get command line option `--allow-releaseinfo-change` which allows to accept this non-interactively, however this is now something we can remember for in two years for Debian Bullseye release :wink:.

closed 2019-08-02 by MichaIng