MichaIng/DietPi #2974: Can someone help me? The system is installed and the alsa driver is installed, but the sound card is still unrecognizable

hifitime, 2019-07-08:
### Creating a bug report/issue
Can someone help me? The system is installed and the alsa driver is installed, but the sound card is still unrecognizable

Thanks for any help
==============================
root@DietPi:~# lsusb
Bus 001 Device 002: ID 16d0:071a MCS
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
================================================
root@DietPi:~# aplay -l
aplay: device_list:270: no soundcards found...
=============================================
root@DietPi:~# alsactl -v
alsactl version 1.1.3
==================================================

root@DietPi:~# alsamixer
cannot open mixer: No such file or directory
### ### ### ### ### ==============================================

root@DietPi:~# dietpi-services restart
[  OK  ] DietPi-Services | Root access verified.

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: restart

[  OK  ] DietPi-Services | restart : avahi-daemon
[  OK  ] DietPi-Services | restart : shairport-sync
[  OK  ] DietPi-Services | restart : squeezelite
[  OK  ] DietPi-Services | restart : gmrender
[  OK  ] DietPi-Services | restart : logitechmediaserver
[  OK  ] DietPi-Services | restart : cron

=================================================

#### Required Information
- DietPi version | DietPi v6.25.3 : 13:46 - Mon 07/08/19
- Distro version | 9.9
- Kernel version | Linux DietPi 4.19.56-mvebu64 #5.89 SMP PREEMPT Tue Jun 25 22:27:47 CEST 2019 aarch64 GNU/Linux
- SBC device | Generic Device (aarch64)

MichaIng, 2019-07-08:
@hifitime 
Many thanks for your report.

Most likely a driver or even a dtoverlay or kernel module is required for your this sound card and your SBC. Which SBC are you using actually?
Check for install instructions on Linux/Debian from the sound card manufacturer.

hifitime, 2019-07-09:
hi

 thanks for your Reply.
my  usb sound card is Amanero Combo384 USB

hifitime, 2019-07-09:
-
 Amanero Combo384 USB.Note，This is the official statement:

NO Drivers needed for Mac OSX 10.6+ and Linux with UAC2 compliant kernel.KS/Wasapi/WDM/ASIO Drivers for Win7,Win8,Win10 32-64bit included at no fee.

hifitime, 2019-07-09:
Whether it's installing native debian, armbian, or DietPi, it's the same problem.

hifitime, 2019-07-09:
Is it possible that it's a SBC-driven problem?

hifitime, 2019-07-09:
I installed Debian first and then ran PREP_SYSTEM_FOR_DIETPI.sh to install it.

MichaIng, 2019-07-09:
@hifitime 
Please for the future edit your comments to add further info instead of creating new ones to keep it a bid clean 😉.

As you said, it is Linux kernel related and this means it depends on the actual machine/SBC you are trying this on. So which SBC is it?

hifitime, 2019-07-09:
SBC is Clearfog base
https://developer.solid-run.com/products/clearfog-base/

MichaIng, 2019-07-09:
@hifitime 
Okay in this case this is something you need to at best ask both, the sound card manufacturer and the SBC manufacturer.

What I would try, to have a wider kernel support, is a raw Buster install, see "Upstream (Buster)" section: https://developer.solid-run.com/knowledge-base/a388-debian/#official-solidrun-images
As there are quire some steps, preparing the installer media and install u-boot manually, you could also try to upgrade the system. But at best do a full SDcard backup first, then e.g.:
```
dietpi-services stop
sed -i 's/stretch/buster/g' /etc/apt/sources.list{,.d/*}
apt update
apt upgrade
apt dist-upgrade
reboot
```

hifitime, 2019-07-09:
Okay, I'll try again.

hifitime, 2019-07-10:
> @hifitime
> Okay in this case this is something you need to at best ask both, the sound card manufacturer and the SBC manufacturer.
> 
> Are there any kernel packages installed or even a 3rd party APT repo for firmware?
> 
> ```
> dpkg -S /lib/modules
> ```
> 
> What I would try, to have a wider kernel support, is a raw Buster install, see "Upstream (Buster)" section: https://developer.solid-run.com/knowledge-base/a388-debian/#official-solidrun-images
> As there are quire some steps, preparing the installer media and install u-boot manually, you could also try to upgrade the system. But at best do a full SDcard backup first, then e.g.:
> 
> ```
> dietpi-services stop
> sed -i 's/stretch/buster/g' /etc/apt/sources.list{,.d/*}
> apt update
> apt upgrade
> apt full-upgrade
> reboot
> ```

Problems remain after the upgrade
Is it possible that the uac2 module of Linux kernel was not compiled?

MichaIng, 2019-07-10:
> Is it possible that the uac2 module of Linux kernel was not compiled?

Most likely. This is why I asked for `dpkg -S /lib/modules` as this should show if there is any e.g. manufacturer DPKG package responsible to kernel/bootloader.

This is the driver that is required for UAC2: https://github.com/torvalds/linux/blob/master/drivers/usb/gadget/function/f_uac2.c
Respectively: `linux/usb/audio-v2.h`
If your SBCs manufacturer has any kernel/firmware source available you could check if that ones are part of it. Or at best ask via their forums/support, perhaps it's implemented as kernel module, which you would then find inside `/lib/modules`.

MichaIng, 2019-10-09:
I mark this issue as closed now. Feel free to reopen if required.

closed 2019-10-09 by MichaIng