MichaIng/DietPi #2975: DietPi-Software | Node-RED: missing directory

Orfait, 2019-07-09:
#### Information
- DietPi version : 
`G_DIETPI_VERSION_CORE=6`
`G_DIETPI_VERSION_SUB=25`
`G_DIETPI_VERSION_RC=3`
`G_GITBRANCH='master'`
`G_GITOWNER='MichaIng'`
- Distro version : `stretch`
- Kernel version : `Linux DietPi 4.19.42-v7+ #1219 SMP Tue May 14 21:20:58 BST 2019 armv7l GNU/Linux`
- SBC device : `RPi 3 Model B (armv7l)`

#### Additional Information
- Software title : Node-RED
- DietPi freshly installed, so also fresh install of Node-RED

#### Steps to reproduce
1. Install Node-RED
2. Start service : `systemctl start node-red.service`

#### Expected behaviour
Service should be running

#### Actual behaviour
Service ends with error related to directory

#### Extra details
If we take a look at `/etc/systemd/system/node-red.service` :
```[Unit]
Description=Node-RED (DietPi)

[Service]
User=nodered
ExecStart=/usr/local/bin/node-red -u /mnt/dietpi_userdata/node-red

[Install]
WantedBy=multi-user.target
```
1. The folder `/mnt/dietpi_userdata/node-red` is missing, create it :
`mkdir /mnt/dietpi_userdata/node-red`
2. Give `nodered` user permission to this directory :
```chmod ug+rwx -R  /mnt/dietpi_userdata/node-red
chown node-red:node-red -R  /mnt/dietpi_userdata/node-red
```

MichaIng, 2019-07-09:
@Orfait 
Many thanks for your report.

Confirmed. Bug introduced with v6.25 in relation with some user creation alignments.

Fixed with: https://github.com/MichaIng/DietPi/commit/2924d0e136bbdabd4e2c9d2936410a71e5760ea9
Permissions are done here: https://github.com/MichaIng/DietPi/blob/master/dietpi/func/dietpi-set_software#L473
Changelog: https://github.com/MichaIng/DietPi/commit/91d76d4a9170b35a50129e907a657b37414611e6

I will also add a patch to the v6.25 patch file to prevent future updaters from running into this.

marcobrianza, 2019-07-22:
hello, stating from a fresh install the user nodered was also missing, created manually and with the above code it fixes.

MichaIng, 2019-07-22:
@marcobrianza 
Thanks for reporting. Found the issue:
- If the group "gpio" does not exist, the whole useradd/usermod command fails. This seems to be the case on non SBCs.
- Can you verify that you tried it on some x86_64 machine?

marcobrianza, 2019-07-22:
yes, it was on virtualbox

MichaIng, 2019-07-22:
@marcobrianza 
Fixed with: https://github.com/MichaIng/DietPi/commit/cc65890a1f7b723187559dce68e16bf3ec3de66f
- Along with some other failsafe coding where we create or modify run users.

Changelog: https://github.com/MichaIng/DietPi/commit/d7604eab07a9f85aebf436df6633965ab4ba31d4

marcobrianza, 2019-07-23:
Great quick fix! thank you
I have switched to dev branch (6.26.0) and can confirm that it works.

On the virtual machine I had to modify the file /DietPi/dietpi.txt so I think that instructions at https://github.com/MichaIng/DietPi/blob/master/BRANCH_SYSTEM.md should be updated since modifying /boot/dietpi.txt did not work for me.

MichaIng, 2019-07-23:
@marcobrianza 
Great, many thanks for testing.

> should be updated since modifying /boot/dietpi.txt did not work for me.

Note:  "`Beta on Fresh image:`", so before booting the first time, it's `dietpi.txt` from the boot partition or dir.
For "`Beta on an existing installation:`" it states
```
G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=beta' /DietPi/dietpi.txt
```
for beta, respectively "dev" for dev branch 😉. Manually editing the RAMdisk location `/DietPi/dietpi.txt` has the same result.

closed 2019-10-15 by MichaIng

Phil1988, 2020-01-08:
Sorry for the late feedback.
I just wanted to report, that this fixes this issue :)