MichaIng/DietPi #2976: Service "pihole-FTL" keeps crashing/stopping; both as HW on RPi 2 Mod B+ & as VM

kuerious, 2019-07-10:
### Creating a bug report/issue

#### Required Information
- DietPi version | CORE=**6**, SUB=**25**, RC=**3**, BRANCH=**master**
- Distro version | **9.9**
- Kernel version | Linux <hostname> 4.9.0-8-amd64 #1 SMP Debian 4.9.144-3.1 (2019-02-19) x86_64 GNU/Linux
- SBC device | Have tried with **RPi Mod B+** and with **Virtual Machine x86_64**
- Power supply used | HW used **2.5A official CanaKit PSU**
- SDcard used | HW used **8GB SanDisk Class10**

#### Additional Information (if applicable)
- Software title | **Pi-hole**
- Was the software title installed freshly or updated/migrated? | **Both**
- Can this issue be replicated on a fresh installation of DietPi? | **Yes, as was done on a VM.**
- Bug report ID | **01c2316b-fc4e-4245-8978-ff8ad4ee13dc**
- Also sent Pi-hole debug report

#### Steps to reproduce

1. Use DietPi on HW w/ 8GB microSD + "Pi-hole" install from "Software - Optimized",  unmodified. Run DoH via [original docs](https://docs.pi-hole.net/guides/dns-over-https/), then with steps listed [here](https://www.reddit.com/r/pihole/comments/aoezvx/how_to_install_pihole_with_dns_over_https_cloud/).
2. Use DietPi via VM w/ 4x Core, 4GB RAM, 8GB store + "Pi-hole" install from "Software - Optimized",  unmodified. Run DoH via [original docs](https://docs.pi-hole.net/guides/dns-over-https/), then with steps listed [here](https://www.reddit.com/r/pihole/comments/aoezvx/how_to_install_pihole_with_dns_over_https_cloud/).

#### Expected behaviour

- Normal, uninterrupted services running. DNS resolution functions as specified.

#### Actual behaviour

- Service **pihole-FTL** crashes 1x to even 5x daily. A restart of the service restores function until next instance. Nothing apparent in log files.

#### Extra details

- Cross-posted at Pi-hole GitHub as [Issue# 614](https://github.com/pi-hole/FTL/issues/614).

kuerious, 2019-07-10:
And another thing ... no errors in the log for service **pihole-FTL**:

https://pastebin.com/Qg0cejwR

MichaIng, 2019-07-10:
@kuerious 
Many thanks for your report.

I remember another case where Pi-hole crashed daily when DietPi-Logclear faces a filled /var/log tmpfs and clears it. We have a special handling for Pi-hole log files but perhaps due to some changes their side those do not work anymore.

Basically the issue is/was that `pihole-FTL` doesn't like the log files to be cleared while running. But this is required when having a RAMlog of course. So we restart the service in a special way that worked in the past. I will run some tests to figure out if any other method is required.

kuerious, 2019-07-11:
As [per your comment](https://github.com/pi-hole/FTL/issues/614#issuecomment-509997573) in the [Pi-hole issue #614](https://github.com/pi-hole/FTL/issues/614), I will close this as being responded to & linked from there.

closed 2019-07-11 by kuerious