MichaIng/DietPi #2977: Asus Tinkerboard install failing

Magno99900, 2019-07-10:
------------------------------------------------------------------
- DietPi has encounted an error, and, is unable to continue        -
- Please create a ticket: https://github.com/Fourdee/DietPi/issues -
- Copy and paste the BLUE lines below, into the ticket             -
--------------------------------------------------------------------

#### Details:
- Date           | Wed 10 Jul 05:36:14 BST 2019
- Bug report     | N/A
- DietPi version | v6.18.14 (Fourdee/master)
- Img creator    | DietPi Core Team
- Pre-image      | ARMbian
- SBC device     | Asus Tinker Board (armv7l) (index=52)
- Kernel version | #270 SMP PREEMPT Wed Sep 19 11:40:40 CEST 2018
- Distro         | stretch (index=4)
- Command        | G_AGUG
- Exit code      | 100
- Software title | DietPi-Update

#### Steps to reproduce:

1. trying to install latest version of dietpi from download page onto known working asus tinkerboard.
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- Tried multiple SD cards and tried re downloading the file

#### Additional logs:
```
Log file contents:
Setting up apt-utils (1.4.9) ...
Setting up tzdata (2019a-0+deb9u1) ...

Current default time zone: 'Europe/London'
Local time is now:      Wed Jul 10 05:35:37 BST 2019.
Universal Time is now:  Wed Jul 10 04:35:37 UTC 2019.
Run 'dpkg-reconfigure tzdata' if you wish to change it.

Setting up systemd-sysv (232-25+deb9u11) ...
Setting up linux-u-boot-tinkerboard-next (5.90) ...
Setting up unzip (6.0-21+deb9u1) ...
Setting up armbian-firmware (5.90) ...
Setting up libssl1.0.2:armhf (1.0.2s-1~deb9u1) ...
Setting up libssh2-1:armhf (1.7.0-1+deb9u1) ...
Processing triggers for libc-bin (2.24-11+deb9u4) ...
Setting up udev (232-25+deb9u11) ...
addgroup: The group `input' already exists as a system group. Exiting.
update-initramfs: deferring update (trigger activated)
Setting up libssl1.1:armhf (1.1.0k-1~deb9u1) ...
Setting up libc-l10n (2.24-11+deb9u4) ...
Processing triggers for systemd (232-25+deb9u11) ...
Setting up openssl (1.1.0k-1~deb9u1) ...
Setting up wget (1.18-5+deb9u3) ...
Setting up htop (2.1.2-3) ...
Setting up gnupg-agent (2.1.18-8~deb9u4) ...
Setting up dirmngr (2.1.18-8~deb9u4) ...
Setting up armbian-tools-stretch (5.75) ...
Setting up linux-dtb-next-rockchip (5.90) ...
Setting up locales (2.24-11+deb9u4) ...
Generating locales (this might take a while)...
  en_GB.UTF-8... done
[... 188 lines cut ...]
Reading state information...
Calculating upgrade...
The following packages have been kept back:
  linux-stretch-root-next-tinkerboard
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up initramfs-tools (0.130) ...
update-initramfs: deferring update (trigger activated)
Processing triggers for initramfs-tools (0.130) ...
update-initramfs: Generating /boot/initrd.img-4.19.57-rockchip
/etc/initramfs/post-update.d//99-uboot: 3: .: Can't open /etc/armbian-release
run-parts: /etc/initramfs/post-update.d//99-uboot exited with return code 2
dpkg: error processing package initramfs-tools (--configure):
 subprocess installed post-installation script returned error exit status 1
Errors were encountered while processing:
 initramfs-tools
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

--------------------------------------------------------------------

[FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.

root@DietPi:~#

MichaIng, 2019-07-10:
@Magno99900 
Many thanks for your report.

Please do the following:
```
cat << _EOF_ > /etc/armbian-release
# PLEASE DO NOT EDIT THIS FILE
BOARD=tinkerboard
BOARD_NAME="Tinkerboard"
BOARDFAMILY=rockchip
BUILD_REPOSITORY_URL=https://github.com/armbian/rkbin
BUILD_REPOSITORY_COMMIT=51d55a4
VERSION=5.90
LINUXFAMILY=rockchip
BRANCH=default
ARCH=arm
IMAGE_TYPE=stable
BOARD_TYPE=conf
INITRD_ARCH=arm
KERNEL_IMAGE_TYPE=zImage
_EOF_
apt-get -y upgrade
```

We need to recreate the ATB image.

Fourdee, 2019-08-04:
@MichaIng 

Buster image uploaded:
https://dietpi.com/downloads/images/DietPi_ASUSTB-ARMv7-Buster.7z

I've ran a basic install test, seems fine. GPU drivers will probably not transition over from Stretch, needs testing when I/comminuty get time.

ToDo:
- Replace dietpi.com with new buster image link?

Fourdee, 2019-08-04:
- :u5408: GPU accel
- Breaks all X11/desktop env's.
- :u5408:  ```rm /etc/X11/xorg.conf```
- Packages appear to break APT. We need to remove GPU driver install from our code.

MichaIng, 2019-08-04:
@Fourdee 
Many thanks for creating this. Indeed our GPU driver packages are Stretch versions. I made and totally forgot some updated archive, however it provides Stretch versions as well: https://dietpi.com/downloads/binaries/asus/testing/

There are GPU drivers for t76x available in Debian Buster repo: [libegl1](https://packages.debian.org/buster/libegl1)
Not 100% sure which package is pulled by default by [libegl1-mesa](https://packages.debian.org/buster/libegl1-mesa), which we install with X.org, but I guess best would be: [mali-t76x-fbdev-driver](https://packages.debian.org/buster/mali-t76x-fbdev-driver)
Simple `xorg.conf` to start with:
```
cat << _EOF_ > /etc/X11/xorg.conf
Section "Device"
	Identifier	"Mali-Fbdev"
	Driver		"fbdev"
	Option		"fbdev"		"/dev/fb0"
EndSection
_EOF_
```

Sadly https://tinkerboarding.co.uk is down to get resources from.

Found an TinkerOS fork on Debian Buster basis that has Kodi pre-installed: https://sourceforge.net/projects/tinkerboard/
Will check for working drivers/configs contained.

MichaIng, 2019-09-02:
I mark this as closed. GPU driver tests will be done in a separate issue.

closed 2019-09-02 by MichaIng