MichaIng/DietPi #2978: Odroid C1 eth0 recognized as eth1 soft V159

billgatos, 2019-07-10:
Hello all, specially Daniel. Tremendous good job on this.

So a few years ago, we talked alot because my odroid c1, only detected the eth as eth1 and not eth0 as usual. At that time you made a specific update to solve the issue.
So my question is can just use the updated version 6.22 on my C1 or can´t go beyond the v69 with update at that time.

Best regards.

### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version | `uname -a`
- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `sed -n 5p /DietPi/dietpi/.hw_model`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...

MichaIng, 2019-07-10:
@billgatos 
Many thanks for your question.

Jep you can install/upgrade to the current DietPi version, it auto-detects the correct/connected network interface automatically.

Note that Odroid C1 support is limited now. The official Debian Stretch based image does not have HDMI/local terminal support, so SSH/serial console only. The Jessie image (available here: https://dietpi.com/downloads/images/legacy/) has now reached the last supported DietPi version (v6.25), so no new updates will be available anymore.

MichaIng, 2019-09-02:
I mark this issue as closed. Feel free to reopen if required.

closed 2019-09-02 by MichaIng