MichaIng/DietPi #2980: DietPi-Software | Mycroft AI setup Voice Matrix

LexiconCode, 2019-07-10:
Mycroft AI set up use to include setting up the audio device. I have a [Voice Matrix](https://www.matrix.one/products/voice) add on board that used to be set up through Mycroft scripts. What would be your suggestion to proceed trying to set up Voice Matrix board with DietPi?

MichaIng, 2019-07-11:
@LexiconCode
Which Mycroft script in particular sets up the sudo device? Didn't know that.

About the Voice Matrix I found this install instructions: https://matrix-io.github.io/matrix-documentation/matrix-core/getting-started/core-installation/
As a promoted setup it should work with RPi. I can add this to DietPi-Config as well, as I anyway plan to rewrite the audio options with next version.

LexiconCode, 2019-07-15:
I'll take a look and event and also review the rewrite the audio options with next version. Unless you want to keep it open I consider this issue closed.

MichaIng, 2019-07-26:
@LexiconCode 
The following should work:
```
curl -sL https://apt.matrix.one/doc/apt-key.gpg | apt-key add -
echo "deb https://apt.matrix.one/raspbian $G_DISTRO_NAME main" > /etc/apt/sources.list.d/matrixlabs.list
G_AGUP
G_AGUG # Assures current kernel is installed
G_AGI matrixio-kernel-modules
reboot
```
Now `aplay -l` should list an entry for the MATRIX Voice, I guess it's `card 1`.
In `/etc/asound.conf` replacing `card 0` with `card 1` and in case `0,0` with `1,0` might work. But honestly I don't know about all the magic that you can do with this the MATRIX Voice, might might require to use different sound card or device 😉.
`aplay -l` at least should give a hint and when the related card has been added to asound.conf, `alsamixer` to tune levels and/or toggle some switches.

LexiconCode, 2019-07-28:
G_AGI matrixio-kernel-modules
                    │  - Exit code: 100
                    │  - DietPi version: v6.25.3 (MichaIng/master) | HW_MODEL:3 | HW_ARCH:2 | DISTRO:4
                    │  - Image creator: DietPi Core Team
                    │  - Pre-image: Raspbian Lite
                    │
                    │ Log file contents:
                    │ Setting up binutils (2.28-5) ...^M
                    │ Setting up cpp-6 (6.3.0-18+rpi1+deb9u1) ...^M
                    │ Setting up libgflags2v5 (2.1.2-4) ...^M
                    │ Setting up libusb-0.1-4:armhf (2:0.1.12-30) ...^M
                    │ Setting up libpython2.7-stdlib:armhf (2.7.13-2+deb9u3) ...^M
                    │ Setting up wiringpi (2.50) ...^M
                    │ Setting up cpp (4:6.3.0-4) ...^M
                    │ Setting up libfftw3-bin (3.3.5-3) ...^M

LexiconCode, 2019-07-28:
Collecting esptool
  Using cached https://files.pythonhosted.org/packages/6c/bf/578443af2270ebeff78fdfd30cef3a02433cc63b34a912936653a48526dd/esptool-2.7.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
    ImportError: No module named ``setuptools``

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-HuMOex/esptool/
dpkg: error processing package matrixio-creator-init (--configure):
 subprocess installed post-installation script returned error exit status 1
dpkg: dependency problems prevent configuration of matrixio-kernel-modules:
 matrixio-kernel-modules depends on matrixio-creator-init; however:

LexiconCode, 2019-07-28:
```
aplay -l
-bash: aplay: command not found
```

LexiconCode, 2019-07-28:
```
root@DietPi:~# sudo aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: Dummy [Dummy], device 0: Dummy PCM [Dummy PCM]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7

```

MichaIng, 2019-07-29:
@LexiconCode 
Wow, that package pulls a lot of dependencies, although `python-setuptools` seems to be missing, good to know. Could you paste:
```
apt show matrixio-kernel-modules
```

> Tried using https://github.com/matrix-io/matrixio-kernel-modules/blob/master/misc/asound.conf

Ah nice they provide one, even two configs. For the microphone `card "MATRIXIOSOUND"` is used. Could you check this is available? `arecord -l`

Also you might play with `alsamixer`, if anything does not work.

I don't like that the card is called `Dummy` so far, but perhaps that is expected. You did a reboot afterwards, just in case any additional module/overlay needs to be loaded on boot?

LexiconCode, 2019-07-29:
I did Reboot afterwards.
```
root@DietPi:~# apt show matrixio-kernel-modules
Package: matrixio-kernel-modules
Version: 0.1.8
Priority: optional
Section: embedded
Maintainer: MATRIX Labs <info@matrix.one>
Installed-Size: 108 kB
Depends: matrixio-creator-init, raspberrypi-kernel-headers, raspberrypi-kernel, dkms
Homepage: https://github.com/matrix-io/matrixio-kernel-modules
License: unknown
Vendor: none
Download-Size: 17.7 kB
APT-Manual-Installed: yes
APT-Sources: https://apt.matrix.one/raspbian stretch/main armhf Packages
Description: Install kernel modules for MATRIX devices.
```

root@DietPi:~# arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: Dummy [Dummy], device 0: Dummy PCM [Dummy PCM]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7

`alsamixer` shows Dummy and default soundcards

MichaIng, 2019-07-29:
@LexiconCode 
> Depends: matrixio-creator-init

Ah okay the tools are pulled as well: `apt depends matrixio-creator-init`

Does it work for playback and recording then?

LexiconCode, 2019-07-30:
> 
> 
> @LexiconCode
> 
> > Depends: matrixio-creator-init
> 
> Ah okay the tools are pulled as well: `apt depends matrixio-creator-init`
> 
> Does it work for playback and recording then?

Sure I'm looking for the to do the record/playback. Once I figure that out I'll let you know.

MichaIng, 2019-07-30:
@LexiconCode 
`aplay` and `arecord` for `.wav` files.
`mpg123` for .mp3 files (`apt install mpg123`)